dingo35/SmartEVSE-3.5 #1: Charging sometimes limites itself MinCurrent

dingo35, 2024-01-05:
see [https://gathering.tweakers.net/forum/list_message/77697986#77697986](url)

Try this version, it is a release candidate for v3.5 and it has the phase detection removed so there is no switching of override_currents anymore:
[firmware3.5-f4479b0192b.zip](https://github.com/dingo35/SmartEVSE-3.5/files/13845564/firmware3.5-f4479b0192b.zip)

thmjanssen, 2024-01-15:
Although we haven't done a lot of actual charging sessions (car remains connected though), so far the override current hasn't been adjusted since running the RC version.

HerrEck, 2024-01-16:
I charged 360 kWh without any override current issue's so far. Will give a new update in 2 weeks

HerrEck, 2024-02-01:
Still no issue's the last 2 weeks

dingo35, 2024-02-01:
Ok then I call this bug resolved in 3.5.0-RC0

closed 2024-02-01 by dingo35