dingo35/SmartEVSE-3.5 #3: Modbus server no data

rvdgaag, 2024-01-07:
Was the internal modbus server functionality removed ? Can't pull any data

dingo35, 2024-01-08:
The modbus TCP bridge is removed because there were several reports the data coming off of it is unstable; as soon as I can fix that problem it will be added back again, but until now I have been unsuccesfull in reproducing the problem.

rvdgaag, 2024-01-08:
I can test as well of you want, have a stable modbus config with 2 EV's and 2 EVSE's

dingo35, 2024-01-08:
I am currently still adding stuff from Stegen's develop branch, once I finished that Ill gladly accept your offer!

rvdgaag, 2024-01-09:
Ok, tried to install a normale 1.18 version back on the evse. One was succesfull but the other one does not respond anymore.

Any suggestions how to get this one alive again

dingo35, 2024-01-09:
1. Do you mean v1.8.0 instead of 1.18 ?
2. Even if the webpage does not respond, you can often go to http://ipadres/update and still flash. This situation can happen when you have flashed a new firmware.bin, but you haven't updated your spiffs.bin yet.... 
3. If everything else fails you can always flash by USB-cable; there is a USB-C connector at the right upper corner of the PCB; the command to flash is:
`pio run -t upload`

rvdgaag, 2024-01-09:
Ok got it back now

closed 2024-01-25 by dingo35