dingo35/SmartEVSE-3.5 #4: Display mirrored and upside-down

Imaginous, 2024-01-22:
**Describe the bug**
Today I plugged in my Tesla, which was above the set charge limit, and checked the charger. 

Just to find the display was upside-down and mirrored.
After a reboot in the webinterface it's all fine again.

**Upload your config**
`{"version":"15:31:52 @Jan  8 2024","mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Imaginous","rssi":-54,"bssid":"7C:10:C9:CD:FE:A8"},"evse":{"temp":25,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":545,"solar_stop_timer":0,"state":"Connected to EV","state_id":1,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":320,"override_current":0,"current_min":6,"current_max":32,"current_main":33,"current_max_circuit":32,"current_max_sum_mains":600,"solar_max_import":10,"solar_start_current":4,"solar_stop_time":30,"enable_C2":"Always Off","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"mqtt.imaginous.nl","port":1883,"topic_prefix":"SmartEVSE","username":"mqtt","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":5,"L1":5,"L2":0,"L3":0,"last_data_update":1705936746,"original_data":{"TOTAL":5,"L1":5,"L2":0,"L3":0}},"backlight":{"timer":0,"status":"OFF"}} `

**Upload a debug log**
No log file

**To Reproduce**
Haven't been able to reproduce it. But I replaced my unit because the previous one went black sometimes. 
It seems the "display stack" gets corrupted. 

**Expected behavior**
The screen being normal. 

**Screenshots**
[image]

fluppie, 2024-01-31:
If I'm right it's contactor related. Try to make a little airgap between de SmartEVSE and the Iskra contactor.  0.5 or 1TE should be enough. If you have the space ofcourse.

dingo35, 2024-01-31:
You should first solve the heat problem you have in your panel, all your other problems (including this one, or screens going black) are the consequence of the temperature getting too high in your panel.

The only enemy of electronics is heat! Apart from that, it is just UNSAFE to have components overheating.....

closed 2024-01-31 by dingo35

Imaginous, 2024-02-01:
I will re-arrange the components to see if this helps.
Thank you both for the suggestions.