dingo35/SmartEVSE-3.5 #5: Hompage does not show any data

daninden, 2024-01-25:
**Describe the bug**
I updated my SmartEVSE to 3.5.0-RC0, and altough the API endpoints are still working, the homepage stays blank.

It does seem to load some resources.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the homepage
2. See nothing

**Expected behavior**
To see the homepage

**Screenshots**
[image]

daninden, 2024-01-25:
I did some quick checks.. looks like the styling.css is not loading properly.

If I save the html page, set the endpoint fixed it is actually working, but only because styling.css has the wrong uri then.
If I fix the uri (include the actual ip endpoint), it starts to fail.

If i load the file directly (http://ip/styling.css) it keeps on waiting, i think the webserver has trouble serving the file (size?).

dingo35, 2024-01-25:
Cannot reproduce this. Did you reflash spiffs.bin?

daninden, 2024-01-25:
yes i did.. but I just redid it.. and now it's ok.. very strange.