dingo35/SmartEVSE-3.5 #7: MaxImport auto corrects to wrong value (RC0)

Imaginous, 2024-01-26:
**Describe the bug**
When setting *Solar Start Current* to for example -7A and the *Min Current* is 6A. The *Import current* is set to 13A.
Wanted *import current* = 0
Seems to be the change to the auto correction of the *import current* value.

**Upload your config**
`{"version":"SERKRI-3.5.0","mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Imaginous","rssi":-54,"bssid":"7C:10:C9:CD:FE:A8"},"evse":{"temp":26,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":102,"solar_stop_timer":0,"state":"Connected to EV","state_id":1,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":32,"current_main":33,"current_max_circuit":32,"current_max_sum_mains":600,"solar_max_import":13,"solar_start_current":7,"solar_stop_time":30,"enable_C2":"Always Off","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"mqtt.imaginous.nl","port":1883,"topic_prefix":"SmartEVSE","username":"mqtt","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":-7,"L1":-7,"L2":0,"L3":0,"last_data_update":1706264738,"original_data":{"TOTAL":-7,"L1":-7,"L2":0,"L3":0}},"backlight":{"timer":0,"status":"OFF"}}`

**To Reproduce**
Steps to reproduce the behavior:
1. Mode Solar?
2. Enter a new value for *Solar Start Current* in the menu or via Home Assistant... wait a few moments and the *Import Current* will be changed to the absolute value of the *Solar Start Current* + *Min Current*.

**Expected behavior**
-7A+6A=-1A so *import current* should stay 0 (or a higher set value).

dingo35, 2024-01-26:
Fixed in d9d0b79425d

closed 2024-01-26 by dingo35

dingo35, 2024-01-26:
You can test here:
[d9d0b79425-3.5RC0.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14066352/d9d0b79425-3.5RC0.zip)