dingo35/SmartEVSE-3.5 #8: Solar mode not working

Imaginous, 2024-01-26:
**Describe the bug**
I have tested with fake P1 data to get *Solar Mode* working. But it did not start.
*Solar Start Current* = 7A
*Import Current* = 0A
*Min Current* = 6A

- P1 meter send -60 (-6A) -> solar did not start as expected.
- P1 meter send -80 (-8A) -> solar still did not start, not expected. Expected the car to start charging at 8A.

**Upload your config**
`{"version":"10:01:36 @Jan 26 2024","mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Imaginous","rssi":-56,"bssid":"7C:10:C9:CD:FE:A8"},"evse":{"temp":26,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":102,"solar_stop_timer":0,"state":"Connected to EV","state_id":1,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":32,"current_main":33,"current_max_circuit":32,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":7,"solar_stop_time":5,"enable_C2":"Always Off","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"mqtt.imaginous.nl","port":1883,"topic_prefix":"SmartEVSE","username":"mqtt","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":127,"L1":127,"L2":0,"L3":0,"last_data_update":1706286654,"original_data":{"TOTAL":127,"L1":127,"L2":0,"L3":0}},"backlight":{"timer":0,"status":"OFF"}}`

**Upload a debug log**
Did only a quick test, have to find some time to do a more rigirous test.

**To Reproduce**
1. Solar mode
2. Send P1 data, negative values since we are returning power to the grid.

**Expected behavior**
The car starts charging when the P1 return current is higher then the *Solar Start Current* and that the message *"charging stopped - waiting for solar"* goes away.

dingo35, 2024-02-04:
This should have been solved in 31e2f3f964 ; it is a strange bug that seems to have been there all the time, even in the original firmware, but it only shows up after rebooting in Solar mode with an EV connected; it will move into B1 state and erroneously not get out of there...

Please test!

Imaginous, 2024-02-05:
I wonder if the fix is correct. I think it will not work with multiple active SmartEVSE's.

The old code summed all active EVSE's, the new code doesn't. Also the new code won't work if the count of active EVSE's is larger then 1.

How is the StartCurrent stored? As a positive or negative number?

This are my 2 cents.

dingo35, 2024-02-05:
The line with Totalcurrent is wrong anyways, that is the actual fix.
I simplified IsCurrentAvailable by integrating two blocks of code, if (ActiveEVSE == 0) and the else clause actually had the same formulas in them.
BUT the else clause has ActiveEVSE++ in it, so another fix is changing ActiveEVSE == 0 to ActiveEVSE == 1.

I hope to finish my automated test case for muliple Active EVSE soon, IF the test fails it is again a bug that has always been there.....

Imaginous, 2024-02-05:
I have build the firmware and will flash it tonight for testing.

I'll keep you posted.

dingo35, 2024-02-20:
Fixed with 68e74b7c3

closed 2024-02-20 by dingo35