dingo35/SmartEVSE-3.5 #11: Erronous EV wattage through eastron kwh meter

djoenez, 2024-02-10:
**Describe the bug**
While charging (and also sometimes while not charging) i get bouncing power data (like also described here: https://github.com/serkri/SmartEVSE-3/issues/230). First check was fixing the wifi (now has a wired AP 1m from the smartevses). second check was modbus cabling (rewired, twisted and shielded now). But unfortunately the power still fluctuates between the set power (3.9kw) and 2.3kw, or when not using it bounces between 0 and 2.3kw.

[image]
So on the left and right i'm charging, in the middle nothing happens. I've stood next to it, my kWh meter keeps  stating 3.9kw, but i can see in on the smartevse webpage it fluctuates from 2.3 to 3.9kw

**Upload your config**

{"version":"SERKRI-3.5.0","mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Oud-Zuid intelligence #1","rssi":-52,"bssid":"84:D8:1B:E7:B6:FE"},"evse":{"temp":32,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":1,"pwm":307,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":180,"override_current":0,"current_min":6,"current_max":18,"current_main":20,"current_max_circuit":18,"current_max_sum_mains":600,"solar_max_import":12,"solar_start_current":6,"solar_stop_time":10,"enable_C2":"Not present","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.1.80","port":1883,"topic_prefix":"EVSE1","username":"mosquitto","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron1P","address":103,"import_active_power":3.9,"total_kwh":2369,"charged_kwh":1.2,"currents":{"TOTAL":173,"L1":173,"L2":0,"L3":0},"import_active_energy":2369,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":151,"L1":26,"L2":169,"L3":-44,"last_data_update":1707570598,"original_data":{"TOTAL":151,"L1":26,"L2":169,"L3":-44}},"backlight":{"timer":120,"status":"ON"}}

**Upload a debug log**
[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 08, function: 04, reg: 0000.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=18.0 A Imeasured=18.4 A MaxCircuit=18 Imeasured_EV=17.3 A, Battery Current = 0.0 A, mode[0m=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1707569996,processed=1707569995.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=18.1 A, Idifference=0.7, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=18.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=18.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C0) Balance: EVSE0:C(18.0A),EVSE1:B1(0.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7[0m:A(0.0A),
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet(ModbusWriteMultipleRequest)(C0) address: 09, function: 0x10, reg: 0020, count: 8, values: 00b4 [0m0000 0000 0000 0000 0000 0000 0000 00b4 00a0 00ad 00b8 0000 00b4 0012 0003 0002 0000 0000 0000 0000 0000 00ad 0000 0000 0000 .
[0m[1;33m(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 18.4 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 2.6 A L2: 18.4 A L3: -5.0 A Isum: 16.0 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[1;32m(D) (Timer100ms)(C1) ModbusRequest 21: Request EVMeter Current Measurement
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 0006, data: 000c.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 67 04 18 41 8b 1a 86 00 00 00 00 00 00 00 00 45 77 cf b2 00 00 00 00 00 00 [0m00 00 
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 67 Function 04
[0m[0;32m(V) (ModbusDecode)(C0)  Register 0006(ModbusDecode)(C0)  Response
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=18.0 A Imeasured=18.4 A MaxCircuit=18 Imeasured_EV=17.3 A, Battery Current = 0.0 A, mode[0m=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1707569996,processed=1707569996.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=18.0 A, Idifference=0.7, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=18.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=18.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C1) Balance: EVSE0:C(18.0A),EVSE1:B1(0.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7[0m:A(0.0A),
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C1) Sent packet(ModbusWriteMultipleRequest)(C1) address: 09, function: 0x10, reg: 0020, count: 8, values: 00b4 [0m0000 0000 0000 0000 0000 0000 0000 00b4 00a0 00ad 00b8 0000 00b4 0012 0003 0002 0000 0000 0000 0000 0000 00ad 0000 0000 0000 .
[0m[1;33m(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 18.4 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C1) L1: 2.6 A L2: 18.4 A L3: -5.0 A Isum: 16.0 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=18.0 A Imeasured=18.4 A MaxCircuit=18 Imeasured_EV=17.3 A, Battery Current = 0.0 A, mode[0m=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1707569997,processed=1707569996.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=18.1 A, Idifference=0.7, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=18.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=18.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C0) Balance: EVSE0:C(18.0A),EVSE1:B1(0.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7[0m:A(0.0A),
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet(ModbusWriteMultipleRequest)(C0) address: 09, function: 0x10, reg: 0020, count: 8, values: 00b4 [0m0000 0000 0000 0000 0000 0000 0000 00b4 009f 00ad 00b8 0000 00b4 0012 0003 0002 0000 0000 0000 0000 0000 00ad 0000 0000 0000 .
[0m[1;33m(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 18.4 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 2.6 A L2: 18.4 A L3: -5.1 A Isum: 15.9 A
[0m[1;32m(D) (Timer100ms)(C1) ModbusRequest 5: Request Energy Node 1
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 66, function: 04, reg: 0048, data: 0002.
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 66, function: 04, reg: 000c, data: 0002.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 66 04 04 43 6d 6e 98 
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04
[0m[0;32m(V) (ModbusDecode)(C0)  Register 000c(ModbusDecode)(C0)  Response
[0m[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 66 04 04 00 00 00 00 
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04
[0m[0;32m(V) (ModbusDecode)(C0)  Register 0000(ModbusDecode)(C0)  Response
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 02, function: 04, reg: 0000, data: 0008.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (19 bytes) 02 04 10 00 09 00 00 00 00 00 01 00 00 00 00 00 00 00 11 
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 02 Function 04
[0m[0;32m(V) (ModbusDecode)(C0)  Register 0000(ModbusDecode)(C0)  Response
[0m[1;32m(D) (receiveNodeStatus)(C0) ReceivedNode[1]Status State:9 Error:0, BalancedMax:170, Mode:1, ConfigChanged:0.
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 03, function: 04, reg: 0000, data: 0008.
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 04, function: 04, reg: 0000, data: 0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 03, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 05, function: 04, reg: 0000, data: 0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 04, function: 04, reg: 0000.
[... cut ...] 0 .
[0m[1;33m(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 18.3 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 2.7 A L2: 18.3 A L3: -5.1 A Isum: 15.9 A

**To Reproduce**
Steps to reproduce the behavior:
Doesn't matter which mode.

**Expected behavior**
No more bouncing to random signals.

**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.

dingo35, 2024-02-10:
From your log I derived your EVMeter is at address 67; when you `grep 67 logfile` you get:
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 0006, data: 000c.
�[0m�[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0) (27 bytes) 67 04 18 41 8b 1a 86 00 00 00 00 00 00 00 00 45 77 cf b2 00 00 00 00 00 00 �[0m00 00
�[0m�[0;32m(V) (ModbusDecode)(C0) valid Modbus packet: Address 67 Function 04
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 0006, data: 000c.
�[0m�[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0) (27 bytes) 67 04 18 41 8b 17 d6 00 00 00 00 00 00 00 00 45 77 ef 69 00 00 00 00 00 00 �[0m00 00
�[0m�[0;32m(V) (ModbusDecode)(C0) valid Modbus packet: Address 67 Function 04
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 0048, data: 0002.
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 000c, data: 0002.
�[0m�[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0) (7 bytes) 67 04 04 45 14 05 e7
�[0m�[0;32m(V) (ModbusDecode)(C0) valid Modbus packet: Address 67 Function 04
�[0m�[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0) (7 bytes) 67 04 04 45 77 ef 69
�[0m�[0;32m(V) (ModbusDecode)(C0) valid Modbus packet: Address 67 Function 04
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 0006, data: 000c.
�[0m�[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0) (27 bytes) 67 04 18 41 8b 20 88 00 00 00 00 00 00 00 00 45 77 d4 89 00 00 00 00 00 00 �[0m00 00
�[0m�[0;32m(V) (ModbusDecode)(C0) valid Modbus packet: Address 67 Function 04
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 0006, data: 000c.
�[0m�[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0) (27 bytes) 67 04 18 41 8b 27 d9 00 00 00 00 00 00 00 00 45 77 cf 76 00 00 00 00 00 00 �[0m00 00
�[0m�[0;32m(V) (ModbusDecode)(C0) valid Modbus packet: Address 67 Function 04
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 0048, data: 0002.
�[0m�[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0) (7 bytes) 67 04 04 45 14 05 fc
�[0m�[0;32m(V) (ModbusDecode)(C0) valid Modbus packet: Address 67 Function 04
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 000c, data: 0002.
�[0m�[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0) (7 bytes) 67 04 04 45 77 e0 0a
�[0m�[0;32m(V) (ModbusDecode)(C0) valid Modbus packet: Address 67 Function 04
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 0006, data: 000c.
�[0m(MBhandleError)(C0) Error response: E2 - CRC check error, address: 67, function: 04, reg: 0006.
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 0006, data: 000c.
�[0m(MBhandleError)(C0) Error response: E2 - CRC check error, address: 67, function: 04, reg: 0006.
�[0m�[0;32m(V) (ModbusSend8)(C1) address: 67, function: 04, reg: 0048, data: 0002.
�[0m�[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0) (7 bytes) 67 04 04 45 14 06 14
�[0m�[0;32m(V) (ModbusDecode)(C0) valid Modbus packet: Address 67 Function 04
�[0m�[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 000c, data: 0002.
�[0m(MBhandleError)(C0) Error response: E2 - CRC check error, address: 67, function: 04, reg: 000c.

The CRC errors in the end of the log file give you those errors, so you seem to have a modbus problem.
Steps to take:
1.  software: please compile with the old modbus-fix, both Stegen and I suspect that that fix is not necessary anymore, but it might make a difference in your situation; just add `extra_scripts = pre:eModbus-fix/fix.py` just before `build_flags` in platformio.ini ; if you checkout v1.8.0 you can see it there.

Please report back with a log as you did before; please take this step first, because I would like to know if this improves your situation, even if your modbus would be unstable....

2. hardware: physically check your modbus wiring; at the ends add 120ohm terminator resistors; make sure you use twisted pair

djoenez, 2024-02-10:
Thanks for your response. I've never compiled anything. so when i have the time i'll read up on that. In the mean time i'll order the terminator. Also while rewiring i changed the order (it was first the third, now the last in the row of 4, both have the same issue). I'll keep you posted.

dingo35, 2024-02-10:
[bd9c6eeb5790-with-modbus-fix.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14230156/bd9c6eeb5790-with-modbus-fix.zip)

Tried to save you some time, this is my current post 3.5.0-RC0 version with modbus fix enabled. Please report back.

djoenez, 2024-02-10:
That indeed saves some time. thanks. Unfortunately it is still there, error signal is always in the range of 2368 to 2377.

[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 08, function: 04, reg: 0000.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=18.0 A Imeasured=18.4 A MaxCircuit=18 Imeasured_EV=17.3 A, Battery Current = 0.0 A, mode[0m=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1707569996,processed=1707569995.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=18.1 A, Idifference=0.7, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=18.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=18.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C0) Balance: EVSE0:C(18.0A),EVSE1:B1(0.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7[0m:A(0.0A),
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet(ModbusWriteMultipleRequest)(C0) address: 09, function: 0x10, reg: 0020, count: 8, values: 00b4 [0m0000 0000 0000 0000 0000 0000 0000 00b4 00a0 00ad 00b8 0000 00b4 0012 0003 0002 0000 0000 0000 0000 0000 00ad 0000 0000 0000 .
[0m[1;33m(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 18.4 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 2.6 A L2: 18.4 A L3: -5.0 A Isum: 16.0 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[1;32m(D) (Timer100ms)(C1) ModbusRequest 21: Request EVMeter Current Measurement
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 67, function: 04, reg: 0006, data: 000c.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 67 04 18 41 8b 1a 86 00 00 00 00 00 00 00 00 45 77 cf b2 00 00 00 00 00 00 [0m00 00 
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 67 Function 04
[0m[0;32m(V) (ModbusDecode)(C0)  Register 0006(ModbusDecode)(C0)  Response
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=18.0 A Imeasured=18.4 A MaxCircuit=18 Imeasured_EV=17.3 A, Battery Current = 0.0 A, mode[0m=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1707569996,processed=1707569996.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=18.0 A, Idifference=0.7, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=18.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=18.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C1) Balance: EVSE0:C(18.0A),EVSE1:B1(0.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7[0m:A(0.0A),
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C1) Sent packet(ModbusWriteMultipleRequest)(C1) address: 09, function: 0x10, reg: 0020, count: 8, values: 00b4 [0m0000 0000 0000 0000 0000 0000 0000 00b4 00a0 00ad 00b8 0000 00b4 0012 0003 0002 0000 0000 0000 0000 0000 00ad 0000 0000 0000 .
[0m[1;33m(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 18.4 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C1) L1: 2.6 A L2: 18.4 A L3: -5.0 A Isum: 16.0 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=18.0 A Imeasured=18.4 A MaxCircuit=18 Imeasured_EV=17.3 A, Battery Current = 0.0 A, mode[0m=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1707569997,processed=1707569996.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=18.1 A, Idifference=0.7, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=18.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=18.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C0) Balance: EVSE0:C(18.0A),EVSE1:B1(0.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7[0m:A(0.0A),
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet(ModbusWriteMultipleRequest)(C0) address: 09, function: 0x10, reg: 0020, count: 8, values: 00b4 [0m0000 0000 0000 0000 0000 0000 0000 00b4 009f 00ad 00b8 0000 00b4 0012 0003 0002 0000 0000 0000 0000 0000 00ad 0000 0000 0000 .
[0m[1;33m(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 18.4 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 2.6 A L2: 18.4 A L3: -5.1 A Isum: 15.9 A
[0m[1;32m(D) (Timer100ms)(C1) ModbusRequest 5: Request Energy Node 1
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 66, function: 04, reg: 0048, data: 0002.
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 66, function: 04, reg: 000c, data: 0002.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 66 04 04 43 6d 6e 98 
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04
[0m[0;32m(V) (ModbusDecode)(C0)  Register 000c(ModbusDecode)(C0)  Response
[0m[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 66 04 04 00 00 00 00 
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04
[0m[0;32m(V) (ModbusDecode)(C0)  Register 0000(ModbusDecode)(C0)  Response
[... cut ...]
[0m[1;32m(D) (ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 02, function: 04, reg: 0000, data: 0008.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (19 bytes) 02 04 10 00 09 00 00 00 00 00 01 00 00 00 00 00 00 00 11 
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 02 Function 04
[0m[0;32m(V) (ModbusDecode)(C0)  Register 0000(ModbusDecode)(C0)  Response
[0m[1;32m(D) (receiveNodeStatus)(C0) ReceivedNode[1]Status State:9 Error:0, BalancedMax:170, Mode:1, ConfigChanged:0.
[0m(operator())(C1) page / (root) requested and sent

dingo35, 2024-02-10:
Well sad for you, but good for the community, that means that the modbus-fix might be unnecessary after all.....

I coincidentally just fixed a bug that has to do with the modbus communication on the EVmeter part, it might also solve your problem:
[13ac242f.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14230514/13ac242f.zip)

djoenez, 2024-02-11:
Hello Dingo, i tried your bugfix. unfortunately the same issue while charging, i don't see any bouncing signals while not charging though! I'll update you once i have the resistor placed. I'll also check if removing and adding ground does anything.

I still find it weird, if the modbus message would be failing, why is it always 2.3xx W? (while the kWh meter says 3900W)

dingo35, 2024-02-11:
I suspect there is a problem in configurations with relatively high modbus traffic: the code uses one single global set of registers for all modbus requests/responses, and answers to one modbus message _might_ mess up another; if it is static data (e.g. config info) that your EVMeter response messes up with, you get the same response every time.

E.g. I had the EVmeter response messing up my config data, constantly the Slave was reporting it was on Mode 186, while only [0-2] are allowed; memory management of the Arduino somehow always reads the same memory address/value, while you would expect random values.

dingo35, 2024-02-11:
If you want I can post a version that has some extended logging on the handling of your modbus messages, so we can see whats going on....

djoenez, 2024-02-11:
If it can help in anyway getting this fixed i'm glad to put some effort in, just tell me what to do :)

dingo35, 2024-02-11:
[db983b41d1.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14233581/db983b41d1.zip)
Flash this and try to log one of those wrong readings....

djoenez, 2024-02-11:
Here it switches from bad to good to bad again
[putty_export.txt](https://github.com/dingo35/SmartEVSE-3.5/files/14233650/putty_export.txt)

dingo35, 2024-02-12:
It is definitely your modbus that is having problems. Please post your /settings of both Master and Slave so we can see your devices. Please also describe which devices are on your modbus. Only 1 master, the SmartEVSE on the bus?

Bogus data from your device at 0x66 is received by the SmartEVSE:
`ESC[0mESC[1;32m(D) (Timer100ms)(C1) ModbusRequest 5: Request Energy Node 1
ESC[0mESC[1;32m(D) (ModbusSend8)(C1) Sent packet address: 66, function: 04, reg: 0048, token:66040048, data: 0002.
ESC[0m(MBhandleError)(C0) Error response: E2 - CRC check error, address: 66, function: 04, reg: 0048.
ESC[0m(CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=18.0 A Imeasured=17.2 A MaxCircuit=18 Imeasured_EV=17.1 A, Battery Current = 0.0 A, mode=1.
ESC[0mESC[1;32m(D) (ModbusSend8)(C1) Sent packet address: 66, function: 04, reg: 000c, token:6604000c, data: 0002.
ESC[0mESC[1;32m(D) (ModbusDecode)(C0) Received packet (3 bytes) 66 84 00
ESC[0mESC[1;32m(D) (ModbusSend8)(C1) Sent packet address: 02, function: 04, reg: 0000, token:02040000, data: 0008.
ESC[0mESC[1;32m(D) (ModbusDecode)(C0) Modbus Exception 0, Address=102, Function=132.
ESC[0mESC[0;32m(V)  Register 0000(MBhandleData)(C0) ERROR: MB.Function=132, token_function=4.
ESC[0mESC[1;32m(D) (ModbusSend8)(C1) Sent packet address: 03, function: 04, reg: 0000, token:03040000, data: 0008.
ESC[0mESC[0;32m(V) (MBhandleError)(C0) Error response: E2 - CRC check error, address: 02, function: 04, reg: 0000.
ESC[0mESC[1;32m(D) (ModbusSend8)(C1) Sent packet address: 04, function: 04, reg: 0000, token:04040000, data: 0008.
`

The request Energy node 1 (address: 66, function: 04, reg: 0048) generates a crc error AND a Modbus Exception because of the 3 byte response 66 84 00 .

Possibly on of your devices has a hardware problem. Another thing you could try is put them on addresses 101, 102 etc., multiple users have reported problems were solved then; I vaguely remember that was with Eastron devices..

djoenez, 2024-02-12:
Hi Dingo,

So the modbus goes like;EVSE1-EVSE2-KWHMETEREVSE2(ADD102) -
KWHMETEREVSE1(ADD103).
they have always been on these addresses, (just swapped kwhmetrtevse1 from
101 to 103 to check). I verified with the evses and there the addresses
also check out. So I'm wondering what is in address 66

dingo35, 2024-02-12:
0x66 = 102 decimal.

dingo35, 2024-02-12:
No 101 and 102 are the decimal values!

dingo35, 2024-02-12:
Also, I notice you have one Eastron3P meter and one Eastron1P meter on your modbus, are you sure that is correct?

I would expect 3p meters in a 3p installation, and 1p meters in a 1p installation.

djoenez, 2024-02-12:
it is correct, they are actually both 1 phase at the moment (waiting for
good weather for a cable upgrade).
I tried the 3P(1P attached) meter both with the 3p or 1p setting in the
EVSE, with both the setting for 1 phase or 3 phase in the Eastron kWh meter
(it has a setting for this particular usecase). all end up with the same
result.

dingo35, 2024-02-12:
No max address is 247 decimal, so put them on 101 and 102 decimal

A 3p Eastron should be configured 3p, irrespective of how its wired.

Eastron has a little quirk: if you are connecting a 3p Eastron only on 1p, you need to feed that 1p to all three phases on the meter, or it will have strange behaviour (like spontaneous reboots).

djoenez, 2024-02-12:
Ok, i'll put the 3phase back to 101 then just to be sure, and configure again as a 3P. i'll wire all phases to L1 as soon as i have some time :)

Thanks!

rvdgaag, 2024-02-16:
I still have similar issues, even with the new version here in this topic. The new version shows the Phase totals now also in the Node, not just in the Master (Master receives the data via API from HA every 2 seconds:

The totals of the EV meter are shown correct (total_kwh":177.1,"charged_kwh":0.6,) but the live TOTAL":0,"L1":0,"L2":0,"L3":0 never appear

{"version":"11:29:11 @Feb 11 2024","mode":"OFF","mode_id":0,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"McWifi_IoT","rssi":-48,"bssid":"D0:21:F9:73:17:8A"},"evse":{"temp":25,"temp_max":65,"connected":true,"access":false,"mode":1,"loadbl":2,"pwm":1024,"solar_stop_timer":0,"state":"Charging Stopped","state_id":9,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":160,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"solar_max_import":2,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Solar Off","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.1.11","port":1883,"topic_prefix":"SmartEVSE2","username":"energie","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":102,"import_active_power":0,"total_kwh":177.1,"charged_kwh":0.6,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":177.1,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":13,"L1":-2,"L2":1,"L3":14,"last_data_update":0,"original_data":{"TOTAL":0,"L1":0,"L2":0,"L3":0}},"backlight":{"timer":0,"status":"OFF"}}

rvdgaag, 2024-02-16:
(setState)(C0) 14:20:38 STATE COMM_C_OK -> C
(I) (setState)(C1) Switching CONTACTOR C2 ON.
(D) (BroadcastWorker)(C0) Received broadcast packet, reg=0030 (15 bytes) 00 00 30 ff ff ff fd 00 00 00 00 ff ff ff fb
(V) (BroadcastWorker)(C0) Irms from MainsMeter received: Index=7, L1=-0.3A,Index=11, L2=0.0A,Index=15, L3=-0.5A,
(D) (ModbusDecode)(C0) Received packet (23 bytes) 09 10 00 20 00 08 10 00 00 00 a0 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10 Register 0020 Request
(V) (MBbroadcast)(C0) Broadcast received, Node 16.0 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (printStatus)(C1) L1: -0.3 A L2: 0.0 A L3: -0.5 A Isum: -0.8 A
(D) (BroadcastWorker)(C0) Received broadcast packet, reg=0030 (15 bytes) 00 00 30 ff ff ff fd 00 00 00 00 ff ff ff fb
(V) (BroadcastWorker)(C0) Irms from MainsMeter received: Index=7, L1=-0.3A,Index=11, L2=0.0A,Index=15, L3=-0.5A,
(D) (ModbusDecode)(C0) Received packet (23 bytes) 09 10 00 20 00 08 10 00 00 00 a0 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10 Register 0020 Request
(V) (MBbroadcast)(C0) Broadcast received, Node 16.0 A
(D) (ModbusDecode)(C0) Received packet (6 bytes) 02 04 00 00 00 08
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 02 Function 04 Register 0000 Request
(D) (MBNodeRequest)(C0) Node answering NodeStatus request value[0]=2 value[1]=0 value[2]=160 value[3]=1 value[4]=0 value[5]=1 value[6]=0 value[7]=16
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (printStatus)(C1) L1: -0.3 A L2: 0.0 A L3: -0.5 A Isum: -0.8 A
(D) (BroadcastWorker)(C0) Received broadcast packet, reg=0030 (15 bytes) 00 00 30 ff ff ff fd 00 00 00 00 ff ff ff fb
(V) (BroadcastWorker)(C0) Irms from MainsMeter received: Index=7, L1=-0.3A,Index=11, L2=0.0A,Index=15, L3=-0.5A,
(D) (ModbusDecode)(C0) Received packet (23 bytes) 09 10 00 20 00 08 10 00 00 00 a0 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10 Register 0020 Request
(V) (MBbroadcast)(C0) Broadcast received, Node 16.0 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (printStatus)(C1) L1: -0.3 A L2: 0.0 A L3: -0.5 A Isum: -0.8 A
(D) (BroadcastWorker)(C0) Received broadcast packet, reg=0030 (15 bytes) 00 00 30 ff ff ff fd 00 00 00 00 ff ff ff fb
(V) (BroadcastWorker)(C0) Irms from MainsMeter received: Index=7, L1=-0.3A,Index=11, L2=0.0A,Index=15, L3=-0.5A,
(D) (ModbusDecode)(C0) Received packet (23 bytes) 09 10 00 20 00 08 10 00 00 00 a0 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10 Register 0020 Request
(V) (MBbroadcast)(C0) Broadcast received, Node 16.0 A
(D) (ModbusDecode)(C0) Received packet (6 bytes) 66 04 00 48 00 02
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04 Register 0048 Request
(D) (ModbusDecode)(C0) Received packet (7 bytes) 66 04 04 43 31 2c 08
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04 Register 0048 Response
(D) (ModbusDecode)(C0) Received packet (6 bytes) 66 04 00 34 00 02
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04 Register 0034 Request
(D) (ModbusDecode)(C0) Received packet (7 bytes) 66 0( 0p4rintSta 8s(C1 )1 0STA T7E9: C
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04 Register 0034 Request
(D) (ModbusDecode)(C0) Received packet (7 bytes) 66 0( 0p4rintSta 8s(C1 )1 0STA T7E9: C
(V)  Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (printStatus)(C1) L1: -0.3 A L2: 0.0 A L3: -0.5 A Isum: -0.8 A
(D) (ModbusDecode)(C0) Received packet (6 bytes) 02 04 00 00 00 08
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 02 Function 04 Register 0000 Request
(D) (MBNodeRequest)(C0) Node answering NodeStatus request value[0]=2 value[1]=0 value[2]=160 value[3]=1 value[4]=0 value[5]=1 value[6]=0 value[7]=16
(D) (BroadcastWorker)(C0) Received broadcast packet, reg=0030 (15 bytes) 00 00 30 ff ff ff fd ff ff ff ff ff ff ff f8
(V) (BroadcastWorker)(C0) Irms from MainsMeter received: Index=7, L1=-0.3A,Index=11, L2=-0.1A,Index=15, L3=-0.8A,
(D) (ModbusDecode)(C0) Received packet (23 bytes) 09 10 00 20 00 08 10 00 00 00 a0 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10 Register 0020 Request
(V) (MBbroadcast)(C0) Broadcast received, Node 16.0 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (printStatus)(C1) L1: -0.3 A L2: -0.1 A L3: -0.8 A Isum: -1.2 A
(D) (BroadcastWorker)(C0) Received broadcast packet, reg=0030 (15 bytes) 00 00 30 ff ff ff fd ff ff ff ff ff ff ff f8
(V) (BroadcastWorker)(C0) Irms from MainsMeter received: Index=7, L1=-0.3A,Index=11, L2=-0.1A,Index=15, L3=-0.8A,
(D) (ModbusDecode)(C0) Received packet (23 bytes) 09 10 00 20 00 08 10 00 00 00 a0 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10 Register 0020 Request
(V) (MBbroadcast)(C0) Broadcast received, Node 16.0 A
[... cut ...]
(V) (MBbroadcast)(C0) Broadcast received, Node 16.0 A
(D) (ModbusDecode)(C0) Received packet (6 bytes) 02 04 00 00 00 08
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 02 Function 04 Register 0000 Request
(D) (MBNodeRequest)(C0) Node answering NodeStatus request value[0]=2 value[1]=0 value[2]=160 value[3]=1 value[4]=0 value[5]=1 value[6]=0 value[7]=16
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A

dingo35, 2024-02-19:
@rvdgaag I could reproduce your problem once, then I reflashed my spiffs.bin and since then I cannot reproduce it.

Does reflashing spiffs.bin help you?

rvdgaag, 2024-02-19:
OK I will try, the second car will probably be loaded again by the end of the week. I'll post the result

dingo35, 2024-02-19:
Seconds question, since I currently can only test with EVMeter on EM_API: if it doesn't work, did it EVER work?
I think this might be untested in both the original 3.0.1 (no currents of EV meter displayed there), as in every Serkri version ever (since no Slave device was available to me to test this)?

If it did ever work, could you find it which was the last version where the EVmeter currents on a Slave SmartEVSE were shown correctly on display and/or webserver screen?

dingo35, 2024-02-20:
Then don't bother testing that version, it was never coded in, and you are the first one to notice.

I don't have a modbus EVMeter available in my test setup now, so I cannot test it myself, but this version should work:
[fba23e963.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14343097/fba23e963.zip)

rvdgaag, 2024-02-20:
Wow thanks, yes it works fine now, also publishing data via MQTT to HA !

This weekend I will test charging the Master and Node 1 at the same time to see what happens but so far so good !

djoenez, 2024-02-25:
hi, so i tried a couple of things:
- Added 120 ohms resistors at start and end --> no improvement
- Attached L2 and L3 on the SDM72 --> no improvement
- Changed out SDM72 for another one --> no improvement (although this one pingponged between 0W and the set value, not ~2400W)
- removed SDM230 out of the change --> no improvement

Also tried above combinations, no improvement unfortunately.

I'm a bit at a loss here, i am using twisted pair wire, but it is not shielded, just bare wire. could it have something to do with it? else i really don't know where to look for to getting this working.

dingo35, 2024-02-25:
Shielding should not be necessay on those short distances, unless you live at a location with strong transmitters ....

Sorry @djoenez but this looks like some hardware problem, and I cant solve that with my software....

rvdgaag, 2024-02-26:
@djoenez, a simple twisted pair wire can be the cause of these issues. I'm using special modbus cable since modbus is vulnerable for cable reflections and external interference (230V 50 HZ 16A causes a lot of interference).

https://webshop.cedel.nl/nl/RS-485-kabel-(Modbus)-Halogeenvrij?gad_source=1&gclid=Cj0KCQiA5-uuBhDzARIsAAa21T9kU6kNxKO9Za59tcjw6xy_xgx-n6i8f2q_cajuhy2WhDY_7_Z549caAuPSEALw_wcB

djoenez, 2024-02-26:
Thanks for the tip! I'll try that. Although the shipping costs are "not maps".

djoenez, 2024-03-04:
So I changed the wiring to the official modbus wiring. And the same thing happens. I'll ask on the tweakers forum if anybody else has issues with the sdm72v2.

I think I swapped out everything now, so I don't know where it comes from :(

rvdgaag, 2024-03-04:
The registers of  sdm72 are different from the  sdm230 so to me it's kind of logic it doesn't work properly.

dingo35, 2024-03-05:
The Eastron single phase meters (like the sdm230) should be configured at "Eastron1p", the 3 phase meters (sdm630, sdm72) as "Eastron3p".

The registers are coded correctly, I am using two SDM72D-M (one v1 and one v2) in my production environment..

djoenez, 2024-03-05:
They are set as 3f and 1f in the respective EVSES. It is not caused by the sdm72. As I have swapped that one for another sdm72 and the same issue was there.

dingo35, 2024-03-05:
Ok I have found a possible problem: in the modbus architecture, the Master device fires a request, and the addressed Slave device answers with a response. 
It might be possible, either by mistake or by disturbances in the modbus, that a request is sent to a Slave while the response of the previous request has not arrived yet. This might lead to a mixup of values.

This version detects (but cannot correct) these kind of situations:
[827a3cf8b23.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14498219/827a3cf8b23.zip)

Please look in the telnet log for strings like:
"Error outstanding request address ......." 
and post the lines before this error line that request that address/function combination, and response to the request.

djoenez, 2024-03-06:
Updated both EVSES, this is the output of the master:

[0m[1;32m(D) (CalcBalancedCurrent)(C0) Balance: EVSE0:C(18.0A) EVSE1:A(0.0A) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(0.0A) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 28, values: 00b4 0000(CalcBalancedCurrent)(C1)[0m Checkpoint 1 Isetbalanced=18.0 A Imeasured=16.0 A MaxCircuit=18 Imeasured_EV=17.2 A, Battery Current = 0.0 A, mode=2.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=18.0 A, Idifference=0.8, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 3 Isetbalanced=14.1 A, IsumImport=7.9, Isum=7.9, ImportCurrent=0.
[0m[1;32m(D) (CalcBalancedCurrent)(C1) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=14.1A, ActiveEVSE=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkp int 4 Isetbalanced=14.1  A0.0
[0m[0;32m(V) ff ffa5(CalcBalance d6C7u9r0rent)(C 15)a 6Bfalance: b5c 3ffc 0008 00E 0VSE0:C( 144b.812A)  3ffc E4b74E1:A(0 .30fAf)c  0000EVSE2:A (000.000A)  59d0EVSE3:A(0.0A) EVSE4:A(0.0A[0m) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m(ModbusWriteMultipleRequest)(C1) Error outstanding request address: 0x09, function: 0x10. Overwriting request with reg: 0x0020, count: 28.
[0m 3ffb
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C1) Sent packet(printStatus)(C0) STATE:  aCd dErrersosr::  00x 0S9t,a rftuCnurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: [0m[0mvau0e sI:measured:  1060.80d A Iset B0a0l0a0nced: 1 40.010 0A
[0m[0;32m(V) f [0;32m(V) fa4 ffff 6790 5a6f 4b5c 3ffc 0008 0000 4b82 3ffc 4b74 3ffc 5954 8019 2090 3ffb
[0m[1;33m(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 16.0 A IsetBalanced: 14.1 A
[0m[1;33m(I) (printStatus)(C1) L1: 1.1 A L2: 16.0 A L3: -9.2 A Isum: 7.9 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 1
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x66, function: 0x04, reg: 0x0048, token:0x66040048, data: 0x0002.
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m(ModbusSend8)(C0) Error outstanding request address: 0x66, function: 0x04. Overwriting Request with reg: 0x000c, data: 0x0002.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x66, function: 0x04, reg: 0x000c, token:0x6604000c, data: 0x0002.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (7 bytes) 66 04 04 43 ce c9 37
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x66 Function 0x04 Register 0x000c Response
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (7 bytes) 66 04 04 00 00 00 00
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x66 Function 0x04 Register 0x0000 Response
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x02, function: 0x04, reg: 0x0000, token:0x02040000, data: 0x0008.

Sidenote, this version did not take over my custom MQTT prefix, so HA got all in a fritz.

dingo35, 2024-03-06:
As I suspected, two errors in there.

As to the MQTT, I forgot to mention, breaking change: we got rid of that strange MAC derived nr in the MQTT prefix, it is the serial number now!

djoenez, 2024-03-06:
ah, i'm "happy" that i'm not crazy :) Thanks for making this check version to see if this is the issue.

No worries on the MQTT breaking change, just had to rename the prefix and reboot HA to get it working again.

dingo35, 2024-03-06:
Just as I suspected:
'''�[0m�[1;32m(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 1
�[0m�[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x66, function: 0x04, reg: 0x0048, token:0x66040048, data: 0x0002.
�[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
�[0m(ModbusSend8)(C0) Error outstanding request address: 0x66, function: 0x04. Overwriting Request with reg: 0x000c, data: 0x0002.
�[0m�[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x66, function: 0x04, reg: 0x000c, token:0x6604000c, data: 0x0002.'''

You see here that when the EVSE requests the energy information from your 0x66 meter, it is overwriting the outstanding request with register 0x000c, which is the request for the power information. 
So for some reason your meters response is so slow that it fails to respond  in time; and then our firmware is not handling that very well, because it is impossible to distinguish between the different responses. A failure of the 1979 modbus protocol is that the response does not include any pointers to the request it is the response to...

Ok we'll work on this, but the underlying problem is that your meter is not responding fast enough; might be meter problem or modbus-hardware problem....
My job is to make sure that this erroneous situation does not get the data of the two requests swapped...

djoenez, 2024-03-06:
Check. So I verified that it is not an issue with this specific sdm72v2 unit (swapped with a different 72v2) .

Thanks for following up. Is it maybe possible to reserve different registers for different requests?

djoenez, 2024-03-06:
Update: With this new version the topic prefix is not saved after a reboot. It reverts to the now new prefix.

djoenez, 2024-03-08:
I have updated both EVSE's to your version. I'll monitor during the day. for now it looks good (EVSE2 is solar charging, and i'm not seeing peaks on my EVSE1 EV charge wattage).

djoenez, 2024-03-08:
Unfortunately it is still there, from 9 to 9:40 EVSE2 was charging and there are again peaks, from 9:40 EVSE1 is charging (to test solar mode) and there are also peaks.
[image]

dingo35, 2024-03-08:
@djoenez Is this about "erroneous EV wattage" (see title) or about unstable charging?

djoenez, 2024-03-08:
This is about the spikes in EV wattage on EVSE1, (i also had a total kwh from EVSE2 that was 0).

On above picture you can see the EV charge power from EVSE1 (master), before 9:00 AM both EVSEs were not charging (and EVSE1 gave spikes to 2523W. from 9:00 am EVSE2 was charging on solar mode. From 9:40 i switched to have EVSE1 solar charge (EVSE2 off), then the spikes do not go from zero to 2523, but from 1209W(the actual charge power) to 2523.

dingo35, 2024-03-08:
Do the logs still give errors on outstanding requests? If so please post...

dingo35, 2024-03-09:
We changed some timings on the modbus, this should solve the problem:
[05612caad1.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14547712/05612caad1.zip)

djoenez, 2024-03-10:
I updated both the EVSE1 and EVSE2 with this version, the cars do not start to charge when going to smart, see below EVSE1 Logging (after a reboot the car charges for 10s, then stops):

[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
[0m(ModbusSend8)(C1) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x65, function: 0x04, reg: 0x0048, token:0x65040048, data: 0x0002.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x65, function: 0x04, reg: 0x0034, token:0x65040034, data: 0x0002.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x02, function: 0x04, reg: 0x0000, token:0x02040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x03, function: 0x04, reg: 0x0000, token:0x03040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x04, function: 0x04, reg: 0x0000, token:0x04040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x05, function: 0x04, reg: 0x0000, token:0x05040000, data: 0x0008.
[0m(ModbusSend8)(C1) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x06, function: 0x04, reg: 0x0000, token:0x06040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x07, function: 0x04, reg: 0x0000, token:0x07040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x08, function: 0x04, reg: 0x0000, token:0x08040000, data: 0x0008.
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
[0m(ModbusSend8)(C1) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x65, function: 0x04, reg: 0x0006, token:0x65040006, data: 0x000c.
[0m[1;32m(D) (Timer100ms)(C1) ModbusRequest 5: Request Energy Node 0
[0m(ModbusSend8)(C1) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x65, function: 0x04, reg: 0x0048, token:0x65040048, data: 0x0002.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x65, function: 0x04, reg: 0x0034, token:0x65040034, data: 0x0002.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x02, function: 0x04, reg: 0x0000, token:0x02040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x03, function: 0x04, reg: 0x0000, token:0x03040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x04, function: 0x04, reg: 0x0000, token:0x04040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x05, function: 0x04, reg: 0x0000, token:0x05040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x06, function: 0x04, reg: 0x0000, token:0x06040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x07, function: 0x04, reg: 0x0000, token:0x07040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x08, function: 0x04, reg: 0x0000, token:0x08040000, data: 0x0008.
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x65, function: 0x04, reg: 0x0006, token:0x65040006, data: 0x000c.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=18.0 A Imeasured=1.0 A MaxCircuit=18 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1710063105,processed=1710063069.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=22.5 A, Idifference=18.0, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=18.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=18.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C1) Balance: EVSE0:B1(0.0A) EVSE1:A(0.0A) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(0.0A) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m(ModbusWriteMultipleRequest)(C1) Error creating request: 0xe8 - Request queue full
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C1) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 11, values: 0000 0000 0000 0000 0000 0000 0000[0m 0000 000a 0003 ffd8
[0m[1;33m(I) (printStatus)(C1) STATE: B1 Error: 10 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.0 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C1) L1: 1.0 A L2: 0.3 A L3: -4.0 A Isum: -2.7 A
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
[0m(ModbusSend8)(C1) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x65, function: 0x04, reg: 0x0048, token:0x65040048, data: 0x0002.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x65, function: 0x04, reg: 0x0034, token:0x65040034, data: 0x0002.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[... cut ...]
[0m[0;32m(V)  address: 0x65, function: 0x04, reg: 0x0048, token:0x65040048, data: 0x0002.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x65, function: 0x04, reg: 0x0034, token:0x65040034, data: 0x0002.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x02, function: 0x04, reg: 0x0000, token:0x02040000, data: 0x0008.
[0m(ModbusSend8)(C0) Error creating request: 0xe8 - Request queue full
[0m[0;32m(V)  address: 0x03, function: 0x04, reg: 0x0000, token:0x03040000, data: 0x0008.

dingo35, 2024-03-10:
Wow you managed to get your modbus request queue full, never saw that before.
I just have one more inspiration, upgrading the modbus library from version 1.7.0 to 1.7.1, we are at unknown territories here....

[c9c27272a.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14549911/c9c27272a.zip)

My fellow developer had Eastron kWh meter on his test rig with a scope on it so analysed all behaviour, and that resulted in those two changes.

Are you sure you are not pounding/overflowing your ESP32 with API/MQTT requests?

djoenez, 2024-03-12:
No problem. I very much apriciate the effort. Will see if I have some time tonight to test this version. Will keep you updated!

dingo35, 2024-03-12:
Great thanks! Please be aware the release of 3.5.0-RC2 is awaiting your test results :-)

djoenez, 2024-03-12:
There are less peaks, but still some on EVSE1 where im not charging.
 I'm charging here on EVSE2, and telnetting EVSE1. I see peaks (fun fact, once starting telnetting (9:30) i got less peaks).
[image]

caught the peak at 9:38

[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=17.0 A Imeasured=16.0 A MaxCircuit=18 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1710275835,processed=1710275833.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=18.0 A, Idifference=4.0, mod=0.
[0m[0;32m(V) (CalcBalancedCur entC)a(lCc1B)a lCahneccekdpCouirnrte 4 Isetbalanced=18.0 A.
[0m[0;32m(V) tbalanced=17.0 A Imeasured=16.0 A MaxCircuit=18 Imeasured_EV=0.0 A, Battery 7.0 A.
C[0mrent = 0.0 A, mode=1.e
d[0m[1;32m(D) Current)(C1) Balance: (CalcBalancVeSdEC0u:rBr1e(n9t.)0A)  phaseLastUpdate=1710E75VSE1:C(17.0A) 1710275835.
[0m[1;32m(D) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(C0a.l0cAB)a lancedCurrent)V(SCE05): AC(h0e.c0kAp)o int 2 IsetbalaVnScEe6d:=A1(70..00 AA), IdiEfVeSrEe7n:cAe(=0.0A) od
[0m0.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=17.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=17.0 A.
[0m[1;32m(D) ltipleRequest)(C1) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 11, values: 005a 00aa 0000 0000 0000 0000 0000 0000 0014 00a[0m0 0002
[0m[1;33m(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 16.0 A IsetBalanced: 17.0 A
[0m[1;33m(I) (printStatus)(C1) L1: 2.0 A L2: 16.0 A L3: 0.2 A Isum: 18.2 A
[0m[1;33m(I) CalcBalancedCurrent)(C0) Balance: EVSE0:B1(9.0A) EVSE1:C(17.0A) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(0.0A) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m(ModbusWriteMultipleRequest)(C0) Error outstanding request address: 0x09, function: 0x10. Overwriting request with reg: 0x0020, count: 11.
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 11, values: 005a 00aa 0000 0000 0000 0000 0000[0m 0000 0014 00a0 0002
[0m[1;33m(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 16.0 A IsetBalanced: 17.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 2.0 A L2: 16.0 A L3: 0.2 A Isum: 18.2 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x65, function: 0x04, reg: 0x0048, token:0x65040048, data: 0x0002.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (7 bytes) 65 04 04 45 21 3f 2f
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x65 Function 0x04 Register 0x0048 Response
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x65, function: 0x04, reg: 0x0034, token:0x65040034, data: 0x0002.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (7 bytes) 65 04 04 00 00 00 00
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x65 Function 0x04 Register 0x0034 Response
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x02, function: 0x04, reg: 0x0000, token:0x02040000, data: 0x0008.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (19 bytes) 02 04 10 00 02 00 00 00 aa 00 01 00 00 00 01 00 00 00 11
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x04 Register 0x0000 Response
[0m[1;32m(D) (receiveNodeStatus)(C0) ReceivedNode[1]Status State:2 Error:0, BalancedMax:170, Mode:1, ConfigChanged:0.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x03, function: 0x04, reg: 0x0000, token:0x03040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 03, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x04, function: 0x04, reg: 0x0000, token:0x04040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 04, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x05, function: 0x04, reg: 0x0000, token:0x05040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 05, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x06, function: 0x04, reg: 0x0000, token:0x06040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 06, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x07, function: 0x04, reg: 0x0000, token:0x07040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 07, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x08, function: 0x04, reg: 0x0000, token:0x08040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 08, function: 04, reg: 0000.
[0m[1;32m(D) (IsCurrentAvailable)(C0) Current available checkpoint D. ActiveEVSE increased by one=2, TotalCurrent=17.0A, StartCurrent=6A, Isum=18.2A, ImportCurrent=0A.
[0m[1;32m(D) (processAllNodeStates)(C0) NodeAdr 1, BalancedError:0
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x02, function: 0x10, reg: 0x0000, token: 0x02100000 count: 4, values: 0002 0000 0000 0001
[... cut ...]
[0m[1;32m(D) (CalcBalancedCurrent)(C0) Balance: EVSE0:B1(9.0A) EVSE1:C(17.0A) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(0.0A) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 11, values: 005a 00aa 0000 0000 0000 0000 0000[0m 0000 0014 00a0 0002
[0m[1;33m(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 16.0 A IsetBalanced: 17.0 A

caught the peak at 9:38

djoenez, 2024-03-12:
And another one, when only charging EVSE1:

[0m[1;32m(D) (CalcBalancedCurrent)(C0) Balance: EVSE0:C(18.0A) EVSE1:B1(9.0A) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(0.0A) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 11, values: 00b4 005a 0000 0000 0000 0000 0000[0m 0000 0011 00ac 0002
[0m[1;33m(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 17.2 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 1.7 A L2: 17.2 A L3: 0.2 A Isum: 19.1 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x65, function: 0x04, reg: 0x0048, token:0x65040048, data: 0x0002.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (7 bytes) 65 04 04 45 21 44 5e
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x65 Function 0x04 Register 0x0048 Response
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x65, function: 0x04, reg: 0x0034, token:0x65040034, data: 0x0002.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (7 bytes) 65 04 04 45 73 ef ee
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x65 Function 0x04 Register 0x0034 Response
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x02, function: 0x04, reg: 0x0000, token:0x02040000, data: 0x0008.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (19 bytes) 02 04 10 00 09 00 00 00 5a 00 01 00 00 00 00 00 00 00 11
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x04 Register 0x0000 Response
[0m[1;32m(D) (receiveNodeStatus)(C0) ReceivedNode[1]Status State:9 Error:0, BalancedMax:170, Mode:1, ConfigChanged:0.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x03, function: 0x04, reg: 0x0000, token:0x03040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 03, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x04, function: 0x04, reg: 0x0000, token:0x04040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 04, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x05, function: 0x04, reg: 0x0000, token:0x05040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 05, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x06, function: 0x04, reg: 0x0000, token:0x06040000, data: 0x0008.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x07, function: 0x04, reg: 0x0000, token:0x07040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 06, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x08, function: 0x04, reg: 0x0000, token:0x08040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 07, function: 04, reg: 0000.
[0m[1;32m(D) (IsCurrentAvailable)(C0) Current available checkpoint D. ActiveEVSE increased by one=2, TotalCurrent=18.0A, StartCurrent=6A, Isum=19.1A, ImportCurrent=0A.
[0m[1;32m(D) (processAllNodeStates)(C0) NodeAdr 1, BalancedError:0
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x02, function: 0x10, reg: 0x0000, token: 0x02100000 count: 4, values: 0009 0000 0000 0001
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 08, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (6 bytes) 02 10 00 00 00 04
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x10 Register 0x0000 Response
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 20: Request EVMeter Current Measurement Node 0
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x65, function: 0x04, reg: 0x0006, token:0x65040006, data: 0x000c.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (27 bytes) 65 04 18 41 89 4b 0c 00 00 00 00 00 00 00 00 45 73 bd 7c 00 00 00 00 00 00 00 00
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x65 Function 0x04 Register 0x0006 Response
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=18.0 A Imeasured=17.2 A MaxCircuit=18 Imeasured_EV=17.1 A, Battery Current = 0.0 A, mode=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1710276411,processed=1710276411.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=18.0 A, Idifference=0.9, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=18.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=18.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C0) Balance: EVSE0:C(18.0A) EVSE1:B1(9.0A) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(0.0A) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 11, values: 00b4 005a 0000 0000 0000 0000 0000[0m 0000 0011 00ac 0002
[0m[1;33m(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 17.2 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 1.7 A L2: 17.2 A L3: 0.2 A Isum: 19.1 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[... cut ...]
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 11, values: 00b4 005a 0000 0000 0000 0000 0000[0m 0000 0012 00ab 0002
[0m[1;33m(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 17.1 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 1.8 A L2: 17.1 A L3: 0.2 A Isum: 19.1 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.

dingo35, 2024-03-12:
Yes this looks _much_ better, no collisions / timeouts / crc errors on your modbus any more, except for the timeouts on addresses 0x03 - 0x08, which is normal because you only have one slave at 0x02.... 

Still doesn't explain those peaks, but we've taken a major step!

Thanks for testing!

dingo35, 2024-03-13:
@djoenez, I have got another one for you, my fellow developer discovered that if we pin the modbus routines to core 1, it is much less sensitive to load from the HTTP server.

Would you be so kind to test this one?
[48c0b8f5.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14587921/48c0b8f5.zip)

djoenez, 2024-03-13:
I think this is very close. saw one peak on EVSE1 (log below) while both EVSEs were charging. No peaks when EVSE2 was charging and EVSE1 was off (for 15 minutes).

[0m[1;32m(D) (IsCurrentAvailable)(C0) Current available checkpoint D. ActiveEVSE increased by one=2, TotalCurrent=17.0A, StartCurrent=6A, Isum=19.4A, ImportCurrent=0A.
[0m[1;32m(D) (processAllNodeStates)(C0) NodeAdr 1, BalancedError:0
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x02, function: 0x10, reg: 0x0000, token: 0x02100000 count: 4, values: 0002 0000 0000 0001
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (6 bytes) 02 10 00 00 00 04
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x10 Register 0x0000 Response
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 20: Request EVMeter Current Measurement Node 1
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x66, function: 0x04, reg: 0x0006, token:0x66040006, data: 0x000c.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=17.0 A Imeasured=16.0 A MaxCircuit=18 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1710354174,processed=1710354171.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=18.0 A, Idifference=4.0, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=18.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=17.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C1) Balance: EVSE0:B1(0.0A) EVSE1:C(17.0A) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(0.0A) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (27 bytes) 66 04 18 41 7d 9a 11 00 00 00(M o0d0busWrit MultipleR e0q0uest)( C010) Sent  p0a0cket 45 61 a7 c4 00 00 00 00addre[0m[0;32m(V)  0 [0;32m(V)  0x0020, token: 0x09100020 count: 11, valuessD:ecode)(C0)  valid Modbu0000cket: A d0d0raeass 0x6 60 0F0u0nction  00x00040 0000 R e0g0i0s0ter 0x 00000000 R e0s0p0[0[0m [0;32m(V)  0x0020, token: 0x09100020 count: 11, valuessD:ecode)(C0)  valid Modbu0000cket: A d0d0raeass 0x6 60 0F0u0nction  00x00040 0000 R e0g0i0s0ter 0x 00000000 R e0s0p0[0[0mo0nse
[0m[0;32m(V)  0021 00a0 0002
[0m[1;33m(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 16.0 A IsetBalanced: 17.0 A
[0m[1;33m(I) (printStatus)(C1) L1: 3.3 A L2: 16.0 A L3: 0.2 A Isum: 19.5 A
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=17.0 A Imeasured=16.0 A MaxCircuit=18 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1710354174,processed=1710354174.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=17.0 A, Idifference=4.0, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=17.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=17.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C0) Balance: EVSE0:B1(0.0A) EVSE1:C(17.0A) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(0.0A) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m(ModbusWriteMultipleRequest)(C0) Error outstanding request address: 0x09, function: 0x10. Overwriting request with reg: 0x0020, count: 11.
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 11, values: 0000 00aa 0000 0000 0000 0000 0000[0m 0000 0021 00a0 0002
[0m[1;33m(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 16.0 A IsetBalanced: 17.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 3.3 A L2: 16.0 A L3: 0.2 A Isum: 19.5 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0

djoenez, 2024-03-13:
Unfortunately more peaks downwards on only EVSE1 charging. here is the log for 2 peaks downwards (just after 7:43) to 2590 W (from 3840 W).

[image]

[0m[1;32m(D) (CalcBalancedCurrent)(C0) Balance: EVSE0:C(18.0A) EVSE1:B1(6.0A) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(0.0A) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 11, values: 00b4 003c 0000 0000 0000 0000 0000[0m 0000 0020 00ab 0002
[0m[1;33m(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 17.1 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 3.2 A L2: 17.1 A L3: 0.2 A Isum: 20.5 A
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x65, function: 0x04, reg: 0x0048, token:0x65040048, data: 0x0002.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (7 bytes) 65 04 04 45 21 e7 60
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x65 Function 0x04 Register 0x0048 Response
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x65, function: 0x04, reg: 0x0034, token:0x65040034, data: 0x0002.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (7 bytes) 65 04 04 45 70 5d a6
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x65 Function 0x04 Register 0x0034 Response
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x02, function: 0x04, reg: 0x0000, token:0x02040000, data: 0x0008.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (19 bytes) 02 04 10 00 09 00 00 00 3c 00 01 00 00 00 00 00 00 00 11
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x04 Register 0x0000 Response
[0m[1;32m(D) (receiveNodeStatus)(C0) ReceivedNode[1]Status State:9 Error:0, BalancedMax:170, Mode:1, ConfigChanged:0.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x03, function: 0x04, reg: 0x0000, token:0x03040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 03, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x04, function: 0x04, reg: 0x0000, token:0x04040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 04, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x05, function: 0x04, reg: 0x0000, token:0x05040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 05, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x06, function: 0x04, reg: 0x0000, token:0x06040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 06, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x07, function: 0x04, reg: 0x0000, token:0x07040000, data: 0x0008.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=18.0 A Imeasured=17.1 A MaxCircuit=18 Imeasured_EV=17.0 A, Battery Current = 0.0 A, mode=1.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1710355431,processed=1710355428.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=18.2 A, Idifference=1.0, mod=0.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=18.0 A.
[0m[0;32m(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=18.0 A.
[0m[1;32m(D) (CalcBalancedCurrent)(C1) Balance: EVSE0:C(18.0A) EVSE1:B1(6.0A) EVSE2:A(0.0A) EVSE3:A(0.0A) EVSE4:A(0.0A) EVSE5:A(0.0A) EVSE6:A(0.0A) EVSE7:A(0.0A) 
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C1) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 11, values: 00b4 003c 0000 0000 0000 0000 0000[0m 0000 0020 00ab 0002
[0m[1;33m(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 17.1 A IsetBalanced: 18.0 A
[0m[1;33m(I) (printStatus)(C1) L1: 3.2 A L2: 17.1 A L3: 0.2 A Isum: 20.5 A
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 07, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x08, function: 0x04, reg: 0x0000, token:0x08040000, data: 0x0008.
[0m(MBhandleError)(C0) Error response: E0 - Timeout, address: 09, function: 10, reg: 0020.
[0m[1;32m(D) (IsCurrentAvailable)(C0) Current available checkpoint D. ActiveEVSE increased by one=2, TotalCurrent=18.0A, StartCurrent=6A, Isum=20.5A, ImportCurrent=0A.
[0m[1;32m(D) (processAllNodeStates)(C0) NodeAdr 1, BalancedError:0
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C0) Sent packet address: 0x02, function: 0x10, reg: 0x0000, token: 0x02100000 count: 4, values: 0009 0000 0000 0001
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 08, function: 04, reg: 0000.
[0m[1;32m(D) (Timer100ms)(C0) ModbusRequest 20: Request EVMeter Current Measurement Node 0
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x65, function: 0x04, reg: 0x0006, token:0x65040006, data: 0x000c.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (6 bytes) 02 10 00 00 00 04
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x10 Register 0x0000 Response
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (27 bytes) 65 04 18 41 88 83 e5 00 00 00 00 00 00 00 00 45 70 6e 09 00 00 00 00 00 00 00 00
[... cut ...]
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x04, function: 0x04, reg: 0x0000, token:0x04040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 04, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x05, function: 0x04, reg: 0x0000, token:0x05040000, data: 0x0008.
[0m[0;32m(V) (MBhandleError)(C0) Error response: E0 - Timeout, address: 05, function: 04, reg: 0000.
[0m[1;32m(D) (ModbusSend8)(C0) Sent packet address: 0x06, function: 0x04, reg: 0x0000, token:0x06040000, data: 0x0008.
[0m

dingo35, 2024-03-14:
@djoenez Another improvement; I accidently pinned the slave modbus proces to core 1, but pinnen the master modbus proces should give better results:
[5373b78c5050.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14607674/5373b78c5050.zip)

djoenez, 2024-03-15:
Unfortunately still popping up (updated at 9:05).
[image]

dingo35, 2024-03-15:
Ok I can see in the config that you posted at the top of this thread that you use MQTT.
Do you also use the REST APi, and/or the HomeAssistant integration?
Could you also describe wich devices are on your modbus?

We want to have a picture of your infrastructure, to find out why you keep having these strange effects.

Also, do you live near strong transmitters?

[13 more replies]

closed 2024-03-25 by dingo35