dingo35/SmartEVSE-3.5 #12: Feature Request: ModbusTCP for Multi (master-slave) SmartEVSE setup and/or far away Sensorbox

fluppie, 2024-02-12:
At the moment you need to make a 4-wire physical connection between the Sensorbox (12V, GND and A & B). It would be nice for example to cover longer distances between electrical boards without the need for extra wiring when adequate WiFi is availale.
For example the Sensorbox2 has a ESP32, if we would make that data available over Modbus TCP, it can work just by plugging it in into a P1 meter (need to check if it can get 5V out of a DSMR5 meter). Saves wiring and labour hours.
Same goes for EVSE's that aren't in the same electrical board but are using the same grid connection with a certain capacity.
Would be nice if those could also connect over Modbus TCP instead of only Modbus RTU.

This is kind of supported by the library: https://emodbus.github.io/modbusbridge

Hopefully this can be considered. I think it can save some wiring/time when building larger setups.

dingo35, 2024-02-12:
Good idea, we already had the modbus-tcp-bridge enabled in v1.8.0 and earlier, but had to disable it because of instabilities. If we can get those out,its not a big deal...

fluppie, 2024-08-13:
I found this https://github.com/arpiecodes/esphome-smartevse-sensorbox , but have not tested. Might use the API and the sensorbox doing http posts to the SmartEVSE's? But don't know how that will work out with 2 or more SmartEVSE's.
Still like the Modbus TCP idea more in combi with request https://github.com/dingo35/SmartEVSE-3.5/issues/81