dingo35/SmartEVSE-3.5 #13: Feature Request: sync stored allowed RFID list from Master EVSE in a Multi setup to the slaves

fluppie, 2024-02-12:
For example you have a Multi SmartEVSE setup where there are 2 or more EVSE's. All EVSE's have their own RFID reader.
To avoid learning 20/40/... cards to every EVSE. It would be nice if we could only learn the RFID's to the master and then it will sync the complete RFID list over Modbus/HTTP/MQTT to the slaves.

Bonus feature: import/export a .txt or .csv file with all allowed RFID tags via the webinterface. This could be nice if you have EVSE's in multiple locations and the people allowed to charge are travelling between these locations.

dingo35, 2024-02-12:
Excellent proposal, forces me to think about a solution that is independent of the link layer modbus/http/mqtt.... its on the list!

fluppie, 2024-02-12:
OCPP, absolutely, would be cool if SmartEVSE was allowed to backoffices so you could put SmartEVSE's into a roaming network. But I doubt this is feasible without (costly?) certifications etc.

On the other hand being able to do it on-device, without some external management system would also make sense.

dingo35, 2024-02-12:
My first gut feeling goes for your "bonus feature", import from .csv or .txt file.

That way users still have the choice to have differing RFID lists per EVSE,  or syncing them by uploading the same file to all EVSE's.

fluppie, 2024-02-12:
Jup and then, NVS or Spiffs :-).

mstegen, 2024-02-25:
Just thinking out loud here, but if the slave would just send the RFID of the card over modbus to the Master, then the Master can decide if it's allowed to charge or not.

dingo35, 2024-02-25:
True, but:
1 losing flexibility to differ authorisations per EVSE (do we need that, or are they all one-rfid-card-should-authorize-all-EVSEs users?
2 losing compatibility with older SmartEVSEs due to protocol change

fluppie, 2024-03-01:
Regarding the RFID card learning. Would also be handy to put it in card learn mode without the need to go to the electrical board and then go into the menu. In a lot of cases there's quite some distance between the RFID reader and the modules. So would be handy to just switch on the learning mode on my phone in the webinterface or HA and learn new tags.

sending "3" to the RFID mqtt topic isn't working, but I guess that the information is one way and it doesn't read any values being sent to that topic. Right?

fluppie, 2024-03-13:
@dingo35 @mstegen I used setmode as an inspiration for setrfid
Seems like it's working. Is it acceptable to do it this way over http post?

```
        if(request->hasParam("setrfid")) {
            String setrfid = request->getParam("setrfid")->value();

            // RFID Reader (0:Disabled / 1:Enabled / 2:Enable One / 3:Learn / 4:Delete / 5:Delete All)
            switch(setrfid.toInt()) {
                case 0: // Disabled
                    RFIDReader = 0;
                    setAccess(1);
                    #if MQTT
                        // Update MQTT faster
                        lastMqttUpdate = 10;
                    #endif
                    break;
                case 1:
                    RFIDReader = 1;
                    break;
                case 2:
                    RFIDReader = 2;
                    break;
                case 3:
                    RFIDReader = 3;
                    break;
                case 4:
                    RFIDReader = 4;
                    break;
                case 5:
                    RFIDReader = 5;
                    DeleteAllRFID();
                    break;
                default:
                    setrfid = "Value not allowed!";
            }
            doc["evse"]["setrfid"] = RFIDReader;
        }
```

dingo35, 2024-03-13:
I would prefer the "bonus" option, uploading .csv or .txt file.

It gives much more flexibility, and doesnt expand our API with a seldomly used option; it also would have to be synced with the MQTT API.

dingo35, 2024-04-28:
Done with e6efc5130

closed 2024-04-28 by dingo35