dingo35/SmartEVSE-3.5 #17: EV goes to sleep while EVSE is making current available

dingo35, 2024-02-25:
Bug reported by "maarten_NL" through PB:  Contactor is on, EVSE says charging while car does not charge.

[0796d9f2b5.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14396994/0796d9f2b5.zip)

This version might solve your problem, please test!

dingo35, 2024-03-05:
New version with bug fixes, please test:
[827a3cf8b23.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14498108/827a3cf8b23.zip)

mintgroen, 2024-03-06:
> Bug reported by "maarten_NL" through PB: Contactor is on, EVSE says charging while car does not charge.
> 
> [0796d9f2b5.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14396994/0796d9f2b5.zip)
> 
> This version might solve your problem, please test!

With this version, this is the part after a Max mains was exceeded for a very short period of time. Then after one minute, the EVSE tried to re-initiate the charging session but after activating the contactor, screen went blank, contactor switched off,

(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -8 ChargeDelay: 3 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 6.5 A IsetBalanced: 17.0 A
(I) (printStatus)(C0) L1: 4.4 A L2: 6.5 A L3: 5.0 A Isum: 15.9 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -8 ChargeDelay: 2 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 6.5 A IsetBalanced: 17.0 A
(I) (printStatus)(C1) L1: 4.4 A L2: 6.5 A L3: 5.1 A Isum: 16.0 A
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -8 ChargeDelay: 1 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 6.5 A IsetBalanced: 17.0 A
(I) (printStatus)(C0) L1: 4.4 A L2: 6.5 A L3: 5.1 A Isum: 16.0 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.1 A IsetBalanced: 17.0 A
(I) (printStatus)(C1) L1: 4.4 A L2: 4.3 A L3: 5.1 A Isum: 13.8 A
(ProximityPin)(C0) PP pin: 1023 (1921 mV) (warning: fixed cable configured so PP probably disconnected, making this reading void)
(I) (EVSEStates)(C0) Cable limit: 17A  Max: 17A
(D) (IsCurrentAvailable)(C0) Current available checkpoint D. ActiveEVSE increased by one=1, TotalCurrent=0.0A, StartCurrent=8A, Isum=13.8A, ImportCurrent=0A.
(setState)(C0) 18:47:50 STATE B1 -> B
(EVSEStates)(C0) Diode OK
(I) (printStatus)(C1) STATE: B Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.1 A IsetBalanced: 17.0 A
(I) (printStatus)(C1) L1: 4.4 A L2: 4.2 A L3: 5.1 A Isum: 13.7 A
(I) (printStatus)(C0) STATE: B Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.1 A IsetBalanced: 17.0 A
(I) (printStatus)(C0) L1: 4.4 A L2: 4.2 A L3: 5.1 A Isum: 13.7 A
(I) (printStatus)(C1) STATE: B Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 17.0 A
(I) (printStatus)(C1) L1: 4.4 A L2: 4.2 A L3: 4.9 A Isum: 13.5 A
(D) (IsCurrentAvailable)(C0) Current available checkpoint D. ActiveEVSE increased by one=1, TotalCurrent=0.0A, StartCurrent=8A, Isum=13.5A, ImportCurrent=0A.
(setState)(C0) 18:47:52 STATE B -> C
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.8 A IsetBalanced: 17.0 A
(I) (printStatus)(C1) L1: 4.4 A L2: 4.2 A L3: 4.8 A Isum: 13.4 A
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.8 A IsetBalanced: 17.0 A
(I) (printStatus)(C0) L1: 4.4 A L2: 4.2 A L3: 4.8 A Isum: 13.4 A
(setState)(C0) 18:47:53 STATE C -> B
(EVSEStates)(C0) Diode OK
(I) (printStatus)(C1) STATE: B Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.7 A IsetBalanced: 17.0 A
(I) (printStatus)(C1) L1: 4.4 A L2: 4.2 A L3: 4.7 A Isum: 13.3 A
(I) (printStatus)(C1) STATE: B Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.7 A IsetBalanced: 17.0 A
(I) (printStatus)(C1) L1: 4.4 A L2: 4.2 A L3: 4.7 A Isum: 13.3 A
(I) (printStatus)(C0) STATE: B Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.7 A IsetBalanced: 17.0 A
(I) (printStatus)(C0) L1: 4.4 A L2: 4.2 A L3: 4.7 A Isum: 13.3 A
(I) (printStatus)(C1) STATE: B Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.7 A IsetBalanced: 17.0 A
(I) (printStatus)(C1) L1: 4.4 A L2: 4.1 A L3: 4.7 A Isum: 13.2 A
(I) (printStatus)(C1) STATE: B Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.6 A IsetBalanced: 17.0 A
(I) (printStatus)(C1) L1: 4.3 A L2: 4.0 A L3: 4.6 A Isum: 12.9 A
(I) (printStatus)(C0) STATE: B Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.6 A IsetBalanced: 17.0 A
(I) (printStatus)(C0) L1: 4.3 A L2: 4.0 A L3: 4.6 A Isum: 12.9 A

dingo35, 2024-03-06:
Ok that's funny I just fixed that bug, see #24 , bottom of the discussion the 1eeb20ba03+1.zip should fix it...

mintgroen, 2024-03-06:
Flashed that version, triggered the symptom by exceeding max mains and the same behavior occurs.

Kona stops the charging, after 60 seconds the EVSE tries to restart the charing, state C but after the contactor close, it reopens again. see line 619 and further of the telnet log:

[log.txt](https://github.com/dingo35/SmartEVSE-3.5/files/14514581/log.txt)

dingo35, 2024-03-06:
I can't see anything strange in those logs; I understand v3.0.1 worked, the easiest way to find this problem is to move through history and narrow it down to the commit that introduced the problem.
So first flash v1.8.0 , if it works then, goto v3.5.0-RC0, but if it doesn't, go back to v1.6.0 or even v1.4.4 ; that's the nice thing about this repo, you can go all the way back to the first commit, which is Stegen's v3.0.1 .

The binaries of v1.8.0 and earlier you can pull off the serkri repo, but it is pretty simple to compile yourself:
https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/configuration.md   see bottom of page.

If you can find the disturbing commit I'm happy to work around it. The alternative is that you park your Kona at my house for a few weeks :-) .

dingo35, 2024-03-25:
@mintgroen Could you please test if this version still has problems with the Kona?

[de779d4d4.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14742832/de779d4d4.zip)

mintgroen, 2024-03-25:
Sure will do, the sun was shining so the Kona is full atm. I'll test asap.

mintgroen, 2024-03-29:
Did some testing and the Kona seems to listen to the EVSE and restarts charging when the power availability returns.

closed 2024-03-29 by dingo35