dingo35/SmartEVSE-3.5 #20: Solar mode unstable charging

djoenez, 2024-02-27:
**Describe the bug**
Since today it is the first day i can charge my car directly from the sun again i saw the following:

PV mode bouncing up and down ( with earlier serkri versions it was stable as a rock (+-10W))
[image]
from the beginning i'm sending current data (API, via HA) every second (as fast as my smart meter can give them) 11:54 i tried to lower the refresh rate to 3s, which helped in first instance. but later it will start bouncing again. (This is on my slave EVSE btw)

Did anything change on the SOLAR PID?

I checked other large appliances within my house but they are not on.

**Upload your config**
"version":"v3.5.0-RC1","mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Oud-Zuid intelligence #1","rssi":-40,"bssid":"84:D8:1B:E7:B6:FE"},"evse":{"temp":24,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":2,"pwm":184,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":108,"override_current":0,"current_min":6,"current_max":17,"current_main":20,"current_max_circuit":18,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":10,"enable_C2":"Not present","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.1.80","port":1883,"topic_prefix":"EVSE2","username":"mosquitto","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron1P","address":102,"import_active_power":2.1,"total_kwh":340.2,"charged_kwh":5.7,"currents":{"TOTAL":96,"L1":96,"L2":0,"L3":0},"import_active_energy":340.2,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":-15,"L1":6,"L2":76,"L3":-97,"last_data_update":0,"original_data":{"TOTAL":0,"L1":0,"L2":0,"L3":0}},"backlight":{"timer":120,"status":"ON"}}

**Upload a debug log**
can do tonight (but there is no sun)

**To Reproduce**
Solar

**Expected behavior**
with earlier serkri version

**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.

dingo35, 2024-02-27:
Is it possible this behaviour is caused by your modbus problems?

wagemakersjpmaa, 2024-03-04:
I see the same behaviour, 
[image]
Using the API I post the values every 4 seconds directly from my smart meter (ESPHOME)
interval:
  - interval: 4sec
    then:
      - http_request.post:
          url: !lambda |-
            std::string url;
            float current_l1 = (id(power_delivered_l1).state - id(returned_l1).state) / id(voltage_l1).state * 1000;
            float current_l2 = (id(power_delivered_l2).state - id(returned_l2).state) / id(voltage_l2).state * 1000;
            float current_l3 = (id(power_delivered_l3).state - id(returned_l3).state) / id(voltage_l3).state * 1000;

            url.append("http://192.168.4.138/currents?L1=");
            url.append(to_string(current_l1*10));
            url.append("&L2=");
            url.append(to_string(current_l2*10));
            url.append("&L3=");
            url.append(to_string(current_l3*10));
            return url.c_str();

RAW data from Smart-EVSE:
{"version":"v3.5.0-RC1","mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"IOT","rssi":-57,"bssid":"18:E8:29:CD:F3:3B"},"evse":{"temp":31,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":0,"pwm":144,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":85,"override_current":0,"current_min":6,"current_max":25,"current_main":40,"current_max_circuit":25,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":7,"solar_stop_time":30,"enable_C2":"Not present","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.1.201","port":1883,"topic_prefix":"SmartEVSE-6766","username":"mqtt","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":14,"import_active_power":5.1,"total_kwh":130.4,"charged_kwh":8.5,"currents":{"TOTAL":216,"L1":73,"L2":71,"L3":72},"import_active_energy":130.4,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":-86,"L1":-37,"L2":-12,"L3":-37,"last_data_update":1709559762,"original_data":{"TOTAL":-86,"L1":-37,"L2":-12,"L3":-37}},"backlight":{"timer":120,"status":"ON"}}

wagemakersjpmaa, 2024-03-06:
Thanks @dingo35 I will check this version and do some testing in Solar mode. Let you know the results.

djoenez, 2024-03-06:
On EVSE1 it already looks good; If my wife is home early i'll check EVSE2.
[image]

wagemakersjpmaa, 2024-03-06:
@dingo35 I did some testing with this interim built, this looks much more stable in handeling the current changes.. Back to the RC1 built since this builds has some other issues. my MQTT is not working and also the EV meter is not working in this interim build.
[image]

dingo35, 2024-03-06:
Ok that is good news; the MQTT is because we have a breaking change, the MQTT topic number changed from the strange MAC derived number to the serial number now; so you have to change the MQTT topic in HomeAssistant .

The EVMeter should still show power and energy, but no longer displays the phase currents on the LCD display; it was clogging the picture.
Does it still show EV power and energy?

wagemakersjpmaa, 2024-03-06:
Ok, understood on the MQTT. The EV meter did not work no data on the display and also no data on the dashboard, for power, and for KWh.

mintgroen, 2024-03-06:
I see the same behavior on SOLAR charging, while the Solar is stable, the EVSE and the Hyundai Kona are doing a up and down session together. 

Solar power has an update rate of 10s, P1 on a proper DSMR5 every second over MQTT.

Yellow is solar, green dots is P1:

[image]

dingo35, 2024-03-06:
I'm sorry but posting graphs without Y-axis makes them pretty meaningless.... 
I think you have a rather complicated setup, with solar & battery, probably all fed by API ; the engine of SmartEVSE works with modbus, which gets updates every 1-2s; so my old System Theory professor taught me the unexpected behaviour of these kinds of control loops, when you start playing with the variables , of which lag-time is not an unimportant one :-).

It might help if you can add parts of the log where the dip goes in, and point out at which parts of the log the solar power info comes in, at which part the P1 comes in... If you know that, you can read in the logs which calculations are made in CalcBalancedCurrent, and WHY it is making those calculations.... I'm happy to change those calculations if you have an improvement....

djoenez, 2024-03-08:
With the newest version i get this on EVSE2:
[image]
sometimes its stable, sometimes it heavily fluctuates, doesn't matter if i update my L currents every 1/3/5/10 seconds.
EVSE1 is stable with updating L currents every 3 seconds (from 9:56):
[image]

dingo35, 2024-03-08:
Is this Solar mode?

djoenez, 2024-03-08:
Yes on both EVSE1 and EVSE2

dingo35, 2024-03-08:
Could you post a piece of a log where they are fluctuating that much?

djoenez, 2024-03-08:
[image]

and the log of EVSE2, starting around 2:06:
[0m(loop)(C1) DINGO: MQTT prefix=EVSE2.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (15 bytes) 02 10 00 00 00 04 08 00 02 00 00 00 76 00 02
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x10 Register 0x0000 Request
[0m[1;33m(I) (setState)(C0) Switching CONTACTOR C2 ON.
[0m[0;32m(V) (setItemValue)(C0) Errors Cleared received!
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (6 bytes) 66 04 00 06 00 0c
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x66 Function 0x04 Register 0x0006 Request
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (27 bytes) 66 04 18 41 42 4e 5c 00 00 00 00 00 00 00 00 45 28 a9 a5 00 00 00 00 00 00 00 00
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x66 Function 0x04 Register 0x0006 Response
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (63 bytes) 09 10 00 20 00 1c 38 00 3e 00 75 00 00 00 00 00 00 00 00 00 00 00 00 00 0b 00 00 00 73 00 00 ff 8a ff ff eb f3 3d [0m2c 4b 78 3f fc 00 08 00 00 00 00 00 00 00 01 00 00 00 00 00 00 5a 00 3f fb
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x09 Function 0x10 Register 0x0020 Request
[0m[0;32m(V) (MBbroadcast)(C0) Broadcast received, Node 11.7 A, MainsMeter Irms L1=1.1A,L2=11.5A,L3=-11.8A,
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (63 bytes) 09 10 00 20 00 1c 38 00 3e 00 76 00 00 00 00 00 00 00 00 00 00 00 00 00 0b 00 00 00 65 00 00 ff 8b ff ff eb f3 3d [0m2c 4b 78 3f fc 00 08 00 00 00 00 00 00 00 01 00 00 5a 7c 80 19 20 90 3f fb
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x09 Function 0x10 Register 0x0020 Request
[0m[0;32m(V) (MBbroadcast)(C0) Broadcast received, Node 11.8 A, MainsMeter Irms L1=1.1A,L2=10.1A,L3=-11.7A,
[0m[1;33m(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
[0m[1;33m(I) (printStatus)(C1) L1: 1.1 A L2: 10.1 A L3: -11.7 A Isum: -0.5 A
[0m(loop)(C1) DINGO: MQTT prefix=EVSE2.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (6 bytes) 02 04 00 00 00 08
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x04 Register 0x0000 Request
[0m[1;32m(D) (MBNodeRequest)(C0) Node answering NodeStatus request value[0]=2 value[1]=0 value[2]=118 value[3]=2 value[4]=0 value[5]=1 value[6]=0 value[7]=17
[0m[1;33m(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
[0m[1;33m(I) (printStatus)(C1) L1: 1.1 A L2: 10.1 A L3: -11.7 A Isum: -0.5 A
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (63 bytes) 09 10 00 20 00 1c 38 00 3e 00 77 00 00 00 00 00 00 00 00 00 00 00 00 00 0b 00 00 00 65 00 00 ff 8b ff ff eb f3 3d [0m2c 4b 78 3f fc 00 08 00 00 00 00 00 00 00 01 00 00 5a 7c 80 19 20 90 3f fb
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x09 Function 0x10 Register 0x0020 Request
[0m[0;32m(V) (MBbroadcast)(C0) Broadcast received, Node 11.9 A, MainsMeter Irms L1=1.1A,L2=10.1A,L3=-11.7A,
[0m(loop)(C1) DINGO: MQTT prefix=EVSE2.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (15 bytes) 02 10 00 00 00 04 08 00 02 00 00 00 77 00 02
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x10 Register 0x0000 Request
[0m[1;33m(I) (setState)(C0) Switching CONTACTOR C2 ON.
[0m[0;32m(V) (setItemValue)(C0) Errors Cleared received!
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (63 bytes) 09 10 00 20 00 1c 38 00 3e 00 78 00 00 00 00 00 00 00 00 00 00 00 00 00 0b 00 00 00 65 00 00 ff 8b ff ff eb f3 3d [0m2c 4b 78 3f fc 00 08 00 00 00 00 00 00 00 01 00 00 00 00 00 00 5a 00 3f fb
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x09 Function 0x10 Register 0x0020 Request
[0m[0;32m(V) (MBbroadcast)(C0) Broadcast received, Node 12.0 A, MainsMeter Irms L1=1.1A,L2=10.1A,L3=-11.7A,
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (63 bytes) 09 10 00 20 00 1c 38 00 3e 00 79 00 00 00 00 00 00 00 00 00 00 00 00 00 0b 00 00 00 64 00 00 ff 8b ff ff eb f3 3d [0m2c 4b 78 3f fc 00 08 00 00 00 00 00 00 00 01 00 00 5a 7c 80 19 20 90 3f fb
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x09 Function 0x10 Register 0x0020 Request
[0m[0;32m(V) (MBbroadcast)(C0) Broadcast received, Node 12.1 A, MainsMeter Irms L1=1.1A,L2=10.0A,L3=-11.7A,
[0m[1;33m(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
[0m[1;33m(I) (printStatus)(C1) L1: 1.1 A L2: 10.0 A L3: -11.7 A Isum: -0.6 A
[0m(loop)(C1) DINGO: MQTT prefix=EVSE2.
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (6 bytes) 66 04 00 48 00 02
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x66 Function 0x04 Register 0x0048 Request
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (7 bytes) 66 04 04 43 d4 f3 f8
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x66 Function 0x04 Register 0x0048 Response
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (6 bytes) 66 04 00 0c 00 02
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x66 Function 0x04 Register 0x000c Request
[0m[1;32m(D) (ModbusDecode)(C0) Received packet (7 bytes) 66 04 04 45 28 91 18
[0m[0;32m(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x66 Function 0x04 Register 0x000c Response
[... cut ...]
[0m[1;32m(D) (ModbusWriteMultipleRequest)(C1) Sent packet address: 0x09, function: 0x10, reg: 0x0020, token: 0x09100020 count: 28, values: 003e 007a 0000 0000 0000 0000 0000[0m 0000 000a 0000 0069 0000 ff88 ffff ebf3 3d2c 4b78 3ffc 0008 0000 4b9e 3ffc 4b90 3ffc 5a7c 8019 2090 3ffb
[0m[1;33m(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 10.5 A IsetBalanced: 12.2 A
[0m[1;33m(I) (printStatus)(C1) L1: 1.0 A L2: 10.5 A L3: -12.0 A Isum: -0.5 A

dingo35, 2024-03-08:
Now you post the part of the log with the small fluctuations (from 2:06 on), why don't you post the real big fluctuations (from 2:09 on) ?!?

djoenez, 2024-03-08:
I didn't know if you wanted to have the log of EVSE1 or EVSE2( the one that is charging, EVSE1 is receiving the power data). Putty does not want me to log 2 telnets at the same time for some reason (even in different folders) So i have EVSE2 from 2:06 and EVSE1 from 2:11 (when i noticed it was not logging to file).

dingo35, 2024-03-25:
[de779d4d4.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14742803/de779d4d4.zip)

Please test this one, your spike problem should have been solved and this is probably connected to it.
If the problem still exists, please post the log of the Master for the moments the big spikes are happening.

dingo35, 2024-03-25:
...... AND please compare it to this one:
[6ce9c01c399cf6.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14743406/6ce9c01c399cf6.zip)
The latter one has the same provision that Smart mode has, that it only changes IsetBalanced when the data processed is actually new....

fluppie, 2024-03-25:
PV mode bouncing up and down ( with earlier serkri versions it was stable as a rock (+-10W))

I think the best resolution possible is 0.6A / 138W

> The PWM duty cycle of the 1 kHz CP signal indicates the maximum allowed mains current. According to the SAE it includes socket outlet, cable and vehicle inlet. In the US, the definition of the [ampacity](https://en.wikipedia.org/wiki/Ampacity) (ampere capacity, or current capacity) is split for continuous and short term operation.[[30]](https://en.wikipedia.org/wiki/SAE_J1772#cite_note-brusa-35) The SAE defines the ampacity value to be derived by a formula based on the 1 ms full cycle (of the 1 kHz signal) with the maximum continuous ampere rating being 0.6 A per 10 µs up to 850 µs (with the lowest (100 µs/10 µs) × 0.6 A = 6 A). Above 850 µs, the formula requires subtraction of 640 µs and multiplying the difference by 2.5. For example ((960 µs − 640 µs)/10 µs) × 2.5 A = 80 A.[[29]](https://en.wikipedia.org/wiki/SAE_J1772#cite_note-finalsaej1772-34)``

Imaginous, 2024-03-25:
I'm solar charging with the latest version and it seems solid here.

No strange effects.

I supply P1 data via the API.

dingo35, 2024-03-25:
@fluppie @Imaginous which versions are you referring too, the RC1, the de77* or the 6ce9 version?

Imaginous, 2024-03-25:
I'm using the 6ce9 version.

fluppie, 2024-03-25:
I was not referring to a specific version. Just saying we can only increment charging current by 0.6 A so +-10W is a little bit optimistic ;-).

```

// Set Charge Current 
// Current in Amps * 10 (160 = 16A)
void SetCurrent(uint16_t current) {
    uint32_t DutyCycle;

    if ((current >= 60) && (current <= 510)) DutyCycle = current / 0.6;
                                                                            // calculate DutyCycle from current
    else if ((current > 510) && (current <= 800)) DutyCycle = (current / 2.5) + 640;
    else DutyCycle = 100;                                                   // invalid, use 6A
    DutyCycle = DutyCycle * 1024 / 1000;                                    // conversion to 1024 = 100%
    SetCPDuty(DutyCycle);
}

// Write duty cycle to pin
// Value in range 0 (0% duty) to 1024 (100% duty)
void SetCPDuty(uint32_t DutyCycle){
    ledcWrite(CP_CHANNEL, DutyCycle);                                       // update PWM signal
    CurrentPWM = DutyCycle;
}
```

djoenez, 2024-03-25:
I will report back with the d4 version first. Will take some time. It is a busy week with not alot of working from home.

djoenez, 2024-03-29:
I don't see much difference between the two version. I still see that my EVSE2 still has some bouncing around, but less than above stated. I think the EVSE2 has more delay since i'm only posting currents to my EVSE1. Would it help if i post them via MQTT also to EVSE2?

mintgroen, 2024-03-29:
Tested de779d4d4.zip:

Reponse to changing solar production and other loads is slower but less overshoot and more stable behavoir:

[image]

On Solar mode, it works much better. I'll test the Smart mode with some intermittent loads when the sun has set.

dingo35, 2024-03-29:
@djoenez , that wouldnt change anything, the Master calculates all the currents for all the Slaves.
The currents are now processed only every 2 seconds (modbus limitation), so dont post more than once a second because you will only be overloading the ESP32 without any change in the charging characteristics...

djoenez, 2024-04-04:
Hi. One thing I notice in both versions that my current session kwh number resets to zero during charging. If I look to the mode or errors there is nothing (no change). But while in sun or solar mode this happens (and I haven't seen it in previous versions).
[image]

dingo35, 2024-04-04:
What is the State during such a "reset" of the Energy/session counter? Is it constantly State C (=charging) or did it go through a number of state changes? And if so, why (lack of sun?) ?

djoenez, 2024-04-05:
The above image was in smart mode. There was no reason to stop charging when I look at my house consumption (no large appliances were on at that time). No change in the state and mode, at the time of reset.

dingo35, 2024-04-05:
Ok thx this is reported also in #45 so I will close this as a duplicate.

closed 2024-04-05 by dingo35