dingo35/SmartEVSE-3.5 #24: screen blank and no power information after pressing middle knob

mreijnen, 2024-02-29:
An Eastron SDM 630 M kWh meter has now been connected via the Modbus and version 3.5.0_RC1 installed. Everything was working fine until this morning I wanted to see how much my wife had charged last night. No info in Home Assistant. The smartEVSE's screen was blank. Pressing the middle button brought the content back, but everything was set to 0. Charging worked fine, fortunately, but no more information on the screen and in HA. The sensor box also no longer produces red or green pulses. Removing power and waiting 1 minute and then reapplying power does not help either. On the smartEVSE web page I see small currents of 0.1-0.2 A. Not the 3x 16A that is currently being charged. 
Reloading 3.5.0_RC0 solves the problem. 

[settings.json](https://github.com/dingo35/SmartEVSE-3.5/files/14445225/settings.json)

Imaginous, 2024-03-03:
Just was in my car for a moment, it activated the charger for heating the car.

I looked at the SmartEVSE and it was also blank and lit.

Pressing the middle button brought it back.

Using RC1. 
[image]

dingo35, 2024-03-03:
I had this myself at my testbench a few times, but never found a way to reproduce this...

If you find a way to reproduce this, please let us know.....

Imaginous, 2024-03-03:
I also noticed that after pressing the middle button the display turns back on, but the data like mains current is not updated.

Also the backlight stays on.
The web interface still reacts normally.
After a reboot everything is normal.

It seems it occurs after a long period (4 days or so) keeping the car connected but not charging.

As soon as it starts charging (and stops) the display lits up and freezes.

I still have an idea it's the display that goes haywired.
I don't know if the "display mode" is configured once or on every update?

Do you know the type of the display?
Then I can lookup the datasheet and dive into it.

dingo35, 2024-03-03:
@mreijnen @Imaginous Could you guys test if this version has the same problem? 
[Uploading 2b134af930a.zip…]()

Imaginous, 2024-03-03:
Installed the update...

Keep you posted.

mreijnen, 2024-03-04:
Installed the update. Screen is visible while charging and I can see return amps from the solarpanels. I can also see the car is connected, but i can't see any power going to the car on the display and in the webinterface. The car says charging with 4kW and the kWh meter measures 3x 1,45kW to the car. 
Jedlix stopt the charging and screen is blank again. Charging just started again and screen is visible again, but still no data from kWh meter visible. Only data from sensorbox 2 of which I have the impression they are not updated. Fixed on one value.

mreijnen, 2024-03-04:
Put RC0 back and it is working fine now.

mreijnen, 2024-03-06:
I am sorry, but it is still not the fix. Modbus data are freezed on startup data and the screen is blank after charging stops.

dingo35, 2024-03-06:
Ok thx, cannot reproduce this so kind of in the dark here....

mreijnen, 2024-03-06:
3.5.0_RC0 still works fine here.

dingo35, 2024-03-06:
Ok there are dozens of commits between RC0 and RC1, a lot of them I can rule out but in 27 of the commits, the problem must have creeped in.

Are you ok in loading these different tests, I think in 4 or 5 firmware reflashes we can have it localized.... I start in the middle, so patch nr. 13, if that still has the problem, we go down to patch nr 6, if not we go up to patch nr 19, etc.

This would be the first test, the middle one of the 27 patches:
[c7c73f8ad2.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14510296/c7c73f8ad2.zip)

mreijnen, 2024-03-06:
OK, I have tested your third patch and is was a bullseye! All good things come in tree. It works now.

dingo35, 2024-03-06:
No you misunderstood, I just "went back in time" to see which of the changes caused the error; the fact that this patch (commit is a better word) is working means that the problem was introduced later on.
BUT we have no reduced our 27 suspect commits to 11.
Next shot, 
[9b719c922.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14510647/9b719c922.zip)
How about this one?

mreijnen, 2024-03-06:
Tested this version and this one has the fault.

dingo35, 2024-03-06:
Thanks for working with me, we now narrowed it down to 6 suspicious commits. 
[01b82ff5ce.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14511349/01b82ff5ce.zip)

How about this one?

mreijnen, 2024-03-06:
This one is also faulty

dingo35, 2024-03-06:
Ok then this one is probably ok:
[506874aad47.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14511803/506874aad47.zip)

mreijnen, 2024-03-06:
Indeed! This one is OK.

dingo35, 2024-03-06:
Ok then this is hopefully the last test: this is the current "master" version of the repo, including all bugfixes, with that one commit reverted:
[1eeb20ba03+1.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14512128/1eeb20ba03%2B1.zip)

So this one should work!

mreijnen, 2024-03-06:
This one works indeed, but now I don't get Amp info per fase on the car side of the screen, but only power and energy on the line.
It does show amps per fase  in the webinterface though.

mreijnen, 2024-03-06:
The screen representation of the amps is not an issue, but a choice. Probably an sensible one.

dingo35, 2024-03-06:
Yes those phase amps clogged up the display, so this is intented behaviour.

Thanks for helping us out!

closed 2024-03-06 by dingo35