dingo35/SmartEVSE-3.5 #26: Battery charging current is not taken into account anymore

bartv, 2024-03-04:
**Describe the bug**
There is more sun again so I use smart/solar mode + home battery interaction for charging my car. With last years SmartEVSE versions it would work nicely. HA reports the the battery current every 10 seconds.

These calls are still working but only sporadicaly for a few seconds I would see the battery panel apearing in the webui. When the battery starts discharging the EVSE will no longer adjust to make sure the car is not charged from the homebattery.

The following screenshot shows the battery SOC when charging the car:

[image]

It only stopped when I manually forced the battery to stop by increasing the power threshold.

**Upload your config**

```json
{
    "version": "SERKRI-3.5.0",
    "mode": "SMART",
    "mode_id": 3,
    "car_connected": true,
    "wifi": {
        "status": "WL_CONNECTED",
        "ssid": "building",
        "rssi": -63,
        "bssid": "82:8A:20:5A:BB:D0"
    },
    "evse": {
        "temp": 39,
        "temp_max": 65,
        "connected": true,
        "access": true,
        "mode": 1,
        "loadbl": 0,
        "pwm": 545,
        "solar_stop_timer": 0,
        "state": "Charging",
        "state_id": 2,
        "error": "None",
        "error_id": 0,
        "rfid": "Not Installed"
    },
    "settings": {
        "charge_current": 320,
        "override_current": 0,
        "current_min": 8,
        "current_max": 32,
        "current_main": 32,
        "current_max_circuit": 32,
        "current_max_sum_mains": 30,
        "solar_max_import": 8,
        "solar_start_current": 0,
        "solar_stop_time": 2,
        "enable_C2": "Always Off",
        "modem": "Not present",
        "mains_meter": "API",
        "starttime": 0,
        "stoptime": 0,
        "repeat": 0
    },
    "mqtt": {
        "host": "192.168.33.2",
        "port": 1883,
        "topic_prefix": "smartevse-2",
        "username": "",
        "password_set": false,
        "status": "Connected"
    },
    "home_battery": {
        "current": 0,
        "last_update": 0
    },
    "ev_meter": {
        "description": "Eastron3P",
        "address": 12,
        "import_active_power": 7,
        "total_kwh": 3048.3,
        "charged_kwh": 29.8,
        "currents": {
            "TOTAL": 298,
            "L1": 0,
            "L2": 0,
            "L3": 298
        },
        "import_active_energy": 3048.3,
        "export_active_energy": 0
    },
    "mains_meter": {
        "import_active_energy": 0,
        "export_active_energy": 0
    },
    "phase_currents": {
        "TOTAL": 216,
        "L1": -30,
        "L2": -30,
        "L3": 276,
        "last_data_update": 1709551807,
        "original_data": {
            "TOTAL": 216,
            "L1": -30,
            "L2": -30,
            "L3": 276
        }
    },
    "backlight": {
        "timer": 120,
        "status": "ON"
    }
}
```

**Upload a debug log**

Will add it later.

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar? Smart (potentially also solar)
2. What steps do you take so the problem arises? Change battery settings or let it discharge to its minimal value

**Expected behavior**
A clear and concise description of what you expected to happen; expected Charging Current?

The EVSE throttles the charging current so that the battery no longer discharges like it does for other consumption in the house.

**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.

bartv, 2024-03-04:
A screen shot from tcpdump to validate that the battery current calls happen:
[image]

bartv, 2024-03-04:
Based on the debug logs it looks like the current evaluation loop does not retain the battery current. When I send the current as fast as the api can handle most messages have the value but there are still some 0s in there:

```
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 22.6 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: -7.3 A L2: -7.4 A L3: 22.6 A Isum: 7.9 A
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0048, token:0x0c040048, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 3e 99 fc
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0048 Response
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0034, token:0x0c040034, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 dc 0d 23
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0034 Response
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0006, token:0x0c040006, data: 0x000c.
(D) (ModbusDecode)(C0) Received packet (27 bytes) 0c 04 18 00 00 00 00 00 00 00 00 41 ee f2 e9 00 00 00 00 00 00 00 00 45 dc 1a 2d
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=22.6 A MaxCircuit=32 Imeasured_EV=29.8 A, Battery Current = -11.4 A, mode=1.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1709552483,processed=1709552483.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=32.0 A, Idifference=2.2, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=32.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 22.6 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: -7.3 A L2: -7.4 A L3: 22.6 A Isum: 7.9 A
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0048, token:0x0c040048, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 3e 9a 0c
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0048 Response
x0002.
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0034, token:0x0c040034, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 dc 51 7f
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0034 Response
(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=32.0 A Imeasured=22.6 A MaxCircuit=32 Imeasured_EV=29.8 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1709552487,processed=1709552483.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=32.5 A, Idifference=2.2, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=32.0 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=32.0 A.
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 22.6 A IsetBalanced: 32.0 A
(I) (printStatus)(C1) L1: -7.3 A L2: -7.4 A L3: 22.6 A Isum: 7.9 A
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0006, token:0x0c040006, data: 0x000c.
(D) (ModbusDecode)(C0) Received packet (27 bytes) 0c 04 18 00 00 00 00 00 00 00 00 41 ee ea 42 00 00 00 00 00 00 00 00 45 dc 6a 83
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=22.6 A MaxCircuit=32 Imeasured_EV=29.8 A, Battery Current = -11.4 A, mode=1.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1709552487,processed=1709552487.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=32.0 A, Idifference=2.2, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=32.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 22.6 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: -7.3 A L2: -7.4 A L3: 22.6 A Isum: 7.9 A
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0048, token:0x0c040048, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 3e 9a 1d
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0048 Response
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0034, token:0x0c040034, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 dc 49 18
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0034 Response
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0006, token:0x0c040006, data: 0x000c.
(D) (ModbusDecode)(C0) Received packet (27 bytes) 0c 04 18 00 00 00 00 00 00 00 00 41 ee b4 1a 00 00 00 00 00 00 00 00 45 dc 31 f7
[... cut ...]
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=22.7 A MaxCircuit=32 Imeasured_EV=29.8 A, Battery Current = -11.4 A, mode=1.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1709552491,processed=1709552491.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=32.0 A, Idifference=2.2, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=32.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.
```

bartv, 2024-03-04:
Some more experimentation. In solar mode it works fine. When the battery is below a certain threshold I send a discharge current so the home battery and car can charge at the same time. This is what you see here in the logs and works fine:

```
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0048, token:0x0c040048, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 3e b9 3b
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0048 Response
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0034, token:0x0c040034, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 90 1b 31
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0034 Response
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0006, token:0x0c040006, data: 0x000c.
(D) (ModbusDecode)(C0) Received packet (27 bytes) 0c 04 18 00 00 00 00 00 00 00 00 41 9d 10 3f 00 00 00 00 00 00 00 00 45 90 15 7b
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=20.3 A Imeasured=16.0 A MaxCircuit=32 Imeasured_EV=19.6 A, Battery Current = -8.9 A, mode=2.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=20.3 A, Idifference=7.6, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 3 Isetbalanced=20.4 A, IsumImport=-0.9, Isum=7.1, ImportCurrent=8.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=20.4A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=20.4 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=20.4 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 16.0 A IsetBalanced: 20.4 A
(I) (printStatus)(C0) L1: -4.4 A L2: -4.5 A L3: 16.0 A Isum: 7.1 A
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0048, token:0x0c040048, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 3e b9 44
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0048 Response
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0034, token:0x0c040034, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 91 7e 9b
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0034 Response
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0006, token:0x0c040006, data: 0x000c.
(D) (ModbusDecode)(C0) Received packet (27 bytes) 0c 04 18 00 00 00 00 00 00 00 00 41 9e 85 58 00 00 00 00 00 00 00 00 45 91 7e 48
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=20.4 A Imeasured=16.0 A MaxCircuit=32 Imeasured_EV=19.8 A, Battery Current = -8.9 A, mode=2.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=20.4 A, Idifference=7.6, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 3 Isetbalanced=20.5 A, IsumImport=-0.9, Isum=7.1, ImportCurrent=8.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=20.5A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=20.5 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=20.5 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 16.0 A IsetBalanced: 20.5 A
(I) (printStatus)(C0) L1: -4.4 A L2: -4.5 A L3: 16.0 A Isum: 7.1 A
(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=20.5 A Imeasured=17.2 A MaxCircuit=32 Imeasured_EV=19.8 A, Battery Current = -8.9 A, mode=2.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=20.5 A, Idifference=6.8, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 3 Isetbalanced=20.0 A, IsumImport=1.4, Isum=9.4, ImportCurrent=8.
(D) (CalcBalancedCurrent)(C1) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=20.0A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=20.0 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=20.0 A.
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 17.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.9 A L2: -3.9 A L3: 17.2 A Isum: 9.4 A
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0048, token:0x0c040048, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 3e b9 50
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0048 Response
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0034, token:0x0c040034, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 8d d4 eb
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0034 Response
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0006, token:0x0c040006, data: 0x000c.
[... cut ...]
(V) (CalcBalancedCurrent)(C0) Checkpoint 3 Isetbalanced=15.0 A, IsumImport=1.5, Isum=9.5, ImportCurrent=8.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=15.0A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=15.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=15.0 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 15.5 A IsetBalanced: 15.0 A
(I) (printStatus)(C0) L1: -3.1 A L2: -2.9 A L3: 15.5 A Isum: 9.5 A
```

dingo35, 2024-03-04:
Ok, I think I see what's going on, users with long API mainsmeter current delivery feeding times were complaining that the loop was very "agressive", because it was regulating every 1-2 s but fresh data was only coming in every 10s.

We changed that so that in Smart mode, it now only regulates when receiving fresh currents. 

This version should also regulate on receipt of homebattery currents:
[14a03f5fc2.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14482295/14a03f5fc2.zip)

Please test!

bartv, 2024-03-04:
> Ok, I think I see what's going on, users with long API mainsmeter current delivery feeding times were complaining that the loop was very "agressive", because it was regulating every 1-2 s but fresh [...]

It is still ping-ponging between the charge current and 0. See logs:

```
(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=32.0 A Imeasured=26.2 A MaxCircuit=32 Imeasured_EV=31.2 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1709562683,processed=1709562681.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=32.2 A, Idifference=0.8, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=32.0 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=32.0 A.
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 26.2 A IsetBalanced: 32.0 A
(I) (printStatus)(C1) L1: -8.0 A L2: -8.4 A L3: 26.2 A Isum: 9.8 A
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0048, token:0x0c040048, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 3e d1 1f
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0048 Response
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0034, token:0x0c040034, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 e3 69 33
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0034 Response
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0006, token:0x0c040006, data: 0x000c.
(D) (ModbusDecode)(C0) Received packet (27 bytes) 0c 04 18 00 00 00 00 00 00 00 00 41 f9 75 e3 00 00 00 00 00 00 00 00 45 e3 84 7c
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=26.2 A MaxCircuit=32 Imeasured_EV=31.1 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1709562683,processed=1709562683.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=32.0 A, Idifference=0.9, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=32.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 26.2 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: -8.0 A L2: -8.4 A L3: 26.2 A Isum: 9.8 A
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0048, token:0x0c040048, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 3e d1 2f
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0048 Response
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0034, token:0x0c040034, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 e3 86 b3
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0034 Response
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0006, token:0x0c040006, data: 0x000c.
(D) (ModbusDecode)(C0) Received packet (27 bytes) 0c 04 18 00 00 00 00 00 00 00 00 41 f9 53 2d 00 00 00 00 00 00 00 00 45 e3 55 dc
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=26.2 A MaxCircuit=32 Imeasured_EV=31.1 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1709562683,processed=1709562683.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=32.0 A, Idifference=0.9, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=32.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 26.2 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: -8.0 A L2: -8.4 A L3: 26.2 A Isum: 9.8 A
(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=32.0 A Imeasured=26.1 A MaxCircuit=32 Imeasured_EV=31.1 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1709562687,processed=1709562683.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=32.2 A, Idifference=0.9, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=32.0 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=32.0 A.
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 26.1 A IsetBalanced: 32.0 A
(I) (printStatus)(C1) L1: -8.0 A L2: -8.4 A L3: 26.1 A Isum: 9.7 A
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0048, token:0x0c040048, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 3e d1 3f
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0048 Response
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0034, token:0x0c040034, data: 0x0002.
(D) (ModbusDecode)(C0) Received packet (7 bytes) 0c 04 04 45 e3 36 5d
[... cut ...]
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=31.2 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=31.2 A.
(I) (printStatus)(C0) STATE: C1 Error: 0 StartCurrent: -0 ChargeDelay: 13 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 26.2 A IsetBalanced: 31.2 A
(I) (printStatus)(C0) L1: -8.0 A L2: -8.4 A L3: 26.2 A Isum: 9.8 A
```

This is also very visible in the webui now. Before I would never see the battery, now it shows once in a while (probably at the update interval of 10s)

[image]

dingo35, 2024-03-05:
When I look into the code, the homebattery-currents are reset to zero unless you are in Solar mode:
`int getBatteryCurrent(void) {
    int currentTime = time(NULL) - 60; // The data should not be older than 1 minute

    if (Mode == MODE_SOLAR && homeBatteryLastUpdate > (currentTime)) {
        return homeBatteryCurrent;
    } else {
        homeBatteryCurrent = 0;
        homeBatteryLastUpdate = 0;
        return 0;
    }
}
`

So I think this has never worked in Normal or Smart mode; your Issue title suggests otherwise?

dingo35, 2024-03-05:
Ok I am happy to enable this stuff for Smart mode too, BUT I can not test this in real life; are you willing and able to test this thoroughly, including the interaction with the Belgium capacity tariff?

bartv, 2024-03-05:
> Ok I am happy to enable this stuff for Smart mode too, BUT I can not test this in real life; are you willing and able to test this thoroughly, including the interaction with the Belgium capacity tariff?

yes because this 100% my use case

bartv, 2024-03-06:
There is a lot of sun now, so testing the smart mode is not going to work during the day. I wil try later when there is a less sun or the sun is down.

I did test the solar mode and I notice that it does not really regulate the discharge current to 0. When the current is negative it will still regulate the charge current up. Attached is a trace log.

[solar_mode_battery.log](https://github.com/dingo35/SmartEVSE-3.5/files/14508600/solar_mode_battery.log)

dingo35, 2024-03-06:
Ok Bart, we have dozens of current variables in here, so you will have to be more clear to convey your message; please reformulate your problem in terms of the variables you see in the log, and in terms of expected value vs. reality value. Would be even better if you copied in the exact lines of the log you are referring to ....

bartv, 2024-03-06:
Sure, let me rephrase. My assumption is that when there is a non-zero `Battery Current`, the evse will regulate the charging current `Isetbalanced` up or down so that it gets to zero.

A negative `Battery Current` means that the home battery is discharging, you are charging the car battery from the home battery. When `Battery Current` is positive it means that the battery is charging and the car can charge at a higher rate.

In my HA is report the charging current of the battery subtracted with how much I want my home battery to charge now, like suggested in the documentation.

The interaction with the capacity tariff is as follows: when my battery reaches 35% SoC I will let it discharge only when the home consumption reaches the max peak. If I go over, the battery will "shave" the peak. I will only reach that peak in case the car is charging. By running it in solar or smart mode (depending on the season) the EVSE should regulate the charging current down again to not go over the peak.

Does this explain? If not, do not hesitate to ask more questions or information.

dingo35, 2024-03-06:
"Sure, let me rephrase. My assumption is that when there is a non-zero Battery Current, the evse will regulate the charging current Isetbalanced up or down so that it gets to zero."

That is a wrong assumption; Battery Current is divided by 3 (3 phases) and that number is used as an offset to the actual MainsMeter current values L1, L2, L3. You can see that on the webserver, you will also see the "Original" values there (which is the "uncorrected", measured value).
So you can use it to manipulate SmartEVSE's behaviour, but you cannot assume it will regulate your Battery Current to zero; I believe the Belgian inventor of this uses it to instruct SmartEVSE not to start charging from the battery, and you can also use it to "share" your solar power with both the car and the battery. But the intelligence would be in the feed of the Battery Current, not in the SmartEVSE.

bartv, 2024-03-06:
Ah. That explains. last year it worked fine in solar mode however now it does not. I figured out it regulates to 8A sum on mains. I have current_min set to 8 because 6 does not work for my car. Before I combined this with `solar_max_import` set to 0 so that it would only charge on solar if the minimal production is 8A.

In your branch of the firmware solar_max_import cannot be lower than current_min. Is there a specific reason for this?

On the regulation of the battery current, I know fully understand how the model works. However, I am not sure this is very usable in smart mode because that would mean you need to have two control loops (the one in SmartEVSE, and one in HA). My battery can deliver 2.8kW while I try to cap the peak usage to 7kW. This is quite a big hysteresis. 

I need to think about this a bit more how I can use this.

dingo35, 2024-03-06:
The restriction of solar_max_import is already taken out in my repository, it was based on my use of Solar Mode but it turned out it blocked many others.

Im going to disable battery currents in Smart mode again, if you have figured it out just post a feature request.

bartv, 2024-03-06:
Yeah, will do and we can close this ticket. I actually think that the current process is too complex to use. But this was probably easiest to implement. 

I will need to dive into the code again to see how it could work with smart mode + capacity tariff.

bartv, 2024-03-06:
and thx for the assistance!

closed 2024-03-06 by dingo35

bartv, 2024-03-06:
I actually did test the version you provided and it works like charm in smart mode. When using it in solar mode with import set to 0, it also works perfect. So for my personal use case it would actually be a good idea to have this in the 3.5.0 release

Based on my experiments today I did figure out some things that I think could simplify how it works. I will try to clean up my notes and post them as a discussion thread.

dingo35, 2024-03-06:
Well I threw it out few hours ago, this will have to wait for v.3.5.1 ....