dingo35/SmartEVSE-3.5 #27: Consumed EV power is not rounded off on display

rhannink, 2024-03-05:
I have been running version 1.8.0 (Serki) and 3.5RC1 (dingo35) and with both firmware versions I have the same problem:

I am using an Eastron SDM72 Kwh meter to measure the power transfered to the EV. I feed the Kwh meter from the bottom so I have to usb the Eastron inverted profile. 

On the display the value is not rondde off and very long, running through the schematisch drawing. (see the attached picture)

I use HA to sent the netmains values to the smartevse (REST API). 
Besides this cosmetic bug, everything is working fine. And my new Volvo EX30 is charged fine.

[image]

This is the config of my SmartEVSE:
```
{"version":"v3.5.0-RC1","mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"name-of-SSID","rssi":-65,"bssid":"AA:BB:CC:DD:EE:FF"},"evse":{"temp":27,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":272,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":160,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":75,"solar_max_import":2,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Not present","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"xxx.xxxx.xxx.xxx","port":1883,"topic_prefix":"SmartEVSE-laadpaal","username":"","password_set":false,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"InvEastrn","address":12,"import_active_power":-10.2,"total_kwh":52.8,"charged_kwh":3.2,"currents":{"TOTAL":470,"L1":157,"L2":157,"L3":156},"import_active_energy":52.8,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":470,"L1":160,"L2":150,"L3":160,"last_data_update":1709661440,"original_data":{"TOTAL":470,"L1":160,"L2":150,"L3":160}},"backlight":{"timer":120,"status":"ON"}}```
```
Here is a small part of the logfile:

```
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 16.0 A IsetBalanced: 16.0 A
(I) (printStatus)(C1) L1: 16.0 A L2: 15.0 A L3: 16.0 A Isum: 47.0 A
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0034, token:0x0c040034, data: 0x0002.
(MBhandleError)(C0) Error response: E0 - Timeout, address: 0c, function: 04, reg: 0034.
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement Node 0
(D) (ModbusSend8)(C0) Sent packet address: 0x0c, function: 0x04, reg: 0x0006, token:0x0c040006, data: 0x000c.
(D) (ModbusDecode)(C0) Received packet (27 bytes) 0c 04 18 41 7b 13 a0 41 7b 2d 9f 41 7b 70 6d c5 56 9d ae c5 55 76 85 c5 57 0b 5b
q(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x0c Function 0x04 Register 0x0006 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=16.0 A Imeasured=16.0 A MaxCircuit=16 Imeasured_EV=15.7 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1709661020,processed=1709661020.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=16.0 A, Idifference=0.3, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=16.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=16.0 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 16.0 A IsetBalanced: 16.0 A
(I) (printStatus)(C0) L1: 16.0 A L2: 15.0 A L3: 16.0 A Isum: 47.0 A
```

Gr. Remco

dingo35, 2024-03-05:
End of March I'll have this EVMeter connected to my testbench, please have patience till then!

dingo35, 2024-03-09:
We had some code changes, not sure if this solves it?

[05612caad1.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14547710/05612caad1.zip)

rhannink, 2024-03-10:
I am afraid its not solving anything. Even worse, I get a red flashing button and display showing that EV meter could not be connected and I should check wiring and configuration.

Gr. Remco

dingo35, 2024-03-11:
Ok I don't have my kWh meter at my current location so I can't test it myself, does this version solve the red flashing button and/or the rounded value?

[c9c27272a.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14557507/c9c27272a.zip)

rhannink, 2024-03-14:
Tot bad, the screen not shows different (power loaded instead of current) But also the non rounded number.

I think you are almost there ;-)
[image]

dingo35, 2024-03-31:
Problem solved in current master.

closed 2024-03-31 by dingo35

rhannink, 2024-04-01:
Thanks @dingo35, how can I test? Just wait for the 3.5 release or do you already have a binar I can test.

Gr. Remco

dingo35, 2024-04-01:
Sure you can test this:

[c75748d34de8.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14821882/c75748d34de8.zip)

rhannink, 2024-04-02:
@Dingo35

Thats strange. I will sent you a screenshot of my smartEVSE tonight, but it only shows current to the car. No power in Kwh is shown on my display. Where your display shows 10KWh mine shows 16A. And the 18.29 and 18.3 KWh is not showing at all.

Gr. Remco

rhannink, 2024-04-03:
The webinterfaces shows version: Version: 11:11:49 @Apr 1 2024
This is what my screen toggles between:
[image]
[image]
[image]

Gr. Remco

rhannink, 2024-04-04:
Check,

I think I already discovered what is the problem. Or I disabled the EV meter or it happened during upgrade.

`{"version":"11:11:49 @Apr  1 2024","mode":"OFF","mode_id":0,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"hannink.org","rssi":-64,"bssid":"F6:9F:C2:F7:C5:23"},"evse":{"temp":23,"temp_max":65,"connected":false,"access":false,"mode":1,"loadbl":0,"pwm":1024,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":160,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":75,"solar_max_import":2,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Not present","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"127.0.0.1","port":1883,"topic_prefix":"SmartEVSE-6828","username":"","password_set":false,"status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"**Disabled**","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":20,"L1":10,"L2":0,"L3":10,"last_data_update":1712212710,"original_data":{"TOTAL":20,"L1":10,"L2":0,"L3":10}},"backlight":{"timer":0,"status":"OFF"}}`

I will turn it on this evening and beste again, sorry for the fuzz....

rhannink, 2024-04-04:
Ah,

I just read the 3.5 RC2 release notes:

_Because we added some kWh meters, the config of your MainsMeter or EVMeter might be changed to the wrong type; we prevent this for future releases by reserving some slots so we can add more meters without changing their config id._

This is probably what happened?

dingo35, 2024-04-04:
Ok mystery solved!

rhannink, 2024-04-05:
@dingo35,

To bad, but the problem is still there. I enabled the inverted eastron meter again. Not the screens toggle between:
[image]
[image]
[image]

I also flashed the Smartevse with the latest 3.5-RC02 firmware, but still same result.

Gr. Remco

dingo35, 2024-04-05:
Flash RC2 in your SmartEVSE, and start actively charging, and then:
1) What does the Eastron show on Total Power on its OWN LCD screen?
2) What does at that moment, the webserver screen of the SmartEVSE show?
3) At that moment, dump the "Raw" output, so we can see what import_active_power reports

It looks like your meter reports some astronomical high power usage, which then is displayed as you report...
If you want to investigate that further:
4) Make a verbose debug log through telnet so we can see what modbus message is sent to your EV meter, and what message is returned....

I can "sort of" reproduce your screen when simulating pumping several GW into the SmartEVSE, but even then the screen is not _exactly_ like yours..... which points to your RC2 differs from my RC2 ?!?

dingo35, 2024-04-05:
Ok your negative power might be the problem here, I have to wait until my EV is back home to see what my production system is doing...

dingo35, 2024-04-05:
Does this improve things:

[88d9d5c18.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14888386/88d9d5c18.zip)

rhannink, 2024-04-07:
Yep,

This version is ok. The display not shows everything ok, it toggles through the different values without any rounding errors.

Thanks for the help. and keep up the good wok on this nice project.

Gr. Remco

rhannink, 2024-09-24:
Hello,

In version 3.6 of the SmartEVSE software, the not rounded off bug is re-introduced. 

Gr. Remco Hannink

dingo35, 2024-09-24:
Thx for your report; that fix is still in there; ...unfortunately no Eastron here at my testsite so I need more info to try to simulate the error. Could you provide:
-the display of the Eastron EV meter
-the display of the webserver
-the telnet log
-the settings dump

rhannink, 2024-09-27:
@dingo35 

Here the requested information. I see that the meter shows a negative power in the webinterface. I was not able to load the debug version / telnet logs. I wil sent these later.

Gr. Remco
[image]
[image]

`{"version":"v3.6.10","serialnr":6828,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"hannink.org","rssi":-64,"bssid":"F6:9F:C2:F7:C5:23"},"evse":{"temp":36,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":272,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":160,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":75,"max_sum_mains_time":0,"solar_max_import":2,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"127.0.0.1","port":1883,"topic_prefix":"SmartEVSE-6828","username":"","password_set":false,"status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"InvEastrn","address":12,"import_active_power":-10.39999962,"total_kwh":3100.100098,"charged_kwh":6.900000095,"currents":{"TOTAL":457,"L1":152,"L2":153,"L3":152},"import_active_energy":3100.100098,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":410,"L1":140,"L2":130,"L3":140,"last_data_update":1727427540,"original_data":{"TOTAL":410,"L1":140,"L2":130,"L3":140}},"backlight":{"timer":120,"status":"ON"}}`

rhannink, 2025-01-19:
@dingo35, this issue is fixed in 3.7.3. Everything works fine now, thanks an keep up the good work…..