dingo35/SmartEVSE-3.5 #31: Feat request: sensor with RFID id that activated charging

hmmbob, 2024-03-10:
**Is your feature request related to a problem? Please describe.**
There's currently no way to tell which of the approved / learned RFID tags enabled access to the charge point. This makes it impossible to use i.e. Home Assistant to track how much kWh is charged based on which RFID tag, usefull for users with multiple cars/cards for (self-) billing purposes.

**Describe the solution you'd like**
A sensor (via mqtt) that displays what RFID tag enabled the SmartEVSE. As an even better step, statistics of kWh charged per RFID tag, but not sure how feasible that would be :)

**Describe alternatives you've considered**
I can't really find another way to distinguish between which car is connected

fluppie, 2024-03-12:
Check nice one. I also put some RFID related FR here https://github.com/dingo35/SmartEVSE-3.5/issues/13
The RFID address is read in OneWire.cpp and stored in the char RFID. So I guess it should be able to post that char to mqtt if the rfid is a match.

`unsigned char RFID[8] = {0, 0, 0, 0, 0, 0, 0, 0};`

```
void CheckRFID(void) {
    unsigned char x;
    // When RFID is enabled, a OneWire RFID reader is expected on the SW input
    uint8_t RFIDReader = getItemValue(MENU_RFIDREADER);
    if (RFIDReader) {                                        // RFID Reader set to Enabled, Learn or Delete
        if (OneWireReadCardId() ) {                                             // Read card ID
            switch (RFIDReader) {
                case 1:                                                         // EnableAll. All learned cards accepted for locking /unlocking
                    x = MatchRFID();
                    if (x && !RFIDstatus) {
                        _LOG_A("RFID card found!\n");
```

hmmbob, 2024-03-13:
To sharpen my request a bit: it would already be helpful if the sensor shows "last RFID tag scanned" (because the new value will have a timestamp in home assistant anyways) if it proves to be too hard to make sure the RFID tag is indeed the one that initiated charging (after all, we can initiate charging in other ways too)

fluppie, 2024-04-10:
https://github.com/dingo35/SmartEVSE-3.5/blob/5707a8f5f0237050ef20e5641e9123581a6082c6/SmartEVSE-3/src/OneWire.cpp#L144

When opening putty and to a telnet to the module you can see this:
```(OneWireReadCardId)(C0) 94(OneWireReadCardId)(C0) 8a(OneWireReadCardId)(C0) 17(OneWireReadCardId)(C0) c3(OneWireReadCardId)(C0) 00(OneWireReadCardId)(C0) 00(OneWireReadCardId)(C0) ```

94 8A 17 C3 00 00 but in reality this is 00 00 C3 17 8A 94 for some reason you need to inverse them per 2 bytes to match laser etched number on those RFID tags.

So now there should be a way to convert RFID[x] to a string and then publish this string via MQTT and or HTTP, show in the webinterface?

fluppie, 2024-04-14:
@dingo35 do you think it's possible to log the tag ID to MQTT / HTTP from the for loop above?

Alfen charging equipement is reporting the above tag as 948A17C3 in their transactions list.

[image]

dingo35, 2024-04-14:
Test this alfpha 3.5.1 version:
[2b608a6869b.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14970553/2b608a6869b.zip)

It should publish:
SmartEVSE-6498/RFIDLastRead 01:01:02:03:04:05:06:F0  on MQTT and [evse][rfid_lastread] on the REST API.

fluppie, 2024-04-14:
Nice! That posts 01:94:8A:17:C3:00:00:2F for the 948A17C3 tag.
Is the 01 a start bit? and 2F the checksum?

dingo35, 2024-04-14:
That first byte is called "family mode" , not sure what it is used for. Last byte is checksum.

I thought export it all and see what the community does with it....

fluppie, 2024-04-14:
Yes this is a nice start. Thanks for this! Did you need to add a lot of code?

FYI for those curious out there how a charging session starts and stops with a backoffice connection :)

[image]

dingo35, 2024-04-14:
So did you bridge SmartEVSE to an OCPP solution? What tools/code did you use?

fluppie, 2024-04-14:
No, this is a screenshot from the Alfen ACE installer software. So we can have an idea if one day we are going the ocpp route :-).

dingo35, 2024-04-15:
@fluppie please take note: in the final version I will strip the first byte ("family code") and last byte ("checksum"), since the standards define they do not belong to the RFID id....

fluppie, 2024-04-15:
Indeed that's ok. I think it's good to see how parties like Alfen handle this, in public spaces you see them a lot.

fluppie, 2024-04-15:
BTW if possible also remove the : between bytes so we do not need to parse it. I think for OCPP it's a must the : are not there.

fluppie, 2024-04-17:
@hmmbob did you see the test version? https://github.com/dingo35/SmartEVSE-3.5/files/14970553/2b608a6869b.zip
Does it also work for you?
@dingo35 will this feature come to the next release?

hmmbob, 2024-04-18:
I did, but I'm away from home so did not have the opportunity to test yet

hmmbob, 2024-04-18:
Updated via VPN and asked my wife to swipe: I'm seeing the RFID data in the raw data of the SmartEVSE,  but not in the frontend or mqtt. Is that expected?

dingo35, 2024-04-18:
I havent tested this myself yet, received an RFID reader but still waiting for NFC tags to arrive ....

hmmbob, 2024-04-18:
It works, it's just missing the MQTT discovery information for HomeAssistant (and still has the 01: .... and checksum).

For testing, your _pinpas_ will also work 😉 💳 

(I had a space after the ~~ so the strikethrough did not work)

hmmbob, 2024-04-18:
> But so far I just tested using MQTT Explorer to see the activity.

Yeah, when I did the same I saw the info as well. It's there 😃

dingo35, 2024-04-18:
Well you guys you should have told that earlier about my pinpas, now I spilled 5 bucks on a couple of rewritable NFC's... :-)

(Of course I had tried it but the NXP app on my phone said "unformatted"; luckily the RFID reader from Stegen is not so picky!!)

This version publishes RFIDLastRead on homeassistant and has the : removed:
[182cf36892.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15025309/182cf36892.zip)

hmmbob, 2024-04-18:
> Well you guys you should have told that earlier about my pinpas, now I spilled 5 bucks on a couple of rewritable NFC's... :-)

Well, I just tried & figured it out to see if I could get you going again 😉  Never used it before 😆 And you'll find another use case for those cards, I'm sure!

> This version publishes RFIDLastRead on homeassistant and has the : removed: [182cf36892.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15025309/182cf36892.zip)

Works!
[image]

Were the units changed for the EVMeter as well? Suddenly they are in Wh (and KWh is pretty much the standard in EV land....) Or were they always like this and didn't I notice this before?
[image]

hmmbob, 2024-04-18:
Never noticed 😬  (But, in all honestly, I was using the `custom_component` for this info and I changed that this morning to fully MQTT). No worries, I'll add a few `round(1)` to my templates.

dingo35, 2024-04-28:
Done with 60e8035e

closed 2024-04-28 by dingo35