dingo35/SmartEVSE-3.5 #37: Switch to switch 1 or 3 phase charging

Elektrofix-OL, 2024-03-22:
**Is your feature request related to a problem? Please describe.**
My PV system charges my house battery (28.8kWh) during the day. When I get home late, I want to charge the car from the house batteries for the next day. However, the battery inverter can only deliver 4 kW and if my car charges with 3 phases, I draw electricity from the grid when someone switches on a consumer.

**Describe the solution you'd like**
I would like to have a button in the web interface where I can decide whether I want to charge with 1 or 3 phases, i.e. this button blocks the C2 output of the SmartEVSE, so that two phases are interrupted, like with solar charging.

**Describe alternatives you've considered**
Until now I have always set the circuit breakers for L2+L3 by hand, but this is stupid because the preheating of the car before departure is taken from the drive battery and not from the mains.
From 5 a.m. onwards, the IOBrocker switches the SmartEVSE to normal charging via API so that the car can be preheated at 6 a.m

**Additional context**
That's how I imagine it, that would be enough. If C2 is set to Disabel in the config, it doesn't even have to be switched out
[image]

dingo35, 2024-03-22:
If you are already using the API, why don't you just use 
curl -X POST http://ipaddress/settings?enable_C2=value , where value = 0-4?

From the manual:

If car is charging and you want to change from 1F to 3F or vice versa:

  - Change mode to OFF
  - Enable or disable C2 contactor
  - Change to desired value: 0 "Not present", 1 "Always Off", 2 "Solar Off", 3 "Always On", 4 "Auto"
  - Examples:
  - If the desired C2 mode is "Solar Off", the string to be sent is 2

We are working on a fully automatic solution so you wouldn't have to switch 1F / 3F manually, so that's why there is no button on the webpage; one user already posted a PR for this. If we can't get it automated we'll put a button on the webpage....

Elektrofix-OL, 2024-03-24:
When I send http://192.168.5.23/settings?enable_C2=0 or http://192.168.5.23/settings?enable_C2=1
I get: 
`{"version":"v3.5.0-RC1","mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"iot","rssi":-71,"bssid":"6A:D7:9A:24:A1:8D"},"evse":{"temp":27,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":0,"pwm":1024,"solar_stop_timer":0,"state":"Charging Stopped - Waiting for Solar","state_id":9,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":20,"current_main":50,"current_max_circuit":25,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":10,"solar_stop_time":1,"enable_C2":"Solar Off","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.5.2","port":1883,"topic_prefix":"SmartEVSE","username":"Lehm-WG","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":1711273336},"ev_meter":{"description":"Disabled","address":10,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":-48,"L1":-17,"L2":-17,"L3":-14,"last_data_update":1711273336,"original_data":{"TOTAL":-48,"L1":-17,"L2":-17,"L3":-14}},"backlight":{"timer":0,"status":"OFF"}}`
C2 is not changed.
I don't know if it has anything to do with configuring with C2, but I also notice that C1 no longer switches off, i.e. it always stays on. Replacing the solid-state relay didn't help either. C1 only drops when I switch off the voltage from the controller for 5 seconds. If the switch-off time is less than 5 seconds, C1+C2 are immediately active again.
I'm only a master electrician, so I can only build and test, programming is not my thing

dingo35, 2024-03-24:
Ok I've had an earlier report on enable_C2 not working correctly, so this is probably a bug in v3.5.0-RC1; in my current location I dont have any contactors available, so please wait a week until I can test myself.

Second, Solid State relais are known to give problems, so first use old fashioned coil  relais, if that works you can switch out your relais for an SSR and see if that particular combination works.

You checked your wiring, using C1 and neutral for the first contactor, and using C2 and neutral for the second?

Elektrofix-OL, 2024-03-24:
I swapped the solid state relays in the EVSE (IC10 & IC11). The behavior at C1 shows an SSR which can no longer delete itself if there is no longer a signal at the input.
The power to the EV is switched by installation contactors; SSR is not a safe separation and may not be used here.

The installation contactors are each connected to C1 and C2 and are returned together to N.

dingo35, 2024-04-22:
In v3.5.1 the enable_C2 setting is reported to work ok, so closing this issue.

closed 2024-04-22 by dingo35