dingo35/SmartEVSE-3.5 #38: SmartEVSE MQTT does not reconnect after network failure

negronoir, 2024-03-25:
**Release v3.5.0-RC1**

Hi,

Doing some network test I realized  that MQTT connection from SmartEVSE to Mosquitto broker is not restarted if a network comms fail happens.

When I start mosquitto broker, everything runs fine. Each device connects: SmartEVSEs, Axpert PV Inverter, HA, Tasmota devices, etc.

```
2024-03-25 09:28:48: New connection from 192.168.1.9:64412 on port 1883.
2024-03-25 09:28:48: New client connected from 192.168.1.9:64412 as SmartEVSE-Tesla (p2, c1, k15).
2024-03-25 09:28:48: Will message specified (7 bytes) (r1, q0).
2024-03-25 09:28:48: 	SmartEVSE-Tesla/connected
2024-03-25 09:28:48: Sending CONNACK to SmartEVSE-Tesla (0, 0)
...
2024-03-25 09:28:53: New connection from 192.168.1.8:60094 on port 1883.
2024-03-25 09:28:53: New client connected from 192.168.1.8:60094 as SmartEVSE-Mercedes (p2, c1, k15).
2024-03-25 09:28:53: Will message specified (7 bytes) (r1, q0).
2024-03-25 09:28:53: 	SmartEVSE-Mercedes/connected
2024-03-25 09:28:53: Sending CONNACK to SmartEVSE-Mercedes (0, 0)
...
2024-03-25 09:29:00: New connection from 192.168.1.194:57588 on port 1883.
2024-03-25 09:29:00: New client connected from 192.168.1.194:57588 as Axpert-0DD700 (p2, c1, k15).
2024-03-25 09:29:00: Will message specified (5 bytes) (r1, q0).
2024-03-25 09:29:00: 	Axpert/Alive
2024-03-25 09:29:00: Sending CONNACK to Axpert-0DD700 (0, 0)
...
2024-03-25 09:29:15: New connection from 192.168.1.4:39804 on port 1883.
2024-03-25 09:29:15: New client connected from 192.168.1.4:39804 as mqtt_home_assistant (p5, c0, k15).
2024-03-25 09:29:15: Will message specified (7 bytes) (r0, q0).
2024-03-25 09:29:15: 	homeassistant/status
2024-03-25 09:29:15: Sending CONNACK to mqtt_home_assistant (0, 0)
...
2024-03-25 09:29:15: New connection from 192.168.1.113:57846 on port 1883.
2024-03-25 09:29:15: New client connected from 192.168.1.113:57846 as DVES_A9C3D4 (p2, c1, k30, u'DVES_USER').
2024-03-25 09:29:15: Will message specified (7 bytes) (r1, q1).
2024-03-25 09:29:15: 	tele/tasmota_A9C3D4/LWT
2024-03-25 09:29:15: Sending CONNACK to DVES_A9C3D4 (0, 0)
```

After some minutes, I power off WiFi AP so all devices except HA (wich is network wired) are disconnected

```
2024-03-25 09:33:40: Client SmartEVSE-Mercedes has exceeded timeout, disconnecting.
2024-03-25 09:33:40: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'SmartEVSE-Mercedes/connected', ... (7 bytes))
...
2024-03-25 09:33:46: Client SmartEVSE-Tesla has exceeded timeout, disconnecting.
2024-03-25 09:33:46: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'SmartEVSE-Tesla/connected', ... (7 bytes))
...
2024-03-25 09:33:46: Client Axpert-0DD700 has exceeded timeout, disconnecting.
2024-03-25 09:33:46: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'Axpert/Alive', ... (5 bytes))
...
2024-03-25 09:34:04: Client DVES_A9C3D4 has exceeded timeout, disconnecting.
2024-03-25 09:34:04: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'tele/tasmota_A9C3D4/LWT', ... (7 bytes))

```
Then I swith on AP and rest of devices reconnect

```
2024-03-25 09:38:29: New connection from 192.168.1.117:56767 on port 1883.
2024-03-25 09:38:29: New client connected from 192.168.1.117:56767 as DVES_A9C3D4 (p2, c1, k30, u'DVES_USER').
2024-03-25 09:38:29: Will message specified (7 bytes) (r1, q1).
2024-03-25 09:38:29: 	tele/tasmota_A9C3D4/LWT
...
2024-03-25 09:39:00: New connection from 192.168.1.194:54328 on port 1883.
2024-03-25 09:39:00: New client connected from 192.168.1.194:54328 as Axpert-0DD700 (p2, c1, k15).
2024-03-25 09:39:00: Will message specified (5 bytes) (r1, q0).
2024-03-25 09:39:00: 	Axpert/Alive
2024-03-25 09:39:00: Sending CONNACK to Axpert-0DD700 (0, 0)
```

But EVSEs don't do it so they remain "unavailable" forever in HA. To make then available again, I must reboot them

Thanks!

dingo35, 2024-03-25:
1. I cannot reproduce this, at my site they reconnect both to wifi and HA when available.  Could you test 
[de779d4d4.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14742763/de779d4d4.zip)

2. If they don't reconnect, are they available to the network (can you ping them), or are they available to the network but not to the MQTT server?

negronoir, 2024-03-25:
Like a charm.

I've tested five cycles of switching off/on the AP and the change works

Thanks!

closed 2024-03-25 by dingo35