dingo35/SmartEVSE-3.5 #40: Contactor switches on/off/on when the charging session starts

mintgroen, 2024-03-29:
**Describe the bug**
When charging starts the EV requests the charging session to start, contactor does click on, clack off, click on and remains on.

**To Reproduce**
3.5 RC1 and 3.5 RC2 both show the same behavior.

Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar? Does not matter
2. What steps do you take so the problem arises? Start charging

**Expected behavior**
Contactor should only close once

I did not recall that I noticed this behavior in the past so I reverted back to Serkri v1.8 and tried 10 times, the symptom did not reappear (yet, I will monitor it closely)

If needed I can flash to 3.5 and post the debug.

dingo35, 2024-03-31:
@mintgroen , Maarten, via PM I thought we agreed you'd tried to find the commit that caused this, since I cannot reproduce the problem (probably EV related).

Any results on that?

mintgroen, 2024-03-31:
Yes, so far Serkri v1.8 works ok.

mintgroen, 2024-04-19:
20 charging sessions later, symptom did not occur again. 

Do you know what the changes in state interpretation are between 1.8 and 3.5?

dingo35, 2024-04-19:
Symptom did not occur again at which version?

Many commits between 1.8.0 and 3.5, I thought you were going to find out which one is the culprit?

dingo35, 2024-05-22:
Inactive for too long....

closed 2024-05-22 by dingo35

mintgroen, 2024-05-22:
After enabling the more detailed logging of the CP voltages Stegen traced it down to a hardware variation which resulted in the faulty state detection. Stegen send another SmartEVSE to test the behaviour, seems that the calibrated ADC still has some margins and other infuences affecting the voltage levels.

Installed the new SmartEVSE today, programmed with 3.6 and did a quick smoke test.  The state detection works ok, relay kicks in and stays in.

dingo35, 2024-05-22:
Ok thx for your feedback Maarten!

mintgroen, 2025-02-06:
Some weeks ago, the issue reappeared. I rolled back to the test version but that did not solve the issue.

Checking with Michael again and analysing the scope recordings I noticed that the PWM was reducing the available power to min after the switch from B to C. 
My assumption is that the Kona detects the PWM incorrectly and reads error instead of 6A. Then the kona reverts back to state B, EVSE again raises the available power, Kona again switches to C and remains C.

Michael advised to set the min current to 10A and try again. That works, even setting the min current to 7A resolves the issue.

So my proposal to resolve the C-B-C switching is to increase the min current to at least 7A in the intial stages of switching to state C. during the charge process, min current can be set at the configured min current.

mintgroen, 2025-02-06:
Another hypothesis, it could be misread due to leading edge of the signal is not completely square, as you can see at 3s, the left top corner is notched:

https://photos.app.goo.gl/FeWVyq9yrw7KpJVX8

It can also be my cheap-o-scope, else I could retry it once with a better tuned trigger.

Once the EVSE goes to state B, the PWM is activated at the currently available power. In my recording it is about 20%, when switching to C, the PWM cuty cylce is reduced (to start at the min current, what is the reason behind that squeezing of the PWM?

Yes setting the min current to 7 solves the symptom, the downside is that the  min current is also the min for solar mode. Resulting is 3A more import while there is less solar.

A workaround could be to only set the min current to 7 when there is no EV connected and switching to 6 when the charging session is running. The EVSE does not listen to MQTT /set/chargecurrent 60?