dingo35/SmartEVSE-3.5 #41: Contactor C1 does not switch off safely

Elektrofix-OL, 2024-03-31:
**Describe the bug**
The contactor C1 stays on even if charging is stopped and the plug is pulled out. This behavior was observed in Solar and Smart modes.

**Upload your config**
`{"version":"v3.5.0-RC1","mode":"SOLAR","mode_id":2,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"iot","rssi":-62,"bssid":"6A:D7:9A:24:A1:8D"},"evse":{"temp":30,"temp_max":65,"connected":false,"access":true,"mode":2,"loadbl":0,"pwm":1024,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":20,"current_main":50,"current_max_circuit":25,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":5,"enable_C2":"Solar Off","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.5.2","port":1883,"topic_prefix":"SmartEVSE","username":"Lehm-WG","password_set":true,"status":"Connected"},"home_battery":{"current":170,"last_update":1711889048},"ev_meter":{"description":"Disabled","address":10,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":-163,"L1":47,"L2":-112,"L3":-98,"last_data_update":1711889048,"original_data":{"TOTAL":5,"L1":103,"L2":-56,"L3":-42}},"backlight":{"timer":0,"status":"OFF"}}
`

**To Reproduce**
1. Plug the vehicle into EVSE (solar mode)
2. Vehicle wakes up and waits for charging (switching C1)
3. Interrupt the charging process from the vehicle by unlocking the plug. (C1 still active) After 10 seconds the plug will unlocked and can be pulled out, C1 is still active.
4. Plug the vehicle again, the vehicle goes on error, because the voltage is present at the plug, without a request from the vehicle.
5. By switching off the supply voltage for the SmartEVSE, C1 drops out. After switching on the supply voltage, C1 switches on again, the vehicle is charging.
6. Same in Smart mode

**Expected behavior**
Compliance with the IEC 61851 charging protocol

**Attempted remedial measures**
My first guess was a defect in the 230v control circuit:
- both SSR (IC10 and IC11) changed on PCB {could have been overloaded}
- Exchange M3 from phase<=>C1 to neutral<=>C1, {perhaps, could not completely go back to the initial value due to increased mains voltage via IC11}
Both measures were unsuccessful, so the 230V circuit is OK.

*** Suspected cause of error ***
The SSR1 continues to be supplied with a signal from the ESP32, although the signal must be 0V.
Perhaps the IO32 in the software is not controlled as a digital output but as a PWM. The SSR switches the output via the 0 transition,  if there was a PWM signal at output IO32, the SSR would not switch off.

*** Software try ***
I noticed the problem with V3.5 RC1, I downgrade to Sekri 1.6.0 with no changes happend. The downgrade is done via OTA (spiffs.bin + firmware.bin)

dingo35, 2024-03-31:
I could not reproduce this on the current master branch; could you please test:
[a04fb335d.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14816182/a04fb335d.zip)

For your convenience I added log messages when the contactors are switched by the firmware, just telnet to your SmartEVSE and grep Switching will give you:
`(setState)(C1) Switching Contactor1 OFF.
(setState)(C1) Switching Contactor2 OFF.
(setState)(C1) Switching Contactor1 ON.
(setState)(C1) Switching Contactor2 ON.
(setState)(C1) Switching Contactor1 OFF.
(setState)(C1) Switching Contactor2 OFF.
`

dingo35, 2024-03-31:
By the way, you are using coil based (non-solid-state) contactors, right?

It is known that the SSR's on the pcb cannot (reliably) drive an SSR that has to switch the mains to the EV. We need galvanic separation there....

Elektrofix-OL, 2024-03-31:
The SSR are on the SmartEVSE V3.0 board
There are 230V coils at the output C1 and C2

Dark days? Today 12.42 kWh came into the house battery and now 1.3 kW comes from the roof. It's pretty bright for me :-))

Elektrofix-OL, 2024-03-31:
I just tested it briefly. In the log C1 and C2 are off, but the output of C1 is 230V. So if the ESP32 does not give a PWM signal to the IO32, then the chip is defective because the SSR1 continues to let current through
`=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2024.03.31 18:20:01 =~=~=~=~=~=~=~=~=~=~=~=
* Debug level set to Info
[0m[1;33m(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.8 A IsetBalanced: 14.0 A
[0m(ProximityPin)(C0) PP pin: 1023 (1943 mV) (warning: fixed cable configured so PP probably disconnected, making this reading void)
[0m[1;33m(I) (EVSEStates)(C1) Cable limit: 20A  Max: 20A
[0m[1;33m(I) (Timer1S)(C0) Waiting for Solar power...
[0m(setState)(C1) 18:21:08 STATE A -> B1
[0m(setState)(C0) Pilot Disconnected
[0m(setState)(C0) Switching Contactor1 OFF.
[0m(setState)(C0) Switching Contactor2 OFF.
[0m[1;33m(I) (printStatus)(C0) STATE: B1 Error: 32 StartCurrent: -6 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.8 A IsetBalanced: 14.5 A
[0m(setState)(C0) Switching Contactor1 OFF.
[0m(setState)(C1) Switching Co(EVStactor2 OFF.Pi
[0mlot C(setState)(C1) Switching Co(EVStactor2 OFF.Pi
[0mlot Connected
[0m(setState)(C0) Switching Contactor1 OFF.
[0m(setState)(C1) Switching Contactor2 OFF.
[0m(setState)(C0) Switching Contactor1 OFF.
[0m(setState)(C1) Switching Contactor2 OFF.
[0m(setState)(C0) Switching Contactor1 OFF.
[0m(setState)(C1) Switching Contactor2 OFF.
[0m[1;33m(I) (printStatus)(C0) STATE: B1 Error: 32 StartCurrent: -6 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.8 A IsetBalanced: 15.5 A
[0m[1;33m(I) (printStatus)(C0) L1: 0.8 A L2: -1.4 A L3: -1.1 A Isum: -1.7 A
[0m(ProximityPin)(C0) PP pin: 1023 (1943 mV) (warning: fixed cable configured so PP probably disconnected, making this reading void)
[0m[1;33m(I) (EVSEStates)(C1) Cable limit: 20A  Max: 20A
[0m(setState)(C1) 18:21:27 STATE B1 -> B
[0m(setState)(C1) Switching Contactor1 OFF.
[0m(setState)(C1) Switching Contactor2 OFF.
[0m(EVSEStates)(C0) Diode OK
[0m(setState)(C0) 18:21:27 STATE B -> C
[0m(setState)(C1) Switching Contactor1 ON.
[0m(setState)(C1) Switching Contactor2 ON.
[0m[1;33m(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.8 A IsetBalanced: 15.5 A
[0m[1;33m(I) (printStatus)(C0) L1: 0.8 A L2: -1.4 A L3: -1.1 A Isum: -1.7 A
[0m(setState)(C0) 18:21:49 STATE C -> B
[0m(setState)(C0) Switching Contactor1 OFF.
[0m(setState)(C0) Switching Contactor2 OFF.
[0m(EVSEStates)(C0) Diode OK
[0m[1;33m(I) (printStatus)(C0) STATE: B Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.2 A IsetBalanced: 20.0 A
[0m[1;33m(I) (printStatus)(C0) L1: 5.2 A L2: -0.8 A L3: -0.6 A Isum: 3.8 A
[0m(setState)(C0) 18:22:57 STATE B -> B1
[0m(setState)(C1) Switching Contactor1 OFF.
[0m(setState)(C1) Switching Contactor2 OFF.
[0m(ProximityPin)(C0) PP pin: 1023 (1943 mV) (warning: fixed cable configured so PP probably disconnected, making this reading void)
[0m[1;33m(I) (EVSEStates)(C1) Cable limit: 20A  Max: 20A
[0m[1;33m(I) (Timer1S)(C0) Waiting for Solar power...
[0m(setState)(C1) Switching Contactor1 OFF.
[0m(setState)(C1) Switching Contactor2 OFF.
[0m(setState)(C0) Switching Contactor1 OFF.
[0m(setState)(C1) Switching Contactor2 OFF.
* Debug: Command received: q
* Closing client connection ...
`

dingo35, 2024-03-31:
So we conclude you have a hardware problem....

closed 2024-03-31 by dingo35

Elektrofix-OL, 2024-03-31:
Yes, it seems so. There must be a defect between esp32 and SSR1. It's only 10 months old, what bad luck. The other four that I installed have been running smoothly for years.

Imaginous, 2024-03-31:
@Elektrofix-OL Just contact Stegen, it will be solved. 
Great guys, good service.

Elektrofix-OL, 2024-04-20:
The solution of the problem:
The SSR in EVSE is not compatible with electronically modified contactors, such as the Hager ESC425S. It is essential to use old-fashioned switches for the EVSE. It works well with a Hager ESC425 (without S).

Technical explanation:
The SSR in the EVSE monitors the zero crossing of the alternating current and then switch C1 or C2 on or off.
The "S" from Hager ESC425S stands for silent and shows that the contactor does not hum. The Hager ESC425S is not an electronic contactor, it has a mechanical switch, but with this contactor the sine wave is manipulated and the SSR no longer recognizes the zero crossing.

dingo35, 2024-04-20:
I dont understand why you keep posting on this subject. It has been known for years that SmartEVSE does not work well with SSR's.
You also pointed out yourself that galvanic separation is necessary, which also rules out SSR's.

Elektrofix-OL, 2024-04-20:
I don't know if there are differences in English, but in Germany we technically differentiate between:
- Solid State Relays (SSR): Electronic switches based on triacs, as can be found in the EVSE circuit diagram as IS10 and IS11
- Contactor: electromagnetic switch for high load currents with a coil as drive and galvanic isolation
- Contactor (hum-free): electromagnetic switch for high load currents with a modified coil as a drive and galvanic isolation

The ESC425S is a normal commercial contactor and not an SSR. So I didn't consider your messages relevant because the ESC425S is technically not an SSR. I have operated the ESC425S with EVSE V2.0 for years without any problems.

It was only Michael who told me that there is problems with exactly this contactor.

I had to specially order the old-fashioned ESC425 contactor because my electrical wholesaler no longer had it in its standard range. The hum-free contactors are now standard.

I posted this information here specifically after my at-home test so that someone else doesn't have to search if they experience the error and stumble across this post.
I haven't read yet that hum-free contactors don't work with the EVSE. It may be due to the language barrier, but after 40 years in electronics I know the differences between SSR and contactors. It's also new to me that a hum-free contactor causes such trouble.
I'm sorry if I took up your time, it's my last post on this topic.
The EVSE is very good and I have already installed three of them.

dingo35, 2024-04-20:
Great that you like them, I didnt want to insult you, but I never heard my old fashioned relais humming... I guess you might notice if you install your SmartEVSE in your bedroom ?!?

Elektrofix-OL, 2024-04-20:
Larger contactors tend to hum with age and switching cycles. Customers then think it's broken and demand recourse from the electrician who installs it, so the electricians install a hum-free device for €10 more and everyone is happy. It probably has something to do with our German mentality: taking note, demanding rights and denouncing neighbors.

Keep it up, you're on the right track

dingo35, 2024-04-20:
Very enlightening, thx!