dingo35/SmartEVSE-3.5 #45: RC2 final meeting reading missing

rvdgaag, 2024-04-05:
[image]

RC2 works fine here, however when the session is finished the total charged kWH is not is not displayed.

{"version":"v3.5.0-RC2","mode":"OFF","mode_id":0,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"McWifi_IoT","rssi":-53,"bssid":"D0:21:F9:73:17:8A"},"evse":{"temp":25,"temp_max":65,"connected":true,"access":false,"mode":1,"loadbl":1,"pwm":1024,"solar_stop_timer":0,"state":"Charging Stopped","state_id":9,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":25,"current_max_sum_mains":65,"solar_max_import":2,"solar_start_current":4,"solar_stop_time":3,"enable_C2":"Solar Off","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.1.11","port":1883,"topic_prefix":"SmartEVSE1","username":"energie","password_set":true,"status":"Connected"},"home_battery":{"current":-38,"last_update":1712293971},"ev_meter":{"description":"Eastron3P","address":101,"import_active_power":0,"total_kwh":9711.900391,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":9711.900391,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":83,"L1":47,"L2":28,"L3":8,"last_data_update":1712293971,"original_data":{"TOTAL":83,"L1":47,"L2":28,"L3":8}},"backlight":{"timer":120,"status":"ON"}}

dingo35, 2024-04-05:
@rvdgaag Was this in Smart mode or in Solar mode?

rvdgaag, 2024-04-05:
It was in the smart mode

dingo35, 2024-04-05:
Now I also have those "resets" to zero (like @djoenez):

[image]

...but they happen exactly when I start charging:

[image]

...so that is intended behaviour.

What state changes can you see that cause a reset? I'm afraid my test rig only shows this kind of behaviour when I am at state B1, which should then be reset to B by the EV... which turns back to normal behaviour.

dingo35, 2024-04-05:
@rvdgaag : You are sure you were charging on the Master, and snapshot and config are all from Master device?

So we are not talking about the Node / Slave device ?

rvdgaag, 2024-04-05:
Yup its the master. The slave unit is only used for my wifes car

dingo35, 2024-04-05:
Well both my fellow dev and I cannot reproduce, nor explain it (going through the code), perhaps djoenez can give us some pointers on his observations / logs...

rvdgaag, 2024-04-06:
yesterdays and last nights charging sessions were ok, the reporting was not reset again. So let's consider this a hickup.

djoenez, 2024-04-06:
[image]
Dove into the reset. i see the smartevse becomes unavailable (so a reboot?). after that it started charging again. I agree with rvdgaag. let's see it as a hickup, and ill monitor if it happens more often and open a new issue if that is the case.

closed 2024-04-06 by dingo35