dingo35/SmartEVSE-3.5 #49: Crashing modules, I think due to wifi ssid and pass with spaces in it

fluppie, 2024-04-13:
**Describe the bug**
I have an installation with two SmartEVSE's that frequently lock-up. Only a hard power cycle makes them run again.
The wifi router is in the same technical room as the ESP's, both have a strong signal, RSSI of -28 to -32 dBm.
But something that might f*ck things up is a SSID similar to "Hello World 24" and the pass being identical as the SSID.
It happens every few days.

**Upload your config**
I have no access to the site without a physical visit. I only pull stats over MQTT, no HTTP port forwards for the webinterfaces.

**Upload a debug log**
N/A

**To Reproduce**
Steps to reproduce the behavior:
Just wait
Normal mode, 16A, 3F installation with EV meters
2x SmartEVSE as Master and Slave

**Expected behavior**
No lock ups :-)
**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.
[image]

dingo35, 2024-04-14:
Well that's a problem, if the spaces are the cause of your lock-ups, because then the problem would not lie into our firmware but in the underlying libraries we use....

You might try this alpha 3.5.1 version, that uses a lot of different libraries:
[26eae13983e.zip](https://github.com/dingo35/SmartEVSE-3.5/files/14970050/26eae13983e.zip)

And the obvious question you probably already tried: if you remove the spaces from ssid and password, is the problem solved then?

fluppie, 2024-04-14:
I've configured the same SSID and password here at home to my two own SmartEVSE's (not in Master/Slave atm) due to one with a Sensorbox mainsmeter and another one using an SDM230 mainsmeter. So let's see if they will also crash.

First I'll try with the current firmware. If crashing occurs I'll try the above fw.

devdems, 2024-04-18:
> Well that's a problem, if the spaces are the cause of your lock-ups, because then the problem would not lie into our firmware but in the underlying libraries we use....
>
> You might try this alpha [...]

@dingo35 is this one built from mqtt branch?

dingo35, 2024-04-18:
No its from the mongoose2-plus-http-client branch; the hex code is the commit id.

dingo35, 2024-04-18:
@fluppie Did you check for signal interference --> "technical room" suggests a larger chance of equipment transmitting pulses &  spikes?

fluppie, 2024-04-19:
No, but it's a out-of-the-box router from the ISP, which are not great for SMB use. We now remotely updated them to 3.5.0 release. I think they were on RC1. So let's see if problems persist or are now solved.
In case issues remain we'll test the above release (which I have running on my own setup with their SSID en pass mimicked).And otherwise we'll propose a WiFi network/router update.

fluppie, 2024-04-23:
So far so good :). So might be an issue with an older RC? If by the end of this week no crashes occured, then we can mark this one as solved.

dingo35, 2024-04-24:
Something might have changed in one of the underlying libraries we use...

closed 2024-04-28 by dingo35