dingo35/SmartEVSE-3.5 #50: MAINSMET setting disappeared with RC2 when setting up from device

stefan-home, 2024-04-15:
**Describe the bug**
When going to configuration on the device the "MAINSMET" setting is no longer visible

**Upload your config**
RC2 has this problem
RC1 is OK

**To Reproduce**
Steps to reproduce the behavior:
1. tested on OFF and NORMAL, both states have this issue
2. What steps do you take: very standard, just hold the mid-button for 2 secs and find the MAINSMET setting by pushing right button. It's not there

**Expected behavior**
Expect MAINSMET setting to be there (alike RC1 and older versions like 1.8)

**Screenshots**
cannot make a screenshot of a screen that is missing

(note: this is my very 1st issue I report on GitHub, made account just for this, so hope I do ok, user stefannn on tweeters)

dingo35, 2024-04-15:
Yes this is intended behaviour, there are a lot of users that just run "Normal Mode" and have no devices attached. They get confused by "MainsMeter" settings, and the manufacturer gets a lot of questions/problems on this subject.

So only if you choose Smart or Solar mode you can configure MainsMeter.

closed 2024-04-15 by dingo35

stefan-home, 2024-04-15:
Thanks,
in that case I would suggest to modify documentation: note, I would volunteer to do if you would like.

For info: I really tried not send an "intended behaviour" issue. Before sending in I checked the documentation and also verified on RC1.

configuration.md / how to configure
MAINSMET Set type of MAINS meter
add: "only available in SMART mode"

dingo35, 2024-04-15:
Yes, this is intentionally left out of the documentation; if you experiment on this detailed level you should familiarize with the source code, and there you would find this option....