dingo35/SmartEVSE-3.5 #51: Allow MIN_CURRENT down to 2A

stefan-home, 2024-04-15:
**Is your feature request related to a problem? Please describe.**
I want to test my car towards very low settings. Reason is that I hope to find a setting that the car keeps having "charging mode" but is not pulling energy. This setting I would potentially like to use as "soft switch-off in case of mains overload". This would allow to dial down the load current to 0A without having to switch the relais.
This wold also prevent the car from sending "charging ready" messages.
However, I cannot test this because MIN_CURRENT cannot be set below 6A.
I tried setting manual on the device.
I also tried seeing using the api.
Both did not allow setting below 6A.
Note: I'm not using smart mode but do the power-control by own home-automation (especially because I have an old meter that does not give power-per-phase)

**Describe the solution you'd like**
As MIN_CURRENT is already a setting intended for developers only, my request is to extend the range down to (example) 2A.

**Describe alternatives you've considered**
A method that could set override current to 0 but keep car in "charge mode" could do the same but I think that is not supported.
Note: in granting this request I will test a bit and potentially find setting that I will use in my home-automation but COULD in a future smartevse-release be added to create such "zero-power charging-active" state.

**Additional context**
n.a.

Imaginous, 2024-04-15:
6A is the minimum current according to the charging standards.

stefan-home, 2024-04-15:
I know... I would like to experiment. And this setting is for experimenting anyways (see comments about Zoe in the documentation on it....)

dingo35, 2024-04-15:
I think your use case does not make any sense, better replace your relais in 5 years than your EV; you WILL be operating outside the specifications of the protocols, and it is unknown what will happen to the circuits in your car.

BUT if your really, really want to live dangerously, you can change MIN_CURRENT compile time. Look at 6baf5afb24c

closed 2024-04-15 by dingo35

stefan-home, 2024-04-15:
Understandable comment.

coding myself would not be difficult.
Setting up the coding & compile environment is a bridge too far at this moment.
I expect I will do that "some time in future", does not fit with worklife now.

If/When I setup I will probably add (and create PR) for power control function for P1-meters that only have combined current (not per phase). I have that functional in my home automation driving the smartens by api.
As said, bridge too far at this moment.

dingo35, 2024-04-15:
Appreciate your help, but I must warn you: I dont think we will accept PR that interfaces P1 meters that dont have separate phase info. It is deemed "unsafe" to assume currents instead of measuring them.

dingo35, 2024-04-15:
..and I understand those "old" meters are going to be replaced anyways, because 2G and/or 3G networks are being phased out...