dingo35/SmartEVSE-3.5 #53: Solar Mode - stopping timer resets on API meter update

demerstraat, 2024-04-17:
**Describe the bug**
When in SOLAR mode, 
"waiting for SOLAR" works as expected: once meter sees current export > solar start setting, charging starts, 
and charging current is adjusted to keep solar import at desired setting. 

however, when solar production drops, this triggers the "stop" timer, as expected,
it starts counting down, but never reaches 0 secs, even though solar production dropped. 
at first sight, it counts down from 600s to 597s and then resets, 
this is coherent with my API meter update interval: at least every 3 seconds, and on any current change >0.1A

**Upload your config**
{
  "version": "v3.5.0",
  "mode": "SOLAR",
  "mode_id": 2,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "meeuw",
    "rssi": -68,
    "bssid": "1C:ED:6F:3D:BD:87"
  },
  "evse": {
    "temp": 23,
    "temp_max": 65,
    "connected": true,
    "access": true,
    "mode": 2,
    "loadbl": 0,
    "pwm": 102,
    "solar_stop_timer": 598,
    "state": "Charging",
    "state_id": 2,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 60,
    "override_current": 0,
    "current_min": 6,
    "current_max": 16,
    "current_main": 16,
    "current_max_circuit": 16,
    "current_max_sum_mains": 18,
    "solar_max_import": 0,
    "solar_start_current": 8,
    "solar_stop_time": 10,
    "enable_C2": "Always Off",
    "modem": "Not present",
    "mains_meter": "API",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0
  },
  "mqtt": {
    "host": "192.168.68.30",
    "port": 1883,
    "topic_prefix": "SmartEVSE",
    "username": "mqtt-ha",
    "password_set": true,
    "status": "Connected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "API",
    "address": 12,
    "import_active_power": 1.200000048,
    "total_kwh": 126,
    "charged_kwh": 5.699999809,
    "currents": {
      "TOTAL": 52,
      "L1": 52,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 35,
    "L1": 56,
    "L2": -10,
    "L3": -11,
    "last_data_update": 1713346761,
    "original_data": {
      "TOTAL": 35,
      "L1": 56,
      "L2": -10,
      "L3": -11
    }
  },
  "backlight": {
    "timer": 120,
    "status": "ON"
  }
}
**Upload a debug log**
Flash the debug version of the firmware (see https://github.com/dingo35/SmartEVSE-3.5/blob/master/SmartEVSE-3/HowToFlash.txt), telnet to your device, capture the debug log, cut it so that it shows JUST BEFORE and JUST AFTER the problem arises, and upload it here.
[evse-debug-log.txt](https://github.com/dingo35/SmartEVSE-3.5/files/15009254/evse-debug-log.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar?
SOLAR
2. What steps do you take so the problem arises?
Waiting for SOLAR > Charging on SOLAR > cut out solar inverter > stopping timer resets and keeps restarting at 600s, never reaches 0, never stops charging

**Expected behavior**
Keep counting down as long as import_current > solar_max_import

dingo35, 2024-04-17:
Thanks for your bugreport, it took me some time to reproduce it. The timer was reset when no new API data arrived between two "ticks" (approx. 2s), so you were right, it had everything to do with your API update frequency.

This should fix it:
[91403995.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15010026/91403995.zip)

demerstraat, 2024-04-17:
wow impressive support response time! 
big thank you!

I can confirm you fix works! (see log)
[log.txt](https://github.com/dingo35/SmartEVSE-3.5/files/15010129/log.txt)

would it also have been a fix with just decreasing the API meter interval to 1second ?

dingo35, 2024-04-17:
Well that would not be fix but a workaround; I didnt notice this since my testbench updates 1/sec ....

demerstraat, 2024-04-17:
:-) glad that it can now be "in alignment" with the allowed 10 second interval for API updates then (y)

demerstraat, 2024-07-10:
noticed this behaviour again in 3.6.5. 
solar stop timer starts counting down but keeps resetting to 600s. 
decreasing API update interval from 3 to 1 seconds lead to intended behaviour (countdown running out 
[log.txt](https://github.com/user-attachments/files/16168115/log.txt)
till 0)

dingo35, 2024-07-10:
Duplicate of #94 ; This is solved in 8c77f1c0c17e9bb0163bce800e13f58de48b5165

demerstraat, 2024-07-10:
Tried the fix https://github.com/dingo35/SmartEVSE-3.5/commit/8c77f1c0c17e9bb0163bce800e13f58de48b5165  but gives the same issue.

In https://github.com/dingo35/SmartEVSE-3.5/issues/94 the issue is immediately stopping, here it is never stopping

dingo35, 2024-07-11:
Thx for your report, this should fix it:
[15aec8ca9bc.zip](https://github.com/user-attachments/files/16174540/15aec8ca9bc.zip)

Could you test?

demerstraat, 2024-07-11:
super!
can confirm it is fixed with this update.
your great work is much appreciated!

closed 2024-07-11 by demerstraat