dingo35/SmartEVSE-3.5 #54: Solar Stop Timer never kicks in and resets itself every 20 seconds when solar generation remains too low

adjego, 2024-04-18:
**Describe the bug**
When using Solar mode, and the solar generation becomes lower than specified, the solar stop timer will kick in (600s). This timer is reset every 20 seconds without a change of generation of solar energy.

**Upload your config**
`{
"version": "v3.5.0",
"mode": "SOLAR",
"mode_id": 2,
"car_connected": true,
"wifi": {
"status": "WL_CONNECTED",
"ssid": "Laadpaal",
"rssi": -65,
"bssid": "18:E8:29:A1:83:85"
},
"evse": {
"temp": 22,
"temp_max": 65,
"connected": true,
"access": true,
"mode": 2,
"loadbl": 0,
"pwm": 102,
"solar_stop_timer": 592,
"state": "Charging",
"state_id": 2,
"error": "None",
"error_id": 0,
"rfid": "Not Installed"
},
"settings": {
"charge_current": 60,
"override_current": 0,
"current_min": 6,
"current_max": 16,
"current_main": 25,
"current_max_circuit": 16,
"current_max_sum_mains": 600,
"solar_max_import": 0,
"solar_start_current": 4,
"solar_stop_time": 10,
"enable_C2": "Solar Off",
"modem": "Not present",
"mains_meter": "API",
"starttime": 0,
"stoptime": 0,
"repeat": 0
},
"mqtt": {
"host": "192.168.1.8",
"port": 1883,
"topic_prefix": "SmartEVSE",
"username": "",
"password_set": false,
"status": "Connected"
},
"home_battery": {
"current": 0,
"last_update": 0
},
"ev_meter": {
"description": "Disabled",
"address": 12,
"import_active_power": 0,
"total_kwh": 0,
"charged_kwh": 0,
"currents": {
"TOTAL": 0,
"L1": 0,
"L2": 0,
[... 101 lines cut ...]
(I t:1420059ms) (printStatus)(C0) L1: 5.1 A L2: 0.0 A L3: 0.1 A Isum: 5.2 A
(V t:1422059ms) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=6.0 A Imeasured=5.2 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=2.
(V t:1422064ms) (CalcBalancedCurrent)(C0) phaseLastUpdate=1713439055.
(V t:1422066ms) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=6.0 A, Idifference=16.0, mod=0.
(V t:1422066ms) (CalcBalancedCurrent)(C0) Checkpoint 3 Isetbalanced=3.4 A, IsumImport=5.3, Isum=5.3, ImportCurrent=0.
(V t:1422067ms) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=6.0 A.
(V t:1422067ms) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=6.0 A.
(I t:1422068ms) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 584 NoCurrent: 0 Imeasured: 5.2 A IsetBalanced: 6.0 A, MainsMeterTimeout=10, EVMeterTimeout=64.
(I t:1422069ms) (printStatus)(C0) L1: 5.2 A L2: 0.0 A L3: 0.1 A Isum: 5.3 A
(V t:1424069ms) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=6.0 A Imeasured=5.2 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=2.
(V t:1424075ms) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=6.0 A, Idifference=16.0, mod=0.
(V t:1424077ms) (CalcBalancedCurrent)(C0) Checkpoint 3 Isetbalanced=6.0 A, IsumImport=5.3, Isum=5.3, ImportCurrent=0.
(D t:1424078ms) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=6.0A, ActiveEVSE=1.
(V t:1424078ms) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=6.0 A.
(V t:1424079ms) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=6.0 A.
(I t:1424079ms) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.2 A IsetBalanced: 6.0 A, MainsMeterTimeout=8, EVMeterTimeout=64.
(I t:1424080ms) (printStatus)(C0) L1: 5.2 A L2: 0.0 A L3: 0.1 A Isum: 5.3 A`

**To Reproduce**
Steps to reproduce the behavior:
1. Solar mode
2. Cloudy weather, but the charging will never stop as the solar stop timer will reset after 20 seconds.

**Expected behavior**
Charging should stop when solar is not available anymore after the timer times out.

hmmbob, 2024-04-18:
Possible duplicate of #53 ?

dingo35, 2024-04-18:
It is...

closed 2024-04-18 by dingo35