dingo35/SmartEVSE-3.5 #58: Feature Request: Show or download logs

devdems, 2024-04-22:
Currently, i did not find any other way to check the logs without enabling the telnet option and read the logs there.

We need to be able to view the logs on the UI or download them. Assuming you have x amount of space dedicated for logs, it would overwrite the old logs if the x is reached. Ideally, we should be able to define the log level on the UI.

dingo35, 2024-04-22:
Why?

devdems, 2024-04-22:
Simple. You are charging the car at night and something happens. You would like to know what happened.

dingo35, 2024-04-22:
So for those few users that are really bughunting: tee your telnet session to a file and rotate that on a daily basis.

Storage is scarce at IOT devices, and we want to keep it for future extensions.

Also this would complicate code and not open up any new  functionality.

Finally, IMHO this kind of info should not be processed on IOT devices; export them to the outside world where they can be manipulated, grepped, spreadsheeted or whatever you would like to do with it!

closed 2024-04-22 by dingo35

devdems, 2024-04-22:
So for example you charge the car at night, it stops charging a couple of times during the night ergo the car is not charged in the morning. What would be your friendly solution and take into consideration an individual without any IT knowledge?

Currently, you cannot see anywhere that the error happened, no notification is sent, nothing. 

In the morning when he checks the device all is fine and dandy because the error from 3AM is not present anymore.

devdems, 2024-04-22:
There are multiple ways to solve this (show logs, send push notification via telegram, pushbullet,..., send email,)

dingo35, 2024-04-22:
If communications are restored, charging is continued. If not you can determine the comms problem...

If you are connected to HomeAssistant you can already monitor all exported variables....

And guess what, in the past two years, not a single night my EV wasnt charged.....so AFAIK its a rather theoretical discussion...

devdems, 2024-04-22:
Ok, it's a nazi regime here. Will not bother with any more items.

@mstegen recommendation. We can all appreciate the @dingo35 what he has done and still doing but I think you should create a repo on your git for 3.5 and have something like a voting pool for new features as I saw a lot of examples of features being closed that were good proposals.

dingo35, 2024-04-22:
Why dont you look at HomeAssistant, it logs all your error and charging states, with a frequency of 6 times / minute?

And/or log with telnet, you are a developer so it would probably cost you less than an hour to write the bash script for that...

Too bad you're going for an ad hominem, instead of reacting to the content of my messages...

dingo35, 2024-04-22:
No there are only two repo's (not counting Serkri since it is 200+ commits behind), the one from the manufacturer and this one.

And the manufacturer will adopt the versions developed here, once proven stable...