dingo35/SmartEVSE-3.5 #59: Change Actuator and Switch Behaviour with delayed charging set.

Ferrox85, 2024-04-22:
**Is your feature request related to a problem? Please describe.**
Not problem related

**Describe the solution you'd like**
If a starting time is set, I'd like when a car is connected (see status EV Connected = YES) that the locking pin is activated. In this way, it is not possible for others to disconnect the cable.

For Example, when charging is set at 1 o'clock in the night (during variable low electric prices) and someone tries to steal the cable at in the evening, it wil disconnect from the charging port. The Car keeps the cable locked till you unlock the car itself.  The Cable won't be stolen, but the car can't charge with the disconnected cable. I Can make a warning when it is disconnected, but I love to sleep and mitigate this possibility.

When the car unlocks while charging. It stops the charging session and then the SMART EVSE releases the cable also (that is already working this way)..

When I want to overrule the set charging time and plugged in car, I pres the switch one time and it wil should charging immediately

When the car unlocks while not charging and pin locked (ahead of planned charging), the switch on the SMART EVSE should unlock with the second press. 

So in short:
If start time set:
  -  Car gets plugged in -> Lock Actuator automatically
 Loop through the following options after plugged in car and connected = yes:
  -                                                         1st pres                                             2nd press                                                      3rd press (return back to original state when the car got plugged in)
  -  Switch/SmartEVSE behavior: {-> Start Charging                            -> Release Cable                                          -> Planned}
  -  Screen 1st line:                        {-> Scheduled                                  -> "charging mode"(Normal/Solar/Smart)  -> Stopped}
  -  Screen 2nd line:                      {-> Start Now                                    -> Unlock                                                       -> Scheduled}
  -  Switch Color:                           {-> blue Flashing (Planned Mode)  -> "charging mode"(Normal/Solar/Smart)  -> off}

I Would love to code this myself, unfortunately I do not have any experience with C and I do not comprehend the code completely to do this myself or make a contribition to this project. Maybe I can be of assistence if somebody can help met through the code?

**Describe alternatives you've considered**

Export the locking actuator directly to the API. So I Can move the actuator with Home Assistant. Not sure this is the best solution. I like low level intelligence on the SMART EVSE for this. I Use the API to set the datetime to charge. The logic for this is already in home assistance and it is illogical to let the hourpices be handled by the SMART EVSE.

dingo35, 2024-04-22:
Thx for your suggestion; I understand the part on locking the actuator when you are delayed charging. This version should do that, please test:
[b938b7afe.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15064344/b938b7afe.zip)
(you do not have to flash spiffs.bin with this alpha 3.6.0 version)

Now on your 1st/2nd/3rd press table, I'm afraid that is uncomprehendable for all users that either have an RFID reader and/or their switch configured for Solar/Smart and /or NO actuator installed.

Now the switch is simply "Access / No Access" or  "Solar / Smart", dependent on how you configure it.
Wouldnt it make more sense to keep the switch as it is and for the more complicated stuff use the webserver screen on your smartphone?

Ferrox85, 2024-04-22:
Hi Dingo,

Thank you. I am not in the opportunity to test this the next 2 weeks, after that I wil test it.

I would use Home Assistant to do that. But my wife isn't fond of that and likes buttons to do that. I'll get that, but can't we add another sequence like those sequences that are already there (a 3rd option for the switch logic)? Set the charging on smart with the web/home assistant (with/without delay) and use the button for the press table I presented? If it isn't possible or logical, no problem at al, I don't know what the options or possibilities are in this case. If you have an other suggestion, please let me know.

With RFID installed it is logical to use the RFID to lock and unlock, so it isn't needed there. If there is no actuator set, then is only the logical choice for the button smart/solar. (or acces / no acces, without actuator)

But with both a actuator and a switch, I think the most logical setting voor the switch is the sequence I mentioned as a plausible option. I know there some settings that are not visible with certain other settings. So that this option only becomes available with a switch and actuator installed?

But anyway, many thanks for the quick update!

Ferrox85, 2024-05-04:
Hi Dingo,

Just tested, when plugged in and setting a starting time afterwards works.

I have still to test the other way around, a time set and then plugging the car in. But I have some programming to do here to make this work.

dingo35, 2024-05-04:
Ok let me know! Cant test myself cause no actuator here....

Ferrox85, 2024-05-05:
The other way around doesn't. It locks right away when the timer is set, Despite no car is connected. So I can't plug the cable in anymore because the actuator is activated. It should lock only when both are true, Car connected and timer set. If one of them isn't true it should wait till both are true and lock the cable.

dingo35, 2024-05-06:
Could you test this:

[1547209e471.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15217183/1547209e471.zip)

RogMoe, 2024-05-06:
My SmartVS sits inside a former Shell Lolo pole and has an socket and RFID. I took its old guts out and installed SmartEVSE as replacement. As soon as I used the RFID, the actuator got activated, regardless. I miss that function now. I usually leave the cable in the socket and take off. It's a quit neighbourhood... But still, is it possible to have a software switch to keep the actuator locked at all times, until unlocked by RFID or app?

dingo35, 2024-05-06:
@rogmoe This is a very specific use case, since most people would have installed a fixed cable (what is effectively what you want). Surely you understand we are not going to add software switches for every single use-case, adding complexity and confusion for less experienced users.

Why don't you just hardwire your actuator to "locked" forever?

RogMoe, 2024-05-06:
...because I sometimes remove the cable if I go away for longer periodes of time since the connector sticks out from the socket and makes it more "bulky" then a thetered cable. It is in a public area after all. The pool comes with a socket. Can't do much about that.
I do understand you're not looking for additional complexity, but can there be a override through API or at least see the satus of the actuator?

If the car is in SOLAR mode and waiting for the sun to come up, it is NOT locked. Anyone can unplug the connector.

Ferrox85, 2024-05-06:
Hmm... it's a really other suggestion than mine. maybe create a separate feature request to discuss this? I like the idea of heaving the actuator state disclosed to the API and GUI and the ability to lock and unlock separately from the SMART EVSE logic. But I don't leave the cable in the socket, so my request here is a little bit different.

RogMoe, 2024-05-06:
I'm sorry to have disrupted your initial discussion.
@dingo35 has read my proposal and responded. Up to him.
Thx!

Ferrox85, 2024-05-06:
> Could you test this:
> 
> [1547209e471.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15217183/1547209e471.zip)

This doesn't do anything unfortunately I reverted to Version: v3.6.0-RC1.

It doen't even lock anymore when a time is set after a cable is plugged in. the other way around still doesn't work.

dingo35, 2024-05-06:
Easiest is to look at the LCD screen, as soon as it replaces --:-- with local time at upper right corner, local time is in...usually takes 30s or so.

Ferrox85, 2024-05-16:
Ok, just tested it.

After waiting it stil doesn't work. The correct time is in the display. The latest release (3.6.0. RC2) doesn't work either.
But it has the debug option. So I tried that. As a result, see attached debug log. First I tried setting a time before plugging in. And reverting to off and setting a time once again. Both do not activate the actuator.

I reverted back again to 3.6.0.RC1.
[Actuator.log](https://github.com/dingo35/SmartEVSE-3.5/files/15330959/Actuator.log)

I hope you have enough info. If more is needed, let me know. (I do not always have time to debug and have a car to charge here at the same time..)

dingo35, 2024-05-22:
Tested at two locations, works here.....

closed 2024-05-22 by dingo35