dingo35/SmartEVSE-3.5 #62: strange "chargedEnergy" glitch

demerstraat, 2024-04-25:
**Describe the bug**
Noticed the "EV charged energy" jumped to the same value as "Total energy charged"

This happened when :
1) EV-1:  EVSE charging session was finished, Charged Energy was reading = 7.5kWh
2) disconnecting the EVSE charging plug from EV-1 and connecting it to EV-2

instead of resetting Charged Energy to "0 kWh", Charged Energy jumped to the value of "Total Energy Charged"

for some reason, during the same charging session of EV-2, "Charged Energy" reset to zero after +- 3h into the session. 
directly afterwards, it jumped back to following "Total Energy Charged"

---
This morning on reconnecting EV-1 to the charger, 
"EV charged energy" reset to zero as expected, and is report correctly the session energy like it should

**Upload your config**
[raw.json](https://github.com/dingo35/SmartEVSE-3.5/files/15106153/raw.json)

**Upload a debug log**
N/A

**To Reproduce**
N/A

**Expected behavior**
A clear and concise description of what you expected to happen; expected Charging Current?

**Screenshots**
[image]
[image]

dingo35, 2024-04-25:
At what frequency are you feeding your EVMeter APIs?

demerstraat, 2024-04-25:
the interval is 3 seconds. 
it is the first time I noticed this over the 4 weeks of using it

dingo35, 2024-04-25:
I suspect it is a glitch that is there immediately going from state B to state C; if that is the problem, this version should solve it:
[592ab0824e.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15107389/592ab0824e.zip)
...please test!

demerstraat, 2024-04-25:
okay, will do, thanks!

this is my MQTT payload for the EV meter: 144:0:0:3286:339536.0

I noticed I send total energy in _float_ and not _int_, possibly related?

dingo35, 2024-04-25:
Well that value is read by a sscanf:

int n = sscanf(payload.c_str(), "%d:%d:%d:%d:%d", &L1, &L2, &L3, &W, &WH);

and it is read as a decimal unsigned integer; it  obviously accepts your values (n=5 is tested) so I don't think that is a problem...

demerstraat, 2024-04-25:
> I suspect it is a glitch that is there immediately going from state B to state C; if that is the problem, this version should solve it: [592ab0824e.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15107389/592ab0824e.zip) ...please test!

quick feedback: 
EV was charging, I did the update during charging, so obviously EVSE rebooted afterwards, 
EV resumed charging and EnergyCharged again followed TotalEnergyCharged.
 
stopped EV charging session, rebooted EVSE from webpage, after reboot EnergyCharged again working as expected, from 0.

demerstraat, 2024-04-26:
will close this issue as solved, as I'm not able to reproduce the behaviour in this version
https://github.com/dingo35/SmartEVSE-3.5/files/15107389/592ab0824e.zip

closed 2024-04-26 by demerstraat