dingo35/SmartEVSE-3.5 #63: Trying to compile current version on W10 / VS Code

fluppie, 2024-04-25:
Trying to compile here, when `extra_scripts = pre:packfs.py` is enabled, it doesn't even start compiling. When commenting out it almost finishes:
```

|-- Update @ 2.0.0
|-- WiFi @ 2.0.0
|-- RemoteDebug @ 3.0.5
Building in release mode
Linking .pio\build\release\firmware.elf
c:/users/philippe/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\release\src\mongoose.c.o:(.literal.packed_open+0x0): undefined reference to `mg_unpack'
c:/users/philippe/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\release\src\mongoose.c.o:(.literal.packed_stat+0x0): undefined reference to `mg_unlist'
c:/users/philippe/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\release\src\mongoose.c.o: in function `packed_open':
C:\Users\Philippe\Documents\GitHub\SmartEVSE-3.5_ORG\SmartEVSE-3/src/mongoose.c:1893: undefined reference to `mg_unpack'
c:/users/philippe/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\release\src\mongoose.c.o: in function `packed_stat':
C:\Users\Philippe\Documents\GitHub\SmartEVSE-3.5_ORG\SmartEVSE-3/src/mongoose.c:1863: undefined reference to `mg_unpack'
c:/users/philippe/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Philippe\Documents\GitHub\SmartEVSE-3.5_ORG\SmartEVSE-3/src/mongoose.c:1865: undefined reference to `mg_unlist'   
c:/users/philippe/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\release\src\mongoose.c.o: in function `packed_list':
C:\Users\Philippe\Documents\GitHub\SmartEVSE-3.5_ORG\SmartEVSE-3/src/mongoose.c:1877: undefined reference to `mg_unlist'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\release\firmware.elf] Error 1
======================================================== [FAILED] Took 22.20 seconds ========================================================
```

When re-enabling packfs.py I get
```
Processing release (platform: espressif32 @ 6.6.0; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
'rm' is not recognized as an internal or external command,
operable program or batch file.
A subdirectory or file pack.tmp already exists.
'cc' is not recognized as an internal or external command,
operable program or batch file.
========================================================= [FAILED] Took 2.10 seconds =========================================================
```

I guess if I would have had a Linux system, it would compile without any issue?

dingo35, 2024-04-26:
Correct, we're working on this!

dingo35, 2024-04-26:
@fluppie Could you try to build current master branch on Win10? I had an old Win8.1 version installed but wound up in a maze of policies and restrictions, which reminded me why I love Linux so much :-)

fluppie, 2024-04-26:
Nope, not yet :-)
```
 *  Executing task in folder SmartEVSE-3: C:\Users\Philippe\.platformio\penv\Scripts\platformio.exe run --environment release 

Processing release (platform: espressif32 @ 6.6.0; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
'rm' is not recognized as an internal or external command,
operable program or batch file.
A subdirectory or file pack.tmp already exists.
'cp' is not recognized as an internal or external command,
operable program or batch file.
========================================================= [FAILED] Took 2.88 seconds =========================================================
 *  The terminal process "C:\Users\Philippe\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'release'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

dingo36, 2024-04-26:
@fluppie same player shoots again...

dingo36, 2024-04-28:
Solved since 342467eb3

dingo35, 2024-04-28:
Solved since 342467eb3

closed 2024-04-28 by dingo35

fluppie, 2024-04-28:
Works :)