dingo35/SmartEVSE-3.5 #64: CP duty cycle calibration/offset

demerstraat, 2024-05-02:
**Describe the bug**

noticing consistent 5-6% offset between "CP current setpoint" as sent by EVSE via CP vs. actual current drawn by EV OBC. 
e.g. setpoint to charge = 16A, actual current drawn = 15.3A. 

the same offset is present whether in SOLAR / SMART / NORMAL mode

**also noticed that "MAINS L1" seems to be within 0.1A of "SET CURRENT". 
is it possible that PWM correction is evaluated against "MAINS L1" instead of against "EV L1" ?** 

**Upload your config**
[raw.json](https://github.com/dingo35/SmartEVSE-3.5/files/15191724/raw.json)

**Upload a debug log**

**To Reproduce**
see above: always present 

**Expected behavior**
actual current drawn by EV ~ EVSE current setpoint 

**Screenshots**
[image]

dingo35, 2024-05-02:
Most EV's draw only 95% of the allowed current.

https://gathering.tweakers.net/forum/list_message/77537224#77537224

closed 2024-05-02 by dingo35

demerstraat, 2024-05-20:
[image]
offset of "10" in PMW duty did the trick to get 100% of "set value"
made pull request