dingo35/SmartEVSE-3.5 #66: v3.6.0-RC1 MQTT is not connection when entering a host name.

Imaginous, 2024-05-03:
**Describe the bug**
In v3.6.0-RC1 MQTT is not connection when entering a host name.
It works with an IP address

**Upload your config**
`{"version":"v3.6.0-RC1","serialnr":6972,"mode":"OFF","mode_id":0,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"Imaginous","rssi":-54,"bssid":"xx:xx:xx:xx:xx:xx"},"evse":{"temp":26,"temp_max":55,"connected":false,"access":false,"mode":1,"loadbl":0,"pwm":1024,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":26,"current_main":33,"current_max_circuit":32,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":8,"solar_stop_time":30,"enable_C2":"Always Off","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"mqtt.imaginous.nl","port":1883,"topic_prefix":"SmartEVSE","username":"mqtt","password_set":true,"status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":12,"L1":12,"L2":0,"L3":0,"last_data_update":1714725257,"original_data":{"TOTAL":12,"L1":12,"L2":0,"L3":0}},"backlight":{"timer":0,"status":"OFF"}}`

**To Reproduce**
Steps to reproduce the behavior:
Enter a host name for the MQTT server - it fails
Enter a IP address for the MQTT server - it works

**Expected behavior**
That a host name also works like in the previous (v3.5.x) versions.

dingo35, 2024-05-03:
Is the hostname you are using by chance a local one?

Mongoose seems to have 8.8.8.8 hardcoded as dns server, so that would explain why it cannot be resolved..

Imaginous, 2024-05-03:
Yes it's a local IP address, being served by my OPNsense router.
DHCP server provides the DNS server address.

dingo35, 2024-05-03:
Fixed in 01cb6f10905e

closed 2024-05-03 by dingo35