dingo35/SmartEVSE-3.5 #68: Configure Sensorbox as an EVMeter

chylek, 2024-05-05:
**Is your feature request related to a problem? Please describe.**
Hi, I have a Sensorbox 2. Thanks to your and serkri's amazing efforts I can use the API to send MainMeter data directly from a different meter that I already have in my Home assistant (saving me a hassle of digging my yard up due to a missing wire).

But I don't want to throw the Sensorbox in trash - is it possible to use it as an EVMeter?

**Describe the solution you'd like**
Allow selecting the Sensorbox as an EVMeter.

**Describe alternatives you've considered**
Propagate messages from the Sensorbox (even when not configured as any meter) to MQTT or REST API. It would then be possible to relay them back using the EVMeter API.
E.g. SmartEVSE gets Sensorbox data -> sends them to MQTT SmartEVSE-XXXX/Sensorbox -> Home Assistant receives the messages -> Home Assistant sends MQTT message to SmartEVSE-XXXX/Set/EVMeter

**Additional context**
N/A

dingo35, 2024-05-07:
No that wont work; the sensorbox delivers L1-L3 currents, which is useful for MainsMeter.

The main info taken off of the EVmeter is Power and Energy, which is not delivered by Sensorbox...

So I suggest you try to make someone else happy with the sensorbox...

chylek, 2024-05-07:
Oh, right, I didn't realize that. That's a shame. 

Although sending the current using MQTT (the alternative approach) would still be a useful info, since I can retrieve the voltage from other meters and compute the power that way. 

But that would be a lot of work on something that would end up being an estimate of the power. Buying a new meter would be a cheaper and better option 🙂

closed 2024-05-07 by chylek