dingo35/SmartEVSE-3.5 #71: V3.6.1-RC1 Constant MQTT updates / cycling through State and Error

vdweegen, 2024-05-07:
**Describe the bug**
When using version v.3.6.1-RC1 I'm getting constant MQTT updates for both my SmartEVSE's (Power Share). Both the fields "State" and "Error" keep cycling through various values.

As I'm using some custom code to track changes in value and notify me via Pushover.

**Upload your config**
Press the "raw" button in the SmartEVSE webserver and upload the json settings file, so we can look at your configuration.

`{"version":"v3.5.1","serialnr":6678,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"WEEBAAHOME","rssi":-60,"bssid":"74:AC:B9:94:33:C5"},"evse":{"temp":32,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":1,"pwm":102,"solar_stop_timer":0,"state":"Connected to EV","state_id":1,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"solar_max_import":2,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Not present","modem":"Not present","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"10.12.40.32","port":1883,"topic_prefix":"SmartEVSE-15494","username":"","password_set":false,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":101,"import_active_power":0,"total_kwh":1123.199951,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":1123.199951,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":39,"L1":22,"L2":3,"L3":14,"last_data_update":1715107357,"original_data":{"TOTAL":39,"L1":22,"L2":3,"L3":14}},"backlight":{"timer":69,"status":"ON"}}`

`{"version":"v3.5.1","serialnr":7130,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"WEEBAAHOME","rssi":-66,"bssid":"74:AC:B9:94:33:C5"},"evse":{"temp":36,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":2,"pwm":221,"solar_stop_timer":0,"state":"Connected to EV","state_id":1,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":130,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"solar_max_import":2,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Not present","modem":"Not present","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"10.12.40.32","port":1883,"topic_prefix":"SmartEVSE-46809","username":"","password_set":false,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":110,"import_active_power":0,"total_kwh":225.3999939,"charged_kwh":225.3999939,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":225.3999939,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":34,"L1":17,"L2":3,"L3":14,"last_data_update":1715107799,"original_data":{"TOTAL":34,"L1":17,"L2":3,"L3":14}},"backlight":{"timer":0,"status":"OFF"}}`

**Upload a debug log**
Forgot to take the bug report before reverting to v3.5.1. Will provide bug report if needed.

**To Reproduce**
Steps to reproduce the behavior:
In SMART mode (power share, two SmartEVSE's on one 16A three-fase breaker), using a Sensorbox. The SmartEVSE's immediately start updating over MQTT when updating to v.3.6.1

**Expected behavior**
The SmartEVSE's don't continuously update the State and Error Values.

**Screenshots**
[image]

dingo35, 2024-05-08:
Ok I could reproduce this, this should solve your problem:
[71c853c67a.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15246200/71c853c67a.zip)

Please test!

dingo35, 2024-05-08:
I think it solves your problem:

[4886937903a.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15247888/4886937903a.zip)

vdweegen, 2024-05-08:
Just flashed and your fix seems to have done the trick. Will test coming days.

`{"version":"12:42:19 @May  8 2024","serialnr":6678,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"WEEBAAHOME","rssi":-65,"bssid":"74:AC:B9:94:33:C5"},"evse":{"temp":29,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":1,"pwm":136,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":80,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"solar_max_import":2,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Not present","modem":"Not present","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"10.12.40.32","port":1883,"topic_prefix":"SmartEVSE-15494","username":"","password_set":false,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":101,"import_active_power":5,"total_kwh":1124.099976,"charged_kwh":0.300000012,"currents":{"TOTAL":229,"L1":79,"L2":76,"L3":74},"import_active_energy":1124.099976,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":260,"L1":119,"L2":75,"L3":66,"last_data_update":1715187711,"original_data":{"TOTAL":260,"L1":119,"L2":75,"L3":66}},"backlight":{"timer":120,"status":"ON"}}`

`{"version":"12:42:19 @May  8 2024","serialnr":7130,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"WEEBAAHOME","rssi":-63,"bssid":"74:AC:B9:94:33:C5"},"evse":{"temp":33,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":2,"pwm":136,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":80,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"solar_max_import":2,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Not present","modem":"Not present","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"10.12.40.32","port":1883,"topic_prefix":"SmartEVSE-46809","username":"","password_set":false,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":110,"import_active_power":0.800000012,"total_kwh":229.3999939,"charged_kwh":229.3999939,"currents":{"TOTAL":36,"L1":36,"L2":0,"L3":0},"import_active_energy":229.3999939,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":264,"L1":123,"L2":75,"L3":66,"last_data_update":1715187703,"original_data":{"TOTAL":264,"L1":123,"L2":75,"L3":66}},"backlight":{"timer":120,"status":"ON"}}`

dingo35, 2024-05-09:
I'm pretty sure 71c853c67a0 solved it, if not please reopen this issue.

closed 2024-05-09 by dingo35