dingo35/SmartEVSE-3.5 #72: Socket lock in API

RogMoe, 2024-05-09:
**Is your feature request related to a problem? Please describe.**
Unattended cable can get unplugged

**Describe the solution you'd like**
Have access to the lock through API and get feedback on the lock status, possibly even have an override button on the internal webpage.

**Describe alternatives you've considered**
none

**Additional context**
If the car is connected in solarmode and waiting to charge, the cable is not locked. Some people have their EVSE located in a public spot, accessable by others. Having control over the lock would be a great feature.

dingo35, 2024-05-09:
I understand your wish, but Im hesitant to develop stuff I cannot test.
Did you test the version in #59 ?

RogMoe, 2024-05-09:
1547209e471.zip you mean?
Nope because didn;t want to interfere with someone else's request.
I don't use a timer-start.
I'm willing to test anything within 24hr if thats any help.

dingo35, 2024-05-09:
Ok so with:
"If the car is connected in solarmode and waiting to charge, the cable is not locked." you didnt mean you were delayed charging?

RogMoe, 2024-05-09:
Yes, didn't mean "delayed charging". I never used delayd charging, delayed as in using the timer function.
Waiting for the sun to come up is "delayed charging" too in a way ;-) I understand the misunderstanding.

You added a function that locks the cable while waiting for the timed charging to begin. I ask you to extend that function to SOLAR mode too.

fluppie, 2024-05-20:
BTW not 100% related, but during some test I hear the EVSE's sending a open actuator to the motors when nothing is plugged in. Is that needed? Or is no one aware of this :)?

dingo35, 2024-05-21:
I noficed this too a few days ago, I'll upload the fix.

dingo35, 2024-06-21:
Different actuator behaviour since v.3.6.1

closed 2024-06-21 by dingo35