dingo35/SmartEVSE-3.5 #73: Missing Voltage and Power measurements missing

odem881, 2024-05-11:
I cannot see voltage and power measurements from mains meter in MQTT nor settings page in UI. We need this to further process them in HA and other tools.

[image]

dingo35, 2024-05-12:
We only need the currents of the MainsMeter to guard them; Sensorbox only gives those. Power is only needed for EVMeter.

The modbus capacity prevents us from reading all registers; we only have 9600 baud, and if ww crank it up to 19200 not all meters support it.

So you either guess U and calculate P, or move your modbus to HA and upload the values to the SmartEVSE.

closed 2024-05-12 by dingo35

devdems, 2024-05-12:
@odem881, @dingo35 is wrong here. I will share you the update for this in couple of days. Power and Voltage is already available in the code, just need to add them to the outputs. I'm getting both those walues from the mainsmeter to HA via MQTT.

dingo35, 2024-05-12:
If it doesnt generate any more modbus traffic, and is valid for all configurable meters, Id be happy to add your code!

devdems, 2024-05-12:
@odem881 here you go  
[firmware.zip](https://github.com/dingo35/SmartEVSE-3.5/files/15287105/firmware.zip)

https://github.com/dingo35/SmartEVSE-3.5/pull/75

odem881, 2024-05-12:
Working like a charm! Thank you. Now i can keep wired connection between mains and the charger!

dingo35, 2024-05-13:
Yes this is exactly what I thought: the data is NOT present in the current code, you are adding two extra ModbusReadInputRequests in the modbus-loop; and you also put them right next to eachother.

The problem with modbus requests is that when you receive a response, it is not identifying which request it came from; you can see from which device the response is coming, and which function code was used, but NOT if this is an answer from the power measurement request, the current measurement request, or the voltage measurement request.
There is even a "matching" mechanism where we store the address, function and register upon request so we can match it on response, but it fails regularly because we miss the register in the response...

In the past we had problems with this:
#62 , #11, and it goes back to the Serkri versions:
https://github.com/serkri/SmartEVSE-3/issues/247, https://github.com/serkri/SmartEVSE-3/issues/35 .

If you only have a MainsMeter and/or an EVMeter connected to your modbus, this might work out withouth any problems. But if you have multiple SmartEVSE's connected, and sometimes lots of other devices ("christmas tree configuration") people will run into communication errors on the modbus.

Because of this we will not add this to the regular code; luckily you can keep your own modifications locally, the advantage of open source.
@devdems: your code looks fine, but I would recommend spreading the MainsMeter request calls instead of concentrating them in modbusrequest 22 and 23.

devdems, 2024-05-13:
@dingo35 let's forget the EVMeter as it is not important for this feature. Here we are connecting to the mains meter and you always have one mains meter. Even if you have multiple SmartEVSE's only master will connect to the mains meter.

With this bottom code I know if the response is Voltage or Power so I'm not sure what you meant by "..but NOT if this is an answer from the power measurement request, the current measurement request, or the voltage measurement request."  

`else if (MB.Register == EMConfig[MainsMeter].**PRegister**) {
                     MainsPower = receivePowerMeasurement(MB.Data, MainsMeter);
        } else if (MB.Register == EMConfig[MainsMeter].**URegister**) {
                     MainsVoltage = receiveVoltageMeasurement(MB.Data, MainsMeter);
        }`

I'm running this config with master + slave + mains meter for almost a month without any issues and do not see any drops in the inputs of voltage and power being sent to HA.

dingo35, 2024-05-13:
You have to look back in the code, in src/modbus.cpp, and you will see that the Register values are stored from the last request placed. So if you request a power measurement , followed by a voltage measurement, and then the power response comes in, that response will be filled in as a voltage value (so now you have 11.000 V reported on your home installation).

Some meters are responding fast enough so this problem doesnt arise, but some dont. 
Our experience with e.g. Eastron is that even meters with the same model names, but different firmware, respond differently to this problem....

So praise yourself lucky that your meters are fast enough!

Have a nice day.

devdems, 2024-05-20:
@dingo35 I see what you mean as I can see dropouts here and there. Looking more into this and it looks like the way that Modbus is implemented here is quite bad. If you check other solutions on Git you can find much better libraries used for this integration. For example, they are pulling multiple values from Modbus in one go and not like here where we are requesting one by one and hoping that the correct response is picked up.

dingo35, 2024-05-20:
Yep that is exactly on my list, but quite low..

fluppie, 2024-05-20:
I think both Tasmota and ESP Easy are using the Reaper library. One issue/pull request talks about multiple reg at once. Guy is doing 1s interval reading with an SDM630. Might help us? https://github.com/reaper7/SDM_Energy_Meter/pull/82