dingo35/SmartEVSE-3.5 #79: Web server not responding anymore in V3.6.0

RonaldSpijk, 2024-05-24:
I've uploaded version 3.6.0 on a newly bought SmartEVSE. Before that it had 3.5.1 for a few days.
It worked for a few days without any problems.
 
After some issues with the Wifi the web server stopped working. MQTT still works.
I can also ping.
When accessing port 80 the port is found, but the connection is dropped immediately.

A tried resetting a few times, doesn't help.
A tried starting the Wifi Portal, this also doesn't work.

Fortunately, charging still works :-)

I'm stuck in the middle here, because when the web server doesn't work, I can't update,  and get logs.
I'm afraid to do anything else, because it might brick my SmartEVSE.

I think this needs good investigation because it might also occur at other users.
And I like some help getting a working version on the SmartEVSE. Would uploading the bin file with the Espressif flash download tool work? Or should I try something else first.

AndrejSNC, 2024-05-25:
What I have discovered today is, I replaced my old SmartEVSE version with V3 controller. It came with 3.0.1 SW version. I connected it to WiFi and set fixed IP address - works great. 
Then I updated it to 3.6.0 it connected back to the old dynamic IP address, which I don't like. I can start WiFi setup in menu, connect to access point but the configuration page does not open on my phone. How can I access this configuration, to set back static IP? Tried 192.168.4.1 and it did not connect.

dingo35, 2024-05-25:
If you set "wifi" from "Enabled" to "Setup" on the LCD screen, you should be able to connect to the SmartEVSE AP via 192.168.4.1 and set a static IP.

If it for some reason that fails, you can always goto /erasesettings and repeat the wifi setup procedure.

deekoowee, 2024-05-29:
Similar problem here. Controlled worked until I upgrade it to 3.6.0
Controller answer for ping at 192.168.4.1, but do not answer to browser. Tried with Chrome and Curl
Also tried the /erasesettings, but no luck.
I tried to flash the firmware with **ESP32 flash download tool v3.9.6** , but nothing changed
Connected serial cable and telnet shows;
[image]
Any suggestions to fix the controller's firmware

dingo35, 2024-05-29:
If you have these problems with 3.6.0, why dont you downgrade to 3.5.1 with USB-C cable? Link is posted in this thread....

deekoowee, 2024-05-30:
Downgraded to 3.5.1 via USB-C cable, rebooted controller, the web page showup.
Showed version 3.6.0 at left top corner of the screen
After few minutes, the webserver has lost, no communication with any browser.
Some mystery, which I don't catch up.

dingo35, 2024-05-30:
Your downgrade obviously did not succeed...

deekoowee, 2024-05-30:
I tried to upload the firmware.bin file with ESP32 flash download tool v3.9.6 (downloaded from [Espressif site](https://www.espressif.com/en/support/download/other-tools))
I used settings like this, only filename was changed.
[image]
the bar at bottom on window runs ok.

mstegen, 2024-05-30:
There are two partitions, the first one starts at 0x10000 
The second one starts at 0x1c0000 
Try flashing both partitions. That should do the trick.

deekoowee, 2024-05-30:
Flashing those two partitions together fixed the webaccess problem

All settings were gone and have to reconfigure the controller

dingo35, 2024-05-30:
The settings were gone because you did /erasesettings, not because of reflashing those two partitions...

deekoowee, 2024-05-31:
Yes, I noticed that.
It is not big deal to make new settings again

dingo35, 2024-05-31:
This should solve the problem:
[83adad145a90903.zip](https://github.com/user-attachments/files/15512629/83adad145a90903.zip)

AlbertHakvoort, 2024-06-05:
> This should solve the problem: [83adad145a90903.zip](https://github.com/user-attachments/files/15512629/83adad145a90903.zip)

I've had the same issue, the 83adad145a90903.zip is fixed the webserver issue

dingo35, 2024-06-05:
Thanks for your reports guys, this should be solved in 3.6.1 .

closed 2024-06-05 by dingo35

Wireheadbe, 2024-11-17:
Mine "bricked" when updating from 3.7.0 to 3.7.2.
Here's how I flashed in a Debian-based distribution:

First, check where the device has popped up.. from experience, it's quite easy to find the last connected device by just looking into the kernel log, and looking in the last couple of lines:
```
sudo dmesg -T
```

Then, install usbtool, and flash the firmware:
```
sudo apt install esptool
esptool --port /dev/ttyUSB0 write_flash 0x10000 firmware.bin
esptool --port /dev/ttyUSB0 write_flash 0x1c0000 firmware.bin
```

It'll take some time to flash to both partitions, so don't stress :-)