dingo35/SmartEVSE-3.5 #83: Diode check behavior

matth-x, 2024-06-01:
Hi, I'm getting more into details with this project, great work!

The charger checks the diode on the EV side in the beginning of the charging session but doesn't monitor it during charging which doesn't feel quite intuitive to me. Is this actually the standard-conform behavior (or the common practice) or is it more a device-specific behavior?

It's not super important to OCPP, just stumbled over it because the diode in my EVSE tester made problems.

mstegen, 2024-06-02:
Hi Matthias, 

Good question, i had a look at the specifications again.

According to the J1772 spec, after the EV has closed S2, switching from State B to State C. The EVSE should check the presence of the diode:

f. The EVSE determines that the EV/PHEV pilot control circuitry is correctly configured by verifying the presence of the
diode. The low side of the pilot pulse must be within the range specified in Table 4.

And the IEC 61851-1 standard:

The EV supply equipment shall check PWM signal low state of -12V, diode presence, at least once before the closing of the supply switching device on the EV supply equipment.

So it's according to spec.

matth-x, 2024-06-02:
Perfect, thank you for the clarification!

closed 2024-06-02 by matth-x