dingo35/SmartEVSE-3.5 #84: Cannot control slave seperately from master (and other way around) in V3.6.1

djoenez, 2024-06-03:
**Describe the bug**
Earlier i was able to first charge the slave, then the master (or the other way around). Now if i switch smart on the master and off on the slave the slave will 'magically" go to smart. same with solar mode. Especially in solar mode it is handy, i much rather have one car charge fast (less loss) then two cars at minimum charge speed.

Reverted back to 3.6.0 and it was ok again.

dingo35, 2024-07-04:
The master and the slave should always be in the same mode, since the master calculates the currents for the slaves. In recent versions we fixed so that users could no longer misconfigure ....

So this is intended behaviour. The only problem (in v.3.6.3) could be that if you put your master in OFF mode the slave shouldn't goin OFF, and vice versa, but I don't think that is your problem?

djoenez, 2024-07-04:
Hi Dingo, 

ah, i didn't write it up well enough then. I have my master (and slave) both in solar, but for some reason when i set my slave in off, it keeps setting it to solar (within 1-5 minutes), so it is exactly as you describe!

dingo35, 2024-07-04:
Sorry you have to get more in detail, because I can't reproduce this.
How do you put the slave in OFF mode, by LCD, by web, by MQTT, by REST API?

And can you post /settings of both master and slave?

djoenez, 2024-07-11:
Sorry for my late reply, in the meantime i updated both of them to 3.6.5 to see if this resolves the problem. unfortunately it did not:

The situation is as follows:
I switch the master to solar (all communication via MQTT for EVSE master and slave). and put the slave into off (acces denied mode). After x minutes (depends on ... I did not find a pattern here). The slave EVSE has switched to acces allowed (there are no automation in my HA that govern either slave mode or acces, so no outside signal is send to the slave). Which is not what i want (i want to first charge EVSE1, and then EVSE2), now they start both charging at minimum power, which increases power loss during charging and also the EVSE2 switches off more (since there is not enough power in the morning/evening/clouds), which my skoda does not like (after 4x on/off i have to replug the car, else the car won't charge).

EVSE1(master):
{"version":"v3.6.5","serialnr":5277,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Oud-Zuid intelligence #1","rssi":-44,"bssid":"84:D8:1B:E7:B6:FE"},"evse":{"temp":42,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":1,"pwm":282,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":166,"override_current":0,"current_min":6,"current_max":18,"current_main":18,"current_max_circuit":18,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":8,"solar_stop_time":15,"enable_C2":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.1.80","port":1883,"topic_prefix":"EVSE1","username":"XXXXX","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":101,"import_active_power":3.599999905,"total_kwh":3402.600098,"charged_kwh":10.30000019,"currents":{"TOTAL":151,"L1":151,"L2":0,"L3":0},"import_active_energy":3402.600098,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":-66,"L1":4,"L2":131,"L3":-201,"last_data_update":1720693888,"original_data":{"TOTAL":-66,"L1":4,"L2":131,"L3":-201}},"backlight":{"timer":120,"status":"ON"}}

EVSE2 (slave):

{"version":"v3.6.5","serialnr":6175,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Oud-Zuid intelligence #1","rssi":-43,"bssid":"84:D8:1B:E7:B6:FE"},"evse":{"temp":41,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":2,"pwm":102,"solar_stop_timer":0,"state":"Connected to EV","state_id":1,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":17,"current_main":18,"current_max_circuit":18,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":8,"solar_stop_time":15,"enable_C2":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.1.80","port":1883,"topic_prefix":"EVSE2","username":"XXXXX","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron1P","address":102,"import_active_power":0,"total_kwh":1148.599976,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":1148.599976,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":-45,"L1":4,"L2":151,"L3":-200,"last_data_update":0,"original_data":{"TOTAL":0,"L1":0,"L2":0,"L3":0}},"backlight":{"timer":60,"status":"ON"}}

dingo35, 2024-07-11:
I think I have a better solution for you. Mind you, this is only alpha version, but you are one of our advanced users:
[4db803fa50.zip](https://github.com/user-attachments/files/16176423/4db803fa50.zip)

With this version, if you select "PWR SHARE = Master", you get a new option:
[image]

I think if you select NodeNr as priority strategy, this is exactly what you want?

djoenez, 2024-07-11:
Thanks for the fast response! And i've never been called an advanced user on Git, so that is a first for me :).

Yes, i'll try it tomorrow (when there is sun and 2 EV's are in need of a charge, which with today is not a guarantee). ill report back asap.

Other question is: I like the way you are thinking about this, and that you can set it in different modes, my request would be to add one more mode; Which would be: Do nothing. I like my appliances to be as dumb as possible, so that i can have all my logic within Home assistant. so the EVSEs should just work and react if a slave starts/stops charging. but they themselves do not control the off/on modi.

dingo35, 2024-07-11:
As to the skoda not liking the on/off switching, if it is an Enyaq try setting the automatic unlock to off  (so it stays locked all the time); in my experience the problem then vanishes...

djoenez, 2024-07-11:
Ah, Unfortunately it is a citigo! there are no settings there unfortunately. Previously i just upped the wait time (so with cloud cover it had enough time to have the cloud move away).

in 2 weeks we'll move the main car from a hyundai ioniq to a kia EV6, i hope that one will behave as well as the ioniq with the EVSEs.

dingo35, 2024-07-12:
It happens when you are charging on the master, and a cloud comes by so the stop-timer starts running. Somehow this awakens the Slave out of the Off mode....

dingo35, 2024-07-16:
Ha ha v3.6.0 has the problem too, you just didn't notice ....

djoenez, 2024-07-16:
Hi Dingo! Correct, although it is not correct that the master pulls the slave out of off mode only when the solar start timer starts counting, i've also seen it on non-cloudy days. 

This afternoon i hopefully will be home on time to test the alpha version SW and have 2 cars that need charging and sun. keep you posted.

djoenez, 2024-07-16:
On your comment on removing the modbus and power share. I don't trust home assistant enough to be fast enough to not have them both charging and full speed for some time, this is for me a safety issue that i much like to leave local. Let's see how the new first master then slave version works :) Thanks alot!

dingo35, 2024-07-16:
You are absolutely right on that safety issue, just trying to prove that dogma's as "keeping your devices as dumb as possible" are not smart; you should always judge on case-to-case basis....

djoenez, 2024-07-17:
No sun unfortunately. I'm still on the latest official 3.6 and I just noticed that when I have the slave in off and put the master in smart the slave will also go into smart mode. Is this a new feature? (I have an automation that checks what the car battery level is and if it is enough for the drive tomorrow, lately do the sun this was not triggered but now the car on the master was low, so it charged to the right level. But since the slave was now also put in smart mode, and when the master was switched to off the slave was not set back to off, so it kept on charging the whole night).

Is it possible to have the slave follow the mode, but keep the access of the slave untouched? (so when switching mode the access is not automatically granted).

dingo35, 2024-07-17:
Im looking into that, but it is not simple...

djoenez, 2024-07-17:
Check. Before this was possible (I don't know which version it was). I do recognize that under the hood alot of things changed (for the better!). Thanks for your work and effort for this project.

djoenez, 2024-07-22:
Good timing, i just installed your previous version, there the slave was still pulled out of off mode. will test this version. To be continued!

djoenez, 2024-07-23:
Yesterday it worked (master in solar, slave stayed in access denied). I'll try today (hopefully some more sun!) the other way around (so slave in solar, master in access denied mode).

djoenez, 2024-07-24:
Also works the other way around. I have control again which car gets to charge. Thanks alot!

dingo35, 2024-07-26:
Fixed in fc0ba6c264

closed 2024-07-26 by dingo35