dingo35/SmartEVSE-3.5 #85: Not switching to 3p in solar

hp197, 2024-06-05:
**Describe the bug**
In solar mode with C2 on auto, the smartevse is not upgrading to 3p charging anymore.
The display on the smartevse says it is in 3f mode (this is set by Force_Single_Phase_Charging?) but its only charging on 1 phase.

What I think is happening:
- On start the charger sets: `Switching_To_Single_Phase = GOING_TO_SWITCH;` in the CalcBalancedCurrent function.
- In the setState function the Switching_To_Single_Phase is handled and set afterwards to `AFTER_SWITCH`.
- The `AFTER_SWITCH` is never reset back to a neutral state, what causes this to never happen anymore:
```
if (!Force_Single_Phase_Charging() && Switching_To_Single_Phase != AFTER_SWITCH) {  // in AUTO mode we start with 3phases
    CONTACTOR2_ON; // Contactor2 ON
}
```
- This also goes wrong in the IsCurrentAvailable function:
```
bool must_be_single_phase_charging = (EnableC2 == ALWAYS_OFF || (Mode == MODE_SOLAR && EnableC2 == SOLAR_OFF) ||
            (Mode == MODE_SOLAR && EnableC2 == AUTO && Switching_To_Single_Phase == AFTER_SWITCH));
int Phases = must_be_single_phase_charging ? 1 : 3;
```

*** Note:
I use EM_API as ingestion into smartevse.
So my main and ev meter can be updated async (what causes the main to report charging be not yet on the ev and the other way around and might fool the phase detection function a bit (I also know the whole phase detection function is under review)

**Upload your config**
```json
{"version":"v3.5.1","serialnr":0,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"lacasita-public","rssi":-52,"bssid":"XX"},"evse":{"temp":35,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":0,"pwm":272,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":160,"override_current":0,"current_min":6,"current_max":16,"current_main":32,"current_max_circuit":48,"current_max_sum_mains":96,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":10,"enable_C2":"Auto","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"lacasitaha.viot.lacasita.watkijkje.nl","port":1883,"topic_prefix":"SmartEVSE-39644","username":"m2me","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"API","address":120,"import_active_power":3.599999905,"total_kwh":2004.400024,"charged_kwh":0.300000012,"currents":{"TOTAL":154,"L1":0,"L2":154,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":-125,"L1":-111,"L2":62,"L3":-76,"last_data_update":1717594010,"original_data":{"TOTAL":-125,"L1":-111,"L2":62,"L3":-76}},"backlight":{"timer":120,"status":"ON"}}
```

**Upload a debug log**
-- can do later if needed

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar?
2. What steps do you take so the problem arises?

hp197, 2024-06-05:
In my own code I fixed it by injecting `Switching_To_Single_Phase = FALSE;` on line 895 (in the IsCurrentAvailable function):

```diff
}

+Switching_To_Single_Phase = FALSE;
 _LOG_D("Current available checkpoint D. ActiveEVSE increased by one=%i, TotalCurrent=%.1fA, StartCurrent=%iA, Isum=%.1fA, ImportCurrent=%iA.\n", ActiveEVSE, (float) TotalCurrent/10, StartCurrent, (float)Isum/10, ImportCurrent);
return 1;
```

dingo35, 2024-06-05:
The upscaling from 1P to 3P has, AFAIK, never been implemented.

There was a PR from a user who was implementing this, but this PR was pulled back. It is a complex part of the software.

For the time being I suggest you use "Solar Off", "Always On" or "Always_Off", or, if that is too static for you, switch those settings via HomeAssistant.

hp197, 2024-06-05:
awh, I might be confused with the serkri fork then....
That explains why I couldn't find a commit where this was changed.

I'll close the bug for now.

dingo35, 2024-06-05:
No this repo is the successor of the serkri distribution, it was never implemented there too. I should know .... :-)