dingo35/SmartEVSE-3.5 #91: SmartEVSE freeze/stuck, since sending home battery information.

vicom147, 2024-06-27:
What happens:
Since a few days we have an Home battery, with Homeassistant I send the current info with the curl to the master EVSE every 10 seconds. But the master freezes/get stuck after some time but not always the same time. When it happens the webpage is unavailable, buttons are not responding, no communication possible at all. I have to switch the power off and on to get it working again. The slave is not freezing, but gives logical the COMM error. 

Config from the master:
{"version":"v3.6.3","serialnr":1234,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"AP","rssi":-50,"bssid":"7E:8A:20:4C:10:75"},"evse":{"temp":32,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":1,"pwm":1024,"solar_stop_timer":0,"state":"Charging Stopped - Waiting for Solar","state_id":9,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":13,"solar_stop_time":5,"enable_C2":"Auto","mains_meter":"Eastron3P","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"","port":,"topic_prefix":"SmartEVSE-49223","username":"","password_set":false,"status":"Disconnected"},"home_battery":{"current":131,"last_update":1719475920},"ev_meter":{"description":"Eastron3P","address":16,"import_active_power":0,"total_kwh":3193.5,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":3193.5,"export_active_energy":0},"mains_meter":{"import_active_energy":14929.09961,"export_active_energy":8982.799805},"phase_currents":{"TOTAL":-117,"L1":-16,"L2":-30,"L3":-71,"last_data_update":1719475926,"original_data":{"TOTAL":12,"L1":27,"L2":13,"L3":-28}},"backlight":{"timer":120,"status":"ON"}}

The problem is in SMART mode but also in SOLAR mode. Cannot reproduce it.
I will upload the debug version. Is it possible to get the log file after the device is rebooted?

dingo35, 2024-06-27:
Ok thanks for your report; the combination of a master/slave setup AND a home battery makes your configuration rather unique, so a real challenge to solve this!
I see two possiblities to debug this:
1) you flash a debug version and keep your telnet client running 24/7, and we hope that the last lines of your log give any clarity
2) you try to get a coredump; this requires some skills, the procedure is described in the file "coredumping.txt" in the repo... this method is much surer to point us directly at the problem, but requires much more preparation

Are you willing and able to do the second route, with my support, of course?

dingo35, 2024-06-27:
Ok that is a bug, clearing the host field should do the job.

But for now, you can clear your entire config by doing /erasesettings,

Or you can do give -DMQTT=0 as compile option....

rvdgaag, 2024-06-28:
I have a similar setup with 2 Smart EVSE's and a 30 kWh battery setup. Will monitor what happens when I charge the cars tomorrow, I didn't experience issues recently.

dingo35, 2024-06-28:
Thx for your help Ronald!

rvdgaag, 2024-06-28:
@dingo35 it's René by the way ;-)

dingo35, 2024-06-28:
O sorry René, confused you with another faithful community member!

rvdgaag, 2024-06-29:
Mine work fine, no freeze up, I feed both mains and ev data with MQTT.

Just the charge overide doesn't work. It always charges at 3x 16A max

vicom147, 2024-07-01:
Strange, since I flashed the debug version it is running fine.

Telnet is not running 24/7, after about 2 hours it stops. The connection is not closed but no info on the screen anymore.

@rvdgaag same here, charge override does not work, keeps charging with 16A

dingo35, 2024-07-01:
Glad your main problem is solved.
Override is only supposed to work in Smart and Normal mode, and only works on the Master; on my testbench this is obeyed, so I cannot reproduce.
Also note the remark on the webserver, after choosing the override current you MUST tick Normal or Smart to have the Override current accepted. You can verify if its accepted by the "Current Details" box, where it displays the value of Override (current, that is).

vicom147, 2024-07-01:
So it is not possible to set the charging current for the slave? 

I saw in another but/issue, it was in a previous version possible to set Master at Solar and Slave at Smart. I like that possibility, because we have 2 PHEV's. Sometimes I wan't to charge Smart because I have to go in the evening, the other car can charge at Solar because we don't need it the other day.

closed 2024-07-01 by dingo35

dingo35, 2024-07-01:
Putting the slave in another mode then the master has never worked, AFAIK it showed the screens of those modes but again, master decides....
So we fixed that bug, modes of master ans slaves are always the same.

vicom147, 2024-07-01:
Ok, I just put the "master" on 6A current. That works, but the slave is on 10A (Fuse is 16A). So it looks like the master does not send the current "limit" to the slave.

dingo35, 2024-07-01:
Yep the override current works locally for the master.

vicom147, 2024-07-07:
Issue is still there, Telnet won't run 24/7 it stops random after certain hours. Tried multiple cliënts. Connection is not closed, but newer info is not showed.

The coredump, need a little bit help with that one. Where do I have to install the esp32-arduino-lib-builder? 
Or is this based on a Linux environment?

dingo35, 2024-07-07:
Yes I only have Linux, perhaps it works under WSL?

vicom147, 2024-07-07:
I'll try, it is installing now. To receive the coredump I need USB-connection with the SmartEVSE? Not possible over wifi I suggest.

dingo35, 2024-07-07:
Can be done both in Windows or Linux; for windows youll need to install a driver, Linux works right out of the box....
Look in the manual, links are on the bottom of the home page of the repository...

vicom147, 2024-07-07:
Looks like it is running with WSL.

Do I need to flash firmware.bin and partitions.bin in this case or only firmware.bin

dingo35, 2024-07-07:
You will have to rename partitions_custom.csv.coredump to partitions_custom.csv, and then pio run -t upload will generate and flash all the necessary files.

vicom147, 2024-07-08:
Flashed with Linux, but when I try to get the coredump I'm getting an error and the SmartEVSE freezes. 

Error: A fatal error occured: Corrupt data, expected 0x1000 bytes but received 0x7b2 bytes (this is with esptool.py --chip esp32 port ... ... read_flash 0x3F0000 65536 coredump.bin)

with esp-coredump command also error, incorrect size of core dump image: 196608, use partition size instead: 65536
After that als fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0x7be bytes

[error.txt](https://github.com/user-attachments/files/16131175/error.txt)

mstegen, 2024-07-09:
Hi @vicom147 perhaps there is something wrong with the SmartEVSE hardware, can you send me an e-mail at michael@stegen.com with your contact details, then we can arrange for it to be swapped for a new one.

Btw in the mean time, if you have your controller near a computer, you can also log through the USB-C 'serial' port.
This is my preferred way to do it, as it does not rely on a network connection/Wifi, and crash dumps will be directly sent to this port as well.
You should then compile the firmware with the following build flags:
    -DDBG=2
    -DCORE_DEBUG_LEVEL=5
    -DLOG_LEVEL=3 (or higher)