dingo35/SmartEVSE-3.5 #92: "no power available" after switching from SOLAR to SMART

demerstraat, 2024-06-29:
**Describe the bug**

**Single Phase charger (C2 not present)**

after SOLAR charging stops because "waiting for solar power"
a mode change to SMART or NORMAL does not start charging, status = "no power available", 
although up to 14.0A is available within my maxsummains limit

>(D) (IsCurrentAvailable)(C1) No current available MaxSumMains line 954. 
ActiveEVSE=1, MinCurrent=6A, Isum=3.2A, MaxSumMains=18A.

after reboot, charging starts in SMART/NORMAL as expected

i believe the culprit is _must_be_single_phase_charging_  in line 943. 
case C2 NOT_PRESENT is not evaluated

**To Reproduce**
car is charging in SOLAR mode. 
no more solar power available, charging stops as expected, status = "waiting for solar power".
from this status, I change mode to SMART or NORMAL, 
charging does not start, status "no power available"

baseload: +-4.0A
maxsummains: 18.0A
min_current: 6.0A
> this should allow the EV to charge between 6.0A and 14.0A

i also tried if increasing maxsummains would help, 
increased maxsummains in steps of 1.0A. 19, 20 and 21 did nothing (=no start of chargedelay 60 countdown)
setting maxssummains to 22 did start charging, 

i believe the culprit is _must_be_single_phase_charging_  in line 943. 
case C2 NOT_PRESENT is not evaluated, 

so available current is evaluated as if a 3phase charger. (3x1x6+4 = 22A), so that's why 22 as maxsummains worked.
:
```
    bool must_be_single_phase_charging = 
    (EnableC2 == ALWAYS_OFF || (Mode == MODE_SOLAR && EnableC2 == SOLAR_OFF) ||
    (Mode == MODE_SOLAR && EnableC2 == AUTO && Switching_To_Single_Phase == AFTER_SWITCH));
    int Phases = must_be_single_phase_charging ? 1 : 3;
    if ((Phases * ActiveEVSE * (MinCurrent * 10) + Isum) > (MaxSumMains * 10))
```
**debug**
[debug_evse.txt](https://github.com/user-attachments/files/16041079/debug_evse.txt)

**Expected behavior**
start in smart mode from lowest possible charging setting

dingo35, 2024-06-30:
So I can see from your logs that your grid is 3 phase; Am I correct in assuming that your EV is charging 1phase and your problem is that SmartEVSE is calculating with charging 3 phase ?

demerstraat, 2024-06-30:
Grid is 3 phase, C1 contactor is 1 phase, C2 contactor not present. 
EVSE configured with C2 not present.

The problem is:
Changeing mode from Solar to Smart does not resume charging (no power available) after charging stopped  correctly in Solar because no more Solar available.

It only does this when stopped from solar, because reboot in mode smart fixes is.

I believe the cause is the must_be_single_phase_charging that assumes 3 phase in case C2 set to "not present".
But i cannot see why the reboot seems to fix is

dingo35, 2024-06-30:
It is an arbritary choice whether we assume 3phase charging or 1phase charging in our calculations.
To stay compatible with the first firmware versions, AND to stay on the safe side, if we cant derive it from the settings, we assume you are 3phase charging.

You can get the behaviour you want by setting C2 to "Always Off". That way the firmware knows you are charging 1phase.

demerstraat, 2024-06-30:
I understand.
just seems strange that when starting "fresh" in Smart vs. going to Smart from Solar gives a different behaviour.

dingo35, 2024-06-30:
It is, until you realise that in Smart charging _everything_ is per phase, so it doesnt matter how many phases you are charging.

With the introduction of StartCurrent and ImportCurrent, which are summed over all phases used, its a different ball game in Solar.

demerstraat, 2024-06-30:
fully aware of that.
the point though was, all with unchangend config, identical limits and consumption: 
```
boot -> smart         -> solar        -> stopped from solar -> smart
     -> **charging**  -> charging     -> no charging        -> **no charging**
```

dingo35, 2024-07-01:
Ok  I get your point now, I will try to reproduce..

dingo35, 2024-07-04:
I cannot reproduce, in my tests when MaxSumMains 22A it starts charging, and below it doesn't, irrespective of reboot or not, which is intended behaviour.

So the "waiting for power" states in your debug low are exactly intented behaviour. The only strange behaviour would be the charging in Smart mode after a fresh reboot, which I cannot reproduce, and you have not included a log for that....

closed 2024-07-04 by dingo35

demerstraat, 2024-07-09:
_"I cannot reproduce, in my tests when MaxSumMains 22A it starts charging, and below it doesn't, irrespective of reboot or not, which is intended behaviour._
"
with C2 set to "Always Off " i get indeed the same and intended behaviour as you mentioned.
did you try it with setting C2 to "NOT_PRESENT" ? 

I reproduced the behavior of the initial post with current version 3.6.5 
somehow after fresh reboot, C2 not_present causes evaluation as single phase when checking for available power in Smart mode (1x6A as lowest to fit within (maxsummains - baseload)), 
whereas once Solar mode has stopped because no power, Smart mode now suddenly assumes 3x6A as lowest to fit within (maxsummains - baseload).

It's not something major and it is indeed avoided with setting C2 to "Always Off" . 
Could also be clarified in configuration.md (not_present assumes C1 = 3 phase contactor)

[log_SmartOK_afterreboot.txt](https://github.com/user-attachments/files/16145004/log_SmartOK_afterreboot.txt)
[log_SmartOK_to_Solar_to_SmartNOK.txt](https://github.com/user-attachments/files/16145006/log_SmartOK_to_Solar_to_SmartNOK.txt)

dingo35, 2024-07-09:
Good suggestion, I added some remarks to the docs, thx!