dingo35/SmartEVSE-3.5 #93: V3.6.4 loosing WiFi Connection after few hours

Imaginous, 2024-07-05:
**Describe the bug**
After  updating from v3.6.3 to v3.6.4 the device disconnected from WiFi after about 2 hours. 

**Upload your config**
'{"version":"v3.6.4","serialnr":6972,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Imaginous","rssi":-53,"bssid":"xx:xx:xx:xx:xx:xx"},"evse":{"temp":31,"temp_max":55,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":102,"solar_stop_timer":0,"state":"Connected to EV","state_id":1,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":26,"current_main":33,"current_max_circuit":32,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":8,"solar_stop_time":30,"enable_C2":"Always Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"mqtt.imaginous.nl","port":1883,"topic_prefix":"SmartEVSE","username":"mqtt","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":51,"L1":51,"L2":0,"L3":0,"last_data_update":1720178257,"original_data":{"TOTAL":51,"L1":51,"L2":0,"L3":0}},"backlight":{"timer":0,"status":"OFF"}} '

Bssid masked with xx

**To Reproduce**
Car is connected, but not charging. 
I send the mainsdata via the API. 
After a few hours the SmartEVSE disappeared from the WiFi network. 
Display states no Wifi connection. 
Disabling and re-enabling WiFi on the display immediately reconnects to WiFi. 

**Expected behavior**
Staying connected. 

**Screenshots**
[image]

dingo35, 2024-07-05:
Thx for your report, very disturbing, so high prio.

Your wifi signal seems very strong (-53), but still, could you check if this might by caused by interfering wifi stations?

Nevertheless it should reconnect automatically....

dingo35, 2024-07-05:
If I turn off my 2.4Ghz signal, and turn it on again, it reconnects automatically.... so its a mystery to me whats going on here....

Imaginous, 2024-07-05:
Nothing has changed in the WiFi department.
Other devices stay connected.

I rebooted the accespoint... (Asus RT-AX86S)... the SmartEVSE doesn't reconnect.

Other devices do.

dingo35, 2024-07-05:
Ok I just had the same experience on my production unit, not on my test bench though ?!?

I pulled back the release, again thx for the report.

dingo35, 2024-07-05:
[d2de94b97.zip](https://github.com/user-attachments/files/16109757/d2de94b97.zip)

Ik ben momenteel deze versie aan het testen, misschien kun je mee testen ?

Imaginous, 2024-07-05:
I just tested the router reboot again with v3.6.3, it also has the same behavior.

V3.6.2 does reconnect fine.

So the regression occurred in v3.6.3.
I assume there was a short interference on my WiFi after which SmartEVSE didn't reconnect.

Imaginous, 2024-07-05:
> [d2de94b97.zip](https://github.com/user-attachments/files/16109757/d2de94b97.zip)
> 
> Im currently testing this version, perhaps you can test in parallel?

This version also doesn't reconnect. Upto v3.6.2 it seems to work.

dingo35, 2024-07-05:
So v3.6.5 is not out yet, which version are you reporting on ?

Imaginous, 2024-07-05:
v3.6.2 keeps reconnecting without a problem

Imaginous, 2024-07-05:
> So v3.6.5 is not out yet, which version are you reporting on ?

Version v3.6.3 / v3.6.4 and the link you gave for parallel testing v3.6.5 all don't reconnect.

They connect after a power cycle, one router reboot and after a second router reboot they don't reconnect.

Version 3.6.2 keeps reconnecting.

dingo35, 2024-07-05:
It seems to be dependent on your LAN setup; on my test bench all versions reconnect fine, on my production unit, that is connected via a repeater, it doesn't. AP = Fritzbox.

What is your network layout + AP brand?

Imaginous, 2024-07-05:
I have an Asus RT-AX86S (with Merlin firmware) configured as access point.
Use OPNsense as firewall.
SmartEVSE is configured via a static DHCP lease.

As said earlier, v3.6.2 doesn't have a problem reconnecting.

dingo35, 2024-07-05:
v3.6.2 is not reconnecting here either; Im starting to suspect my repeater has problems....

dingo35, 2024-07-05:
..... and now v3.6.3 and v3.6.4 are reconnecting reliably ..... hard to debug if there is no sound reproduction scenario.....

I noticed that the reconnection time can differ from 20s to 2 minutes; @Imaginous can you confirm that v3.6.4 will reconnect if you give it 2 minutes time?

Imaginous, 2024-07-05:
Waited for 5 minutes with v3.6.4, no reconnect.

With v3.6.2 it's 2 seconds.

dingo35, 2024-07-05:
Ok thx. Still cant find a pattern here...

Imaginous, 2024-07-05:
Maybe just poll every X time if WiFi is unconnected and restart the WiFi bit.
I don't know what the difference is between v3.6.2 and v3.6.3?

dingo35, 2024-07-05:
Is the static lease configured static on your AP, or was it configured static on the SmartEVSE?

Do you have the same behaviour if you let it have a dynamic ip address?

Imaginous, 2024-07-05:
It's static on rhe OPNsense router, coupled to the MAC address.

If I remove the static address it still goes wrong.
Even the access point doesn't see the MAC address.
It seems the SmartEVSE isn't trying to connect.

I wonder what happens in the WiFi disable routine. Since it always reconnects after disabling / enabling.

dingo35, 2024-07-06:
[v3.6.4a.zip](https://github.com/user-attachments/files/16114954/v3.6.4a.zip)
[v3.6.4c.zip](https://github.com/user-attachments/files/16114955/v3.6.4c.zip)
[v3.6.4b.zip](https://github.com/user-attachments/files/16114956/v3.6.4b.zip)

Since you seem to be the only one that can reliably reproduce the reconnect bug, could you please test these 3 versions; do they reconnect in your setup and if so, how long does it take them?

Imaginous, 2024-07-06:
SmartEVSE reconnect test:
A - about 5 seconds to reconnect (feels a bit faster then B) 
B - about 5 seconds to reconnect 
C - 7:51 - 7:57 no reconnect after 6 minutes

Imaginous, 2024-07-06:
Does the outcome make sense to you?

Version C doesn't reconnect, version A feels faster than B. I'm running on A at the moment.

Even after multiple access point reboots it keeps reconnecting fast.

dingo35, 2024-07-06:
@Imaginous this makes perfect sense, 4a has this code;
[image]

4b had instead of reconnect() a WiFi.begin() in it, which is basically what is done after a Wifi Disable/Enable. A reconnect() has less work then a begin() , so it makes sense 4a is slightly faster then 4b.

Ok lets run 4a for a few hours, could you report say at the end of the day if you are still stable, so I can get v3.6.5 out with this change in it?

dingo35, 2024-07-06:
By the way, this code was in there up till v3.6.1, and was removed in v3.6.2 . So it didn't make sense v3.6.2 was working for you ....

Imaginous, 2024-07-06:
I will keep you posted.

Thanks for the swift support.

Imaginous, 2024-07-06:
It's still working, car is charging at the moment.

dingo35, 2024-07-07:
@Imaginous The 4a version gives some troubles with usb-serial flashed firmware, somehow ...

[v3.6.4d.zip](https://github.com/user-attachments/files/16119435/v3.6.4d.zip)

Could you test this version for us?

dingo35, 2024-07-07:
@Imaginous Ok I'm getting a report that 3.6.4d is not working well at my fellow developer's LAN, this one works fine at both our infrastructures, how about yours?

[v3.6.4e.zip](https://github.com/user-attachments/files/16119783/v3.6.4e.zip)

(hopefully this is the one, sorry to bother you....)

Imaginous, 2024-07-07:
> @Imaginous Ok I'm getting a report that 3.6.4d is not working well at my fellow developer's LAN, this one works fine at both our infrastructures, how about yours?
> 
> [v3.6.4e.zip](https://github.com/user-attachments/files/16119783/v3.6.4e.zip)
> 
> (hopefully this is the one, sorry to bother you....)
Sorry for the late reply, just got home. 

Tested the E variant, works fine here. Tested 2 reconnects both worked fine and connected within 5 seconds.

Car is now charging, this also works fine.

dingo35, 2024-07-07:
Ok thx for your help, going to release it first thing in the morning!

sonar98, 2024-07-10:
I'm still experiencing problems with my WiFi connection with the latest release. the problems started a few releases ago.

dingo35, 2024-07-10:
Did you try the 3.6.4e version in this thread?

sonar98, 2024-07-10:
No not yet. 
I just installed it to test it.

dingo35, 2024-07-10:
Ok; please be precise in your descriptions. "Problems", "a few releases ago" aint gonna be enough to start debugging...

sonar98, 2024-07-11:
I tried the 3.6.4e version. WiFi stayed connected much longer than 3.6.5 did. It stayed disconnected for over an hour. Had to powercycle to get connection again.  
WiFi coverage is pretty bad in my garage.

I installed 3.6.1 now. I will check if it stays connected with this version.

dingo35, 2024-07-11:
Usually you dont need to powercycle, disabling and re-enabling wifi should work too.

sonar98, 2024-07-12:
No disconnects or WiFi issues since yesterday after installing 3.6.1. When I block my EVSE in my access point and enable it again it reconnects right away. 

Today I tested with v3.6.2. When I block the connection and enable it again the EVSE does not reconnect anymore (waited over 5 minutes) and I have to toggle the wifi on the device to get it connected again.

dingo35, 2024-07-12:
That issue is known in v3.6.2, v3.6.3 and v3.6.4 .
V3.6.4e and v3.6.5 are basically the same as to the reconnect mechanism...

sonar98, 2024-07-12:
Tried v3.6.5 again and it won't reconnect. Conclusion for my situation is that v3.6.1 was the last version that was stable.

My AP's are UniFi U6 Pro.

dingo35, 2024-07-12:
Only try a and b ....

dingo35, 2024-07-12:
[turnofsetautoreconnectonmanualreconnect.zip](https://github.com/user-attachments/files/16194772/turnofsetautoreconnectonmanualreconnect.zip)

What does this one do?

sonar98, 2024-07-12:
This one does not reconnect.

My procedure to test is:
- load web interface
- Unifi: block device
- refresh web interface -> nothing happens
- Unifi: unblock device
- refresh web interface

This cycle is done within a few seconds.
Could it be that the connection is available again within the reconnect loop and that it then somewhere gets stuck in the loop?

sonar98, 2024-07-12:
And I don't know if this is influencing anything, I'm using smart mode and I'm getting the data for the currents from home assistant and if the connection is lost it's giving the serial connection error because it's not updated anymore.

sonar98, 2024-07-12:
Yes that is true but the device is also showing as offline in my Unifi network manager until I manually switch off/on WiFi on the EVSE.

dingo35, 2024-07-12:
Yes lets introduce another variable, the network manager ?!?!
Ping is reliable, third party software we dont know how it reacts. 
Also with ping you can see intermittent connections and packet loss..

But if you know better  how to debug, be my guest...

dingo35, 2024-07-12:
Also you should monitor at least 60 seconds to see if a wifi reconnect happens....

sonar98, 2024-07-12:
It is not an extra variable it is where I manage my WiFi accesspoint to block and unblock to test.
But no problem I will use ping to detect that it is not available.

I waited at least 5 minutes for a reconnect.

dingo35, 2024-07-12:
Ok great! What does this one do:

[conditionalcallofmgr_poll.zip](https://github.com/user-attachments/files/16195355/conditionalcallofmgr_poll.zip)

sonar98, 2024-07-12:
After installing this one connection to WiFi did not come back up. Tried to switch off and on on the EVSE but still no luck.
ESPTouch mode nothing was discovered.

Had to power cycle to get WiFi to connect. Then I tried a reboot through the web interface. 
WiFi did not connect. Had to power cycle again to get a connection.

dingo35, 2024-07-12:
Ok running out of inspiration right now, back tomorrow!

mstegen, 2024-07-12:
While not exactly the same, i think my setup might be similar.
I use three UAP AC Lite's throughout the house/garage, these are controlled by a Unifi Gateway Ultra.
When i block a SmartEVSE from the gateway's web interface, and then unblock, it will reconnect immediately.

As there are a ton of settings related to WiFi in the Gateway, maybe some of them are not correctly configured?
Anyway here are my (auto) settings (switched to manual to make them visible):

[image]

mif11k, 2024-07-12:
Hi guys,

I am following the wi-fi reconnect problem and I must say that 3.6.2 -
3.6.5 and todays bugfix release behave the same for me to. I have a Cisco
AP and I am ready to help. If you want me to test any new builds, you are
welcome.

BR,
Michal

dingo35, 2024-07-13:
@sonar98 could you post your settings?

dingo35, 2024-07-13:
Having problems with 4a might be Unify specific:
https://www.reddit.com/r/esp32/comments/xmxumq/esp32_unifi/

dingo35, 2024-07-13:
Nobody in this thread using hidden ssid, right?

Imaginous, 2024-07-13:
My SSID is visible and I'm using WPA2-personal with AES.

dingo35, 2024-07-13:
Ok guys, I am currently testing this version, which has the workaround from v3.6.5 removed but also a delay that might be disturbing stuff. Reconnecting fine here, hope it does the same for you:
[noworkaroundnodelay.zip](https://github.com/user-attachments/files/16201519/noworkaroundnodelay.zip)

Imaginous, 2024-07-13:
This new version reconnects fine here.

sonar98, 2024-07-13:
> While not exactly the same, i think my setup might be similar. I use three UAP AC Lite's throughout the house/garage, these are controlled by a Unifi Gateway Ultra. When i block a SmartEVSE from the [...]

My settings are default as well.

sonar98, 2024-07-13:
> Ok guys, I am currently testing this version, which has the workaround from v3.6.5 removed but also a delay that might be disturbing stuff. Reconnecting fine here, hope it does the same for you: [noworkaroundnodelay.zip](https://github.com/user-attachments/files/16201519/noworkaroundnodelay.zip)

This version does not reconnect with my setup. I had to reconnect manually (waited over 5 minutes) to get it working again.

[26 more replies]

closed 2024-07-16 by dingo35