dingo35/SmartEVSE-3.5 #94: Solar charging no power available

Imaginous, 2024-07-08:
**Describe the bug**
Solar Charging aborted after a few seconds when there is not enough sun. 
The count down starts 3600 seconds in my case. But after 5 seconds it stops with the message no power a available. 

**Upload your config**
'{"version":"v3.6.5","serialnr":6972,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Imaginous","rssi":-50,"bssid":"7C:10:C9:xx:xx:xx"},"evse":{"temp":36,"temp_max":55,"connected":true,"access":true,"mode":2,"loadbl":0,"pwm":131,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":77,"override_current":0,"current_min":6,"current_max":26,"current_main":33,"current_max_circuit":32,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":8,"solar_stop_time":60,"enable_C2":"Always Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"mqtt.imaginous.nl","port":1883,"topic_prefix":"SmartEVSE","username":"mqtt","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":-74,"L1":-74,"L2":0,"L3":0,"last_data_update":1720431831,"original_data":{"TOTAL":-74,"L1":-74,"L2":0,"L3":0}},"backlight":{"timer":120,"status":"ON"}}4' 

**To Reproduce**
Steps to reproduce the behavior:
1. SOLAR mode without enough sun

**Expected behavior**
Keep charging at 6A until timer has expired. 

**Screenshots**
[image]

dingo35, 2024-07-08:
Ok I can reproduce this, thx for your report!

Imaginous, 2024-07-08:
Sorry to find a bug again ;)

Keep up the good work, I know as developer how eager you are to push the update... and then... a bug.

dingo35, 2024-07-08:
Yes, and worse, now I cannot reproduce the problem anymore ...?!?

mstegen, 2024-07-08:
I found a way to reproduce the bug.
-Start charging in solar mode. 
-Slowly increase phase currents, Sum of phases should be around 0A-1A
-Controller stops charging.

dingo35, 2024-07-08:
@Imaginous this should solve it:
[8c77f1c0c1.zip](https://github.com/user-attachments/files/16129106/8c77f1c0c1.zip)

Imaginous, 2024-07-08:
Installed it, no sun at the moment.
I will keep you posted.

Also did a WiFi reconnect test, that is still okay.

Thanks again.

Imaginous, 2024-07-09:
Today the car charged fine, it seems this test version is working.

dingo35, 2024-07-09:
Same here! Thx for testing..

closed 2024-07-09 by dingo35