dingo35/SmartEVSE-3.5 #95: Taxability according to §14a of the Energy Industry Act (Germany)

Elektrofix-OL, 2024-07-08:
**Is your feature request related to a problem? Please describe.**
The new § 14a of the Energy Industry Act came into force in January 2024.
This affects all consumption devices with an output of 4.2 kW or more that have been put into operation since January 1, 2024.
An EVSE can only be used if it is “dimmable”.

**Describe the solution you'd like**
The energy supplier provides a relay that is opened when the power needs to be reduced to 4.2kW.
In Smart mode, the SmartEVSE could check whether there is an interruption in the PP line. If the EVSE measures a resistance >4700 ohm, it switches to a basic value of 6A and "dimming" is active.
In solar mode this is not necessary because the SmatEVSE should not be connected to the grid.

**Describe alternatives you've considered**
Interruption of the CP connection to cancel the charging process and thus fulfill the requirements of §14a

Elektrofix-OL, 2024-07-14:
Your energy supplier will only impose the Relias (NC) on you if your wallbox is registered this year and the power connection is in Germany. In addition, he must have an intelligent measuring system installed for you.
He is not yet allowed to force existing systems to dim.

I also don't understand why you immediately close the request without giving others a chance to comment.
Isn't SmartEVSE so widespread in Germany that it's worth working on problems in Germany?

dingo35, 2024-07-14:
No I close the comment because:
a) it is does a technically below par suggestion
b) since I cant obtain the necessary hardware, I cant implement it.

I'm happy to honour any PR from you, as long as it obeys the charging standards. Switching CP and or PP with a contactor does not fit in there.

ocaj-nl, 2024-07-17:
A more fitting technical solution would be to have a separate input-device (like the sensorbox) that is connected through modbus to the SmartEVSE. This device could then read in the input relay from the energy-supplier and forward this information to the SmartEVSE. Then the SmartEVSE would need some "dimming-logic" to make sure that the system behaves as expected.

Of course, this would also require some "sensorbox ++"-device which can read a relais-contact, as I don't think the current sensorbox supports that?

(just some thought on the topic, I don't know whether such solution would be compliant with the German law)

dingo35, 2024-07-17:
You are correct. But we need the specifications of such a device. Is the relay coupled to a MQTT or REST sensor enough? What are the legal and technical requirements?

dingo35, 2024-07-17:
So anybody could implement this themselves by defining a switch in HomeAssistant that reacts to the relay, or even thru MQTT?

Elektrofix-OL, 2024-07-17:
It would be a possibility, not elegant since HomeAssistant and the WLAN are there in between, but feasible. I would then have to send an API to the SmartEVSE every minute to ensure that no one makes a change when the dimming is active. 

I haven't managed to pass a ChargeCurrentOverride with 60 to the SmartEVSE via Mqtt yet, it jumps back to 0.
I use IoBroker

dingo35, 2024-07-17:
Are you sure you are in the right mode when applying OverrideCurrent?
Try the web interface first....

Elektrofix-OL, 2024-07-17:
I'm in Smart mode and then enter the value 80 in mqtt.0.SmartEVSE.ChargeCurrentOverride. After a few seconds the value is written back to 60

I can do everything on the web and everything runs via API.
V3.6.5

dingo35, 2024-07-18:
Thx for your report!
"I can do everything on the web and everything runs via API."  

Now we are both non native speakers, what do you mean by this?
Do you mean that everything works well with the built-in webserver? Or not?
Which API? REST, MQTT?

dingo35, 2024-07-18:
RTFM: Your topic should be "SmartEVSE-xxxx/Set/CurrentOverride"

Elektrofix-OL, 2024-07-18:
Yupp, I'm German and do English via Google.

Everything works well via the web interface, I can set all the restrictions there.
Limiting to 6A also works in the API: curl -X POST "http://192.168.5.23/settings?mode=3&override_current=60" -d ' '
Only Mqtt doesn't work. I have no influence on the topic, IoBroker does it alone.

Elektrofix-OL, 2024-07-18:
The idea is correct and would also work with §14.
The sensor box 1.0 had four inputs, three were for the CT terminals L1, L2, L3, one was unused.

But recently I swapped an EVSE2.0 for another EVSE2.0, unfortunately the sensor box doesn't work (communication error), I think something has been changed. The two EVSE had different software versions on them.

Elektrofix-OL, 2024-07-18:
> A more fitting technical solution would be to have a separate input-device (like the sensorbox) that is connected through modbus to the SmartEVSE. This device could then read in the input relay from [...]

The idea is correct and would also work with §14.
The sensor box 1.0 had four inputs, three were for the CT terminals L1, L2, L3, one was unused.

But recently I swapped an EVSE2.0 for another EVSE2.0, unfortunately the sensor box doesn't work (communication error), I think something has been changed. The two EVSE had different software versions on them.

mstegen, 2024-07-18:
The old sensorbox 1 used a different protocol. It's however possible to reprogram it, but you will need to use a pickit programmer for that. The software is available here:
https://github.com/SmartEVSE/Sensorbox-1

Extra Inputs on the SmartEVSE are difficult to add, as there is no room left to connect a wire.

Perhaps the Sensorbox 2 can be used, as it has a RJ12 connector with 2 I/O lines, now configured as RX/TX for communicating with a smartmeter, but in Germany this is not used, so these lines might be used for other purposes.

Elektrofix-OL, 2024-07-20:
Then the software in the sensor box must be adjusted: If EVSE in Mode 4, then TX = Input; Input = false Throttle power from the grid

In Mode 4, the EVSE must monitor and regulate the total power from the network. The maximum grid performance must then be adjustable in a menu, because §14a allows for higher performance if, for example, there is a heat pump and wallbox in the house.

If I take the unbalanced load regulation into account in addition to §14a, then it would also be possible to monitor the grid current per phase instead of the grid power

Yes, we Germans are a strange people with all the rules and regulations, even as a native speaker it's not easy to read if you're not a professional in the subject.

I have no idea if this feature can be easily integrated into the EVSE. I'm a master electrician and not a software fan.
I will soon install an EVSE with my colleague and operate it illegally because I cannot register it without controllability.

dingo35, 2024-08-05:
How about if we add an option to the existing switch connections?

Now we have options to use the existing switch connections for switching Access On / Off or switching Smart vs. Solar; it would be relatively simple to add an option "Power Reduction Relay" that would limit the power usage to MinCurrent (or is the 6A fixed by law?)

Of course this way people using the relay would not be able to use the switch for other purposes, but switching Access On/Of and Smart/Solar can already be done by the buttons on the LCD screen, so perhaps not a big problem?

Also: what would be the limit in case of a Master and e.g. 3 Slave chargers? Would there be 1 relay, and would the current have to be limited to 6A, or 4 x 6A? Or would every slave have its own relay?

Elektrofix-OL, 2024-08-05:
I had already thought about the switch, I would have thought that there would be too many functions on the switch.
The grid operator only switches one relay per grid connection to control §14a devices.
Depending on the device, the power is dimmed; there is a formula for calculating this.

The power limit of 4.2kW ​​was chosen, because it allows 3-phase charging to be reduced to the minimum without having to switch to 1-phase charging. This means that the performance can be increased again later, because the dimmer is limited to a maximum of 2 hours.

What I could imagine:
- The switching input is taken as "Power Reduction Relay", if the input is true, it will not be reduced.
- The reduced maximum power of the EVSE is entered in a new menu.

If someone has multiple EVSEs, the benefit can be divided among the EVSEs. However, the reduced power must be calculated and changed because the power only needs to be dimmed for the controllable §14a devices. Appliances such as electric stoves, lamps and ovens do not need to be dimmed, so the reduced maximum power cannot be determined at the grid connection point, ist must be as a sum from the EV meter.

The electricity from a PV system can add to the reduced maximum output because it is not drawn from the grid. PV + reduced maximum power = charging power

Luxury idea:
If the switch button were an opener, it would still be technically possible to switch between solar and smart in non-reduced mode. The SmartEVSE would only have to react to falling edges with a duration of 0.5 seconds

Elektrofix-OL, 2024-08-05:
Storing the reduced maximum performance requires no more than one byte with a comma shift. 256=25.6 kW
8 charging points = 18,9 kW

dingo35, 2024-08-06:
If you connect your energy-provider-provided relay to the Switch connector and GND, this should work:

[118a39bce87.zip](https://github.com/user-attachments/files/16507627/118a39bce87.zip)

Please test!

Elektrofix-OL, 2024-08-07:
I'm currently traveling for the company and will be able to test it out this weekend, Thanks

Elektrofix-OL, 2024-08-18:
I tested the version. My measurement setup:
Fuses => EV electricity meter => Contactor 4-pole (C1) => Contactor 2-pole (C2) => Auto
A button (NO) and the RG LED are connected to the SmartEVSE.
So dimming is always active. When I press the button, switching off dimming is simulated.

My observations:
- Solar: It is running normally.
- Smart: no dimming detectable
- Normal: Dimming works and can be turned off and on again. However, it is set to 6A current and not to the 4300W output of the EV electricity meter.

My wishes:
- Dimming in smart mode.
- Dimming according to the power of the EV meter  
- It would also be good to have an indicator that shows that the EVSE is charging in a dimmed state, perhaps in faded red instead of faded green

Elektrofix-OL, 2024-08-25:
I installed version 3.6.9, but there is no option to select the “Grid Relay”.
When I searched the code, I didn't find anything called "Relay".
Wasn't that compiled?

dingo35, 2024-08-25:
No, look at the release notes, its a bugfix release.

Elektrofix-OL, 2024-08-25:
OK, but in the SmartEVSE-3.5/docs/configuration.md has the option son in it, so I thought it was already in the firmware

dingo35, 2024-09-07:
[3d4359e3.zip](https://github.com/user-attachments/files/16918037/3d4359e3.zip)

This version should have Smart mode limited charging fixed; also it shows "LIMITED" in the LCD display when appropriate.

The EVMeter is only taken into account to determine the number of phases that are used to charge. If the EV determines to use less then the amount that is made available, it is of no use to make even more available....

Please test....

Elektrofix-OL, 2024-09-14:
I tested the version.
My measurement setup: see above
My observations:
Solar: It's running normally.
Smart/Normal: Dimming works and can be turned off and on again. However, dimming only works if the grid-relay is closed and then opened. If the grid-relay is open when restarting, dimming will not be activated. This behavior is the same every time the SmartEVSE is restarted.
This is not always the case when switching from solar to smart/normal. Dimming often does not occur if the grid-Raley was open during the switchover. After closing and reopening the grid-relay, dimming takes place.

dingo35, 2024-09-14:
Thanks for the elaborate testing!

You are right, the variable that determines whether the relay is open or closed is only set on switching. I have to rethink this part of the code...

Elektrofix-OL, 2024-09-22:
There is no rush.
I had an accident at work and am on sick leave until October 6th, so I hardly drive. Tests are not possible until October.

dingo35, 2024-10-04:
@Elektrofix-OL Thats too bad Karsten, hope you are feeling better ...

This should fix it:
[912006ad.zip](https://github.com/user-attachments/files/17259996/912006ad.zip)

Elektrofix-OL, 2024-10-06:
I tested the version.
My measurement setup: see above
My observations:
Solar: I didn't get a chance to test it, there wasn't enough sun
Smart/Normal: Dimming works straight away if the input is chose to Grid-Ralay.
Issuse:
When restarting, the SmartEVSE does not boot properly. The display flashes and shifts the display vertically. It cannot be reached via WiFi and charging no longer works. The LED button flashes red briefly (0.25 seconds) and then turns green.
A restart with the grid relay closed ensures that the display is stable, but charging is not possible.

Test canceled and current version installed, system then runs normally.

dingo35, 2024-10-25:
Hi Karsten, in this release it should work without the quirks you experienced:
https://github.com/dingo35/SmartEVSE-3.5/releases/tag/v3.7.0

Elektrofix-OL, 2024-10-27:
Sorry Issuse ist still alive:
When restarting, the SmartEVSE does not boot properly. The display flashes and shifts the display vertically. It cannot be reached via WiFi and charging no longer works. The LED button flashes red briefly (0.25 seconds) and then turns green.
A restart with the grid relay closed ensures that the display is stable,

dingo35, 2024-10-27:
I cannot reproduce the problem; could you perhaps:
1) post the _exact_ /settings that you are using when reproducing this problem
2) in platformio.ini, change these values from 0 to 5:
```
    -DCORE_DEBUG_LEVEL=5
    -DLOG_LEVEL=5

```
and then compile the v3.7.0 source code with this command line:
` PLATFORMIO_BUILD_FLAGS='-DDBG=2' pio run  -t upload`

...and monitor your USB port (in my case /dev/ttyUSB0) to see what the ESP32 is reporting when those problems occur?

Elektrofix-OL, 2024-10-27:
1. I use 192.168.9.23/update, the update is taken from the network. The GridRelay is open
2. I set the GridRelay via the display
3. I have left the menu
4. I designed the fuses for the SmartEVSE
5. I wait 10 seconds.
6. I put the fuse back in.
7. Display flickers
8. I try to enter the menu, but the reset takes less than 2 seconds.
9. I have designed the fuses for the SmartEVSE
10. I'll wait 10 seconds. The GridRelay is closed.
11. I put the fuse back in. The display is normal.

I don't have much experience with PlatformIO, I know what you want, but I've never compiled a firmware for ESP32.

Elektrofix-OL, 2024-10-27:
I made another attempt
1. I use 192.168.9.23/update, the update is taken from the network. The GridRelay is open
2. I set the GridRelay via the display
3. I have left the menu
4. I have designed the fuses for the SmartEVSE
5. I wait 10 seconds.
6. I rebooted via the interface.
7. I put the fuse back in. The display didn't flicker. Car could be loaded.
8. I have another update from the network.
9. I have designed the fuses for the SmartEVSE.
10. I'll wait 10 seconds. The GridRelay is closed.
11. I put the fuse back in. The display is black, as is the RGB.
I can't get EVSE to work anymore either.

I will try to send the software to the SmartEVSE via USB in the next few days, I hope I can find instructions online.

Elektrofix-OL, 2024-10-27:
I have access to the SmartEVSE again. Despite the display being black, I was able to install new firmware at 192.168.5.23/update. The update page was not as usual, only consisted of a few lines of text and was arranged at the top left. I could only select one file, but then the display came back. It flickered again until I removed the GridRelay from the setting, then everything worked normally.

dingo35, 2024-10-27:
Please goto http://192.168.x.23/settings and upload what your browser reports...

dingo35, 2024-10-27:
Also, please be aware that there is a bug in the update screen of v3.7.0 .
You should update to v3.7.1 by extracting firmware.bin from the zip file of the release (github repo), and update manually via "choose file" .

Elektrofix-OL, 2024-10-28:
The Settings:
`{"version":"v3.7.1","serialnr":6622,"mode":"SMART","mode_id":3,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"iot","rssi":-44,"bssid":"6A:D7:9A:24:A1:8D"},"evse":{"temp":24,"temp_max":65,"connected":false,"access":true,"mode":1,"loadbl":0,"pwm":1024,"custombutton":false,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":20,"current_main":50,"current_max_circuit":25,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":1,"enable_C2":"Solar Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.5.2","port":1883,"topic_prefix":"SmartEVSE","username":"Lehm-WG","password_set":true,"status":"Connected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":11,"import_active_power":0,"total_kwh":883.7999878,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":883.7999878,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":2,"L1":-47,"L2":8,"L3":41,"last_data_update":1730127872,"original_data":{"TOTAL":2,"L1":-47,"L2":8,"L3":41}},"backlight":{"timer":0,"status":"OFF"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}`

From Debug-Logging this errors came up:
`(MBhandleError)(C1) Error response: E2 - CRC check error, address: 0b, function: 04, reg: 0006.
(MBhandleError)(C1) Error response: E2 - CRC check error, address: 0b, function: 04, reg: 0048.
(MBhandleError)(C1) Error response: E2 - CRC check error, address: 0b, function: 04, reg: 0006.
(MBhandleError)(C1) Error response: E5 - Packet length error, address: 0b, function: 04, reg: 0006.
(MBhandleError)(C1) Error response: E2 - CRC check error, address: 0b, function: 04, reg: 0006.
(MBhandleError)(C1) Error response: E2 - CRC check error, address: 0b, function: 04, reg: 0006.
(MBhandleError)(C1) Error response: E5 - Packet length error, address: 0b, function: 04, reg: 0048.
(MBhandleError)(C1) Error response: E2 - CRC check error, address: 0b, function: 04, reg: 0006.
(MBhandleError)(C1) Error response: E2 - CRC check error, address: 0b, function: 04, reg: 0006.
`
Things only got worse with the 3.7.1 update. When the GridRelay is selected the display flickers until I close the relay, when I open it again it flickers again. If I take the GridRelay out of the config again, it works normally again.
Somehow the worm is in there and I have no idea why that is.

dingo35, 2024-10-28:
Im sorry Karsten but I left Windows 8.1 so I cant reproduce this in Linux; but as you can see it has to do with a header file in the SPI library; something seems to be wrong in your setup.

Are you on the most recent pio version? pio --version reports 6.1.16 on my system

dingo35, 2024-10-28:
It might also have to do with wrong IDE configuration, see https://stackoverflow.com/questions/9274026/how-to-solve-this-arduino-error-error-pins-arduino-h-no-such-file-or-directory

This is why I use plain platformio, without any IDE ; "A machine that knows too much is more of a handicap than a help."

dingo35, 2024-10-28:
You are not using a solid state relay are you?

Elektrofix-OL, 2024-10-28:
I didn‘t use a solid state relay outside from the SmartEVSE anymore. I use oldschool coil contactors on contakt C1 an C2.

dingo35, 2025-04-01:
Closed due to inactivity

closed 2025-04-01 by dingo35