dingo35/SmartEVSE-3.5 #98: RFID card support

matth-x, 2024-07-12:
Moving the discussion from the OCPP PR (https://github.com/dingo35/SmartEVSE-3.5/pull/82) here.

I compared the RFID readouts with the NXP PN532 and the 7th digit is different. To be honest, I haven't heard yet that OCPP backends would ignore the 7th digit (as discussed in the OCPP PR).

In addition to that, the RFID reader didn't recognize cards with only 4 digits. However, I'm not sure if 4-digit cards are commonly used.

@mstegen @fluppie

mstegen, 2024-07-12:
The onewire reader only supports 6 bytes of the UUID unfortunately. The last byte is the CRC byte.
4 byte cards should be read as [1st][2nd][3rd][4th] 00 00

fluppie, 2024-07-12:
Yes we should (with the current RFID reader) ditch the 7th. RFIDLastRead is doing it "right".
RFIDLastRead is also giving the same output as the LocalList in the Alfen ACE Installer software.

Would be nice that the RFID list we can upload with rfid.txt can be used as a locallist for when the evse is offline or the backend steve/eflux etc would be offline. In the OCPP commands I also see the ability to fetch the local list and also send a new or updated locallist. SendLocalList & GetLocalListVersion.
It can also be used to pre-authorize charging etc for a "quicker" response/reaction time. I think this could improve overal user experience.

Some snapshots of ACE installer to have an idea what kind of stuff can be configured.
[image]

[image]

[image]

matth-x, 2024-07-14:
Update: the RFID reader does recognize 4-byte cards, sorry for getting it wrong. Still, we need to be able to distinguish between 4-byte and 7-byte identifiers (since 4-byte readings are padded with 0). Could it be possible to encode the ID length into RFID[0] maybe?

@fluppie Thanks for the input. OCPP local lists are already supported, however they are not synchronized with the SmartEVSE built-in local lists. It could be one of the next action items to use the rfid.txt list with OCPP.

Also, it's a good idea to just truncate the 7th byte for now.

dingo35, 2024-07-14:
Cant we treat the 4byte identifiers, padded with 0's to 6byte, the same as the 6 byte identifiers? Why would we need to distinguish?

Second topic, which is now washed away with the closing of the previous PR, I still think we need the "freevend" switch, without authorization, on the dashboard.... We might even consider putting its default to "On" ?

matth-x, 2024-07-14:
The issue with the trailing 0s is the interoperability with other chargers. If you already have a few chargers on your parking lot and want to add a SmartEVSE, you would need to duplicate each identifier in the management system (probably by hand). Also, if someone tries out the SmartEVSE and sees this behavior, he would certainly regard it as a defect.

Okay, the freevend settings are done. I believe that it would be quite unusual to enable the freevend mode by default so I didn't do it for now. But we can revisit this for sure when we learn more about how people want to use OCPP on the SmartEVSE.

fluppie, 2024-07-14:
Our friends at Alfen are not adding the 0s. So I guess it's best practice to not add them?
[image]

3 tags from the above list are currently read as:
```
23:28:30.385 -> [MO] Send: [2,"1000006","Authorize",{"idTag":"948A17C300002F"}]

23:28:34.538 -> [MO] Send: [2,"1000009","Authorize",{"idTag":"049951D1000040"}]

23:28:38.559 -> [MO] Send: [2,"1000012","Authorize",{"idTag":"542E52D10000F8"}]
```

MarcelVuijst, 2024-09-16:
Hi.

Great project!
Can anyone show me how the rfix.txt must look like? I have tried to find an example but no results yet.
is there also a way to get a list of the stored RFID cards back from a Smart EVSE V3?

Thanks.
Marcel.

MarcelVuijst, 2024-09-17:
Thanks Dingo35! 

It works now. I have RFID tags with a serial number of 8 digest an had to add 4 zero's at the end.

fluppie, 2024-12-09:
@dingo35 @matth-x
Coming back on this. When using for example Wallbox charging point in Steve and 4-byte RFID's are scanned. The tags are not recognized by Steve, since they were put in with the trailing 0's to have compatibility with SmartEVSE.
It's an installation with 20 SmartEVSE's and 2 Wallboxes. Many users, so I don't want to create duplicates as a workaround.
Are we able to detect the true length of a scanned RFID badge? Are the 0's added by SmartEVSE or by the RFID reader itself?

dingo35, 2024-12-09:
No that is a real problem in the RFID standards, the length of the UID is unknown ...🥴

fluppie, 2024-12-09:
This is from the Wallbox
[image]
Tag is known in Steve as 147E2B870000
I have 60+ tags in the db and I cannot find a tag where 00 is a byte. So maybe there's a "safe way" to remove per 2 zero's?

4 byte tag, remove 00 00 ( 00 00 00 with the new readers I think?)
5 byte tag (if that exists) 00 ( 00 00 ) 
6 byte tag, don't remove anything ( 00 )
7 byte tag, only supported by new RFID readers

If I would let Alfen's join, I guess I would see the same "problem" as with the Wallbox chargers.

mstegen, 2024-12-09:
IIRC there are only 4 , 7 and 10 byte UID's.
The new reader supports 4 and 7, and will add the extra (3) zeros to the UID if it reads a 4 byte UID.

fluppie, 2024-12-10:
OK, so we would only need a solution for 4byte, that detects 00 00 (customers with the old RFID reader) and 00 00 00 (new reader) and then remove those to act more like Alfen and Wallbox EVSE's?
Is that something feasible?

dingo35, 2024-12-10:
Except that 7 byte cards with last three bytes 0x00 are valid cards....

dingo35, 2024-12-10:
https://stackoverflow.com/questions/37837730/mifare-cards-distinguish-between-4-byte-and-7-byte-uids#38009709

AFAIK our readers do not support the ATQA command....

dingo35, 2024-12-14:
@fluppie Could you test EXTENSIVELY, since you are one of the few users who can check it at the backend:
[0cbac1140de3.zip](https://github.com/user-attachments/files/18135691/0cbac1140de3.zip)
 
Strategy is now:
If we suspect the RFID was read with the old RFID reader, and bytes 5 and 6 are zero, we send a 4 byte UID, if not zero, we send a 6 byte UID (since the 7th byte is missing).
If we suspect the RFID was read with the new RFID reader, and bytes 5, 6,7 are zero, we send a 4 byte UID, if not zero, we send a 7 byte UID.

fluppie, 2024-12-16:
Yes, I have around 25-30 charging points use the old reader.
I only have one testversion at home with the new reader, but not in a final setup. Just on my desk for testing.

closed 2025-02-19 by dingo35

fluppie, 2025-02-19:
Just a short feedback, I have currently 4 connectors on a build based on Github 10/02. There we use the 'old' readers and 4 byte tags. That works as expected. Will also upgrade 2 connectors next week to the new version reader to see how it handles longer tags.