dingo35/SmartEVSE-3.5 #105: mqtt brokerer access password cannot contain '#' characters

rob040, 2024-07-19:
**Describe the bug**
With my setup it was impossible to access the mqtt brokerer (Mosquitto) with smartevse, whereas other IOT devices had no problems. 
The Mosquitto brokerer (running on raspberry pi as Home Assistant addon) log reports "Client SmartEVSE-xxxx disconnected, not authorised.", whereas I was sure the given PW was correct.
I needed to see what PW was actually send, so I created a debug version that printed the used mqqt password. From logfiles it became clear that the send password was missing the '#' character and following.
The issue must be in the smartevse webpage js script of MQTT config part

**Upload your config**
password is not shown, only password_set:true

**Upload a debug log**

**To Reproduce**
(evse mode is not important, software version 3.5.x and 3.6.x, tested on 3.6.3)
1. on smartevse webpage MQTT edit settings, fill all fields and enter password "rty!@#$%^&*()_+=-"
2. on (firefox) browser, open developer tools console
3. Hit the webpage "Save" button.
4. see on browser console the posted url.
   In my case: 
  `POST http://smartevse-xxxx/settings?mqtt_update=1&mqtt_host=192.168.x.xxx&mqtt_port=1883&mqtt_username=xxxxxxx&mqtt_password=rty!@`&mqtt_topic_prefix=SmartEVSE-xxxx

==> the password is incomplete

Since the text is not quoted nor escaped, there are more special characters that cause problems. (like &, quote, space, others?)

**Expected behavior**
password character set should be as broad as server allows (follow common practice), or documentation and config web page should make the limitations clear.

rob040, 2024-07-19:
Note: With this bug, I had frequent, as in every few days, **ESP hangups** that did not recover. i.e. webpage unreachable, front panel buttons were non-functional, no charging possible. No watchdog reset though.
In debug log I saw that all the MQTT activity continues despite of the authentication failure. Perhaps this causes memory leakage, causing it to hang after some time.

rob040, 2024-07-20:
Even after changing all my MQTT clients and the brokerer to use a PW with only letters and digits, the SmartEVSE still does not connect to brokerer: "MQTT:Disconnected"

dingo35, 2024-07-21:
First, upgrade to v3.6.6 .
Since hundreds of users are using MQTT succesfully, I suspect you have a configuration problem.
You could have a DNS problem, which you could diagnose by filling in IP address instead of a hostname.

rob040, 2024-07-21:
Updated to v3.6.6, debug version. rebooted. 
No DNS issues afaik; SMARTevse.local is found, and MQTT server gets IP address:

After several tries on webpage edit MQTT settings to set new PW, it finally got through it thick head.
I suspect that the regular GET does interfere with POST settings.

rob040, 2024-07-21:
What is the MainsMeter Set topic format?
a `SmartEVSE-xxxx/Set/MainsMeter=-50,-63,-72` is not recognized. Nothing on debug output other than L1,2,3: 0.0 A.

Any idea of creating a MQTT_API.md document similar to REST_API.md ?

dingo35, 2024-07-21:
RTFM

closed 2024-07-21 by dingo35

rob040, 2024-07-21:
Missed that part.
The values are not comma separated but colon separated.

rob040, 2024-07-21:
@dingo, you are again very quick in closing tickets ! :-(
There are multiple real issues listed here that needs attention.

dingo35, 2024-07-21:
Tickets are meant for bugs, not for people who have failed to read the manual.
Please be a productive member of the community instead of being a leecher.

What's your contribution going to be?