dingo35/SmartEVSE-3.5 #106: Wifi configuration: static IP address is lost after power cycle; switches to DHCP obtained IP

rob040, 2024-07-20:
**Describe the bug**
Wifi configuration initiated via LCD menu and setting a static IP address, this is lost after power cycle and switches to DHCP obtained IP

**Upload your config**
FW version 3.6.3

**Upload a debug log**
n.a.

**To Reproduce**
1. Enter wifi setup mode, 
2. Connect to AP (hurry up, there is only 90 seconds time to connect)
3. Browse to 192.168.4.1 (enter IP in browser, automatic redirection is not working)
4. Fill in IP settings: 192.168.2.35,192.168.2.254, 255.255.0.0, 192.168.2.254 (SSID and PW were already set)
5. Save, wait for loss of AP, connect to LAN WiFi and to new static IP (usually works)
6. power-cycle EVSE (leave power off for 20-30 sec)
7. Connect to IP (no response) 
8. check IP on LCD menu: IP is from DHCP pool
(in my network, DHCP pool is 192.168.2.100 - ...200)

**Expected behavior**
No loss of settings

**Screenshots**
setup:
<TODO: add screenshot> <can't attach pictures right now, problem with Firefox?>
reboot and power-cycle
<TODO: add screenshot>

dingo35, 2024-07-20:
Please upgrade to newest community release v3.6.6 when posting bug reports; a lot has changed in the wifi area since v3.6.3.

dingo35, 2024-07-20:
As to static IP,  since we abandoned WifiManager you should configure this in your router/AP....

rob040, 2024-07-20:
> As to static IP, since we abandoned WifiManager you should configure this in your router/AP....

Static IP cannot be configured in router, only a DHCP lease can be fixed in router.

dingo35, 2024-07-20:
Yes, and that fixes your ip address. Makes it "static", as in not changing.

closed 2024-07-20 by dingo35

rob040, 2024-07-20:
:-(

rob040, 2024-07-21:
> Yes, and that fixes your ip address. Makes it "static", as in not changing.

That is not a static IP address!
Static IP are set in the client and do not depend on DHCP.
The chosen method of req esptouch app is not future proof, as is your interpretation of 'static' IP

dingo35, 2024-07-21:
You are absolutely right. But since the invention of mDNS, normal users have no use case for a "real" hard static ip, usually not even for a hardened DHCP lease.
What remains are expert users with VLAN's, iptables, interface bonding; they usually manage to solve their own problems....

Please understand we have to navigate between large groups of users' requirements vs individual requirements.