dingo35/SmartEVSE-3.5 #107: Is it possible to start the charge from 1A?

legeantbleu, 2024-07-22:
Hello,

I have tesla model 3 which is able to start charging from 1A but the smart EVSE only allows you to start charging from 6A.

Is it possible to change this parameter?

Thank you.

Imaginous, 2024-07-22:
This is not possible. 
Also 6A is the minimum according to the standards.

dingo35, 2024-07-22:
As @Imaginous said, this is not according to standards. But if you are an expert user, and you know what you are doing, and you don't mind damaging stuff, AND you know how to compile yourself, you can find in the manual:
```
 If your want to give special compiler options, e.g. lower the minimum current to 5A:  PLATFORMIO_BUILD_FLAGS='-DDBG=1 -DMIN_CURRENT=5' pio run
```

legeantbleu, 2024-07-24:
Ok thanks.