dingo35/SmartEVSE-3.5 #108: Unable to compile using instructions provided

wouterbouvy, 2024-07-23:
**Describe the bug**
After following the instructions in the configuration.md I'm unable to compile due to errors in the mongoose webserver.

**Upload your config**
N/A

**Upload a debug log**
`Linking .pio/build/release/firmware.elf
/Users/wouter/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o:(.literal.packed_open+0x0): undefined reference to `mg_unpack'
/Users/wouter/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o:(.literal.packed_stat+0x0): undefined reference to `mg_unlist'
/Users/wouter/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o: in function `packed_open':
/Users/wouter/temp/SmartEVSE-3.5/SmartEVSE-3/src/mongoose.c:1893: undefined reference to `mg_unpack'
/Users/wouter/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o: in function `packed_stat':
/Users/wouter/temp/SmartEVSE-3.5/SmartEVSE-3/src/mongoose.c:1863: undefined reference to `mg_unpack'
/Users/wouter/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/wouter/temp/SmartEVSE-3.5/SmartEVSE-3/src/mongoose.c:1865: undefined reference to `mg_unlist'
/Users/wouter/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o: in function `mg_unpacked':
/Users/wouter/temp/SmartEVSE-3.5/SmartEVSE-3/src/mongoose.c:1850: undefined reference to `mg_unpack'
/Users/wouter/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/release/src/mongoose.c.o: in function `packed_list':
/Users/wouter/temp/SmartEVSE-3.5/SmartEVSE-3/src/mongoose.c:1877: undefined reference to `mg_unlist'
collect2: error: ld returned 1 exit status
*** [.pio/build/release/firmware.elf] Error 1
========================== [FAILED] Took 9.45 seconds ==========================`

**To Reproduce**
Install PlatformIO
Clone repo
Build with `pio run`

**Expected behavior**
A shiny new build firmware

**Screenshots**
[image]

dingo35, 2024-07-23:
I'm sorry, I cannot reproduce this, in both Linux and Windows it compiles fine, so you will have to look for the problem in your dev environment.....

If you look at the error messages you can see that they are unjust, since e.g. mg_,unpack _is_ defined.

Try pio run -t clean and retry?!?

wouterbouvy, 2024-07-24:
Running you command didn't help unfortunately.
I tried this on 2 different MacOS machines without success. So I thought I might be MacOS related.

However, after installing a fresh Ubuntu 24.04 LTS VM, installing platform and trying to compile, I run into the same issue.

dingo35, 2024-07-24:
If you look at the repository, in 
.github/workflows/*.yaml, you see how every push Ubuntu is installed, platformio is installed, and compilation is performed.

Follow the scripts....

closed 2024-07-24 by wouterbouvy