dingo35/SmartEVSE-3.5 #109: ESPHome code generaties API return code 411

rhannink, 2024-07-24:
**Describe the bug**
Using the ESPhome code on an ESP32 results since some time in a 411 return code and 2 warings that the error code was not valig en errorcode was reset

ESPHOME 2024.7.1
SMARTEVSE v3.6.6

**Upload your config**
```
interval:
  - interval: 5sec
    then:
      - http_request.post:
          url: !lambda |-
            std::string url;

            float l1 = id(returnedl1).state > 0 ? - id(currentl1).state : id(currentl1).state;
            float l2 = id(returnedl2).state > 0 ? - id(currentl2).state : id(currentl2).state;
            float l3 = id(returnedl3).state > 0 ? - id(currentl3).state : id(currentl3).state;

            url.append("http://192.168.1.153/currents?L1=");
            url.append(to_string(l1*10));
            url.append("&L2=");
            url.append(to_string(l2*10));
            url.append("&L3=");
            url.append(to_string(l3*10));
            return url.c_str();
```

**Debug log**
```
[11:20:30][E][http_request.arduino:117]: HTTP Request failed; URL: http://192.168.1.153/currents?L1=-20.000000&L2=-30.000000&L3=-30.000000; Code: 411
[11:20:30][E][component:164]: Component http_request set Error flag: unspecified
[11:20:31][E][component:176]: Component http_request cleared Error flag
```

**Solution**
add the following to the code (all the way to the bottom)
```
          headers:
            Content-Length: 0
```

dingo35, 2024-07-26:
Should be fixed in 854f659 . 

I meticulously copied your change, but I know how picky HA is on indentations; also we have a Dos/Windows file here (lines ending by ^M), which is not my platform.

Would you be so kind to test the file "ET-SM01\ smart\ meter\ module.yaml" in the master branch in your production environment?

hmmbob, 2024-07-26:
By the looks of it, spacing is all right

I think that moving it before the `url` is easier to read though:

```yaml
      - http_request.post:
          headers:
            Content-Length: 0
          url: !lambda |-
              std::string url;

              float l1 = id(returnedl1).state > 0 ? - id(currentl1).state : id(currentl1).state;
              float l2 = id(returnedl2).state > 0 ? - id(currentl2).state : id(currentl2).state;
              float l3 = id(returnedl3).state > 0 ? - id(currentl3).state : id(currentl3).state;

              url.append("http://192.168.207.121/currents?L1=");
              url.append(to_string(l1*10));
              url.append("&L2=");
              url.append(to_string(l2*10));
              url.append("&L3=");
              url.append(to_string(l3*10));
              return url.c_str();
```

rhannink, 2024-07-30:
> Should be fixed in [854f659](https://github.com/dingo35/SmartEVSE-3.5/commit/854f659fb05ed02b2fe1de25a3c8aa1c92b544a8) .
>
> I meticulously copied your change, but I know how picky HA is on [...]

I tested this code and it works fine.

Gr. remco

rhannink, 2024-07-30:
> By the looks of it, spacing is all right
>
> I think that moving it before the `url` is easier to read though:
>
> ```yaml
>       - http_request.post:
>           headers:
>             Content-Length: [...]

I think you are right. Putting the **header:** before the **url:** makes more sense.

Gr. Remco

closed 2024-07-30 by dingo35